espressif/esp-idf #5165: A stack overflow in task *** while running esp_http_client_perform on large HTTP header

agentzex, 2020-04-21:
This issue related to issue #5028 and commit   #471b40b

Hi
I just encountered this error while putting a pretty large header:

1.    I first used an https url and then I just got an "A stack overflow in task " panic , so it seems the print only working for http requests but not https, might want to add it.
2.    After i tried with http url instead, I got this error which explained better why there is a stack overflow, but what I failed to understand though, is how to fix this issue? I didn't see anyway to enlarge the headers buffer size, the "esp_http_client_set_header()" function didn't return any error and only after calling "esp_http_client_perform()" I received this error on runtime. My header is around 350 chars which is pretty standard for API tokens used in many http requests.
3.    Is it possible that if the buffer size for the headers is too small, to return an error on esp_http_client_set_header() instead? that way we can check the return value and avoid runtime panic errors.

Alvin1Zhang, 2020-04-23:
@agentzex Thanks for the report, we will look into.

shubhamkulkarni97, 2020-04-27:
> 1. The stackoverflow error wasn't related to the headers length size. It was related to the fact I run the code from app_main() with default stack size (3584) which was probably too small to fit the response from the http request, and it's probably not related to the headers length or the fact of using https url rather than http url.

You get stack overflow issue as you are using local buffer to store the response and the stack size is not adequate. To fix this issue, you can dynamically allocate memory to get the response. You can refer to event_handler in esp_http_client example https://github.com/espressif/esp-idf/blob/741960d5ce0394e8ecb99deef8a5ebf579cb45a2/examples/protocols/esp_http_client/main/esp_http_client_example.c#L41

> 2. I still got the "HTTP_HEADER: Buffer length is small to fit all the headers" error printed out but it gave me the response although throwing this error (no runtime panic). So my previous questions (2 and 3) remains, how to increase the headers buffer size to fit large headers? and if it's possible to return an error on esp_http_client_set_header() instead.

To increase buffer size, you can increase `buffer_size_tx` in `esp_http_client_config_t`, which is by default set to 512.
Even if the buffer size is small to fit all the headers at a time, last fitting header is sent. Hence it is not necessary to return an error on `esp_http_client_set_header()`.

agentzex, 2020-04-27:
Regarding first question: But the buffer I send in http_event_handler is still subtracted from the size of the current stack size no? if so how is it dynamic? wouldn't the crash be the same if the stack size is still smaller than http response size?
Why not trying to get the response size from the http header before actually writing it to the the buffer?

shubhamkulkarni97, 2020-04-27:
Buffer is allocated using `malloc` as shown here:
https://github.com/espressif/esp-idf/blob/741960d5ce0394e8ecb99deef8a5ebf579cb45a2/examples/protocols/esp_http_client/main/esp_http_client_example.c#L70
Hence, the memory will be allocated from heap and not from stack. This will not result in stack overflow even if stack size is less.

agentzex, 2020-04-27:
I tried 2 cases. When I did the HTTP call from app_main() with the default stack size, I got "a stack overflow in task main detected" (screenshot 1.jpg) . When I did the same exact HTTP call from a task created, I got "multi_heap_assert..." (screenshot 2.jpg).
In both cases, the device aborted and rebooted, so I don't think it matters if it's stack overflow or heap overflow , it's still crashing the device if the response is too big. The crash happened on the call to esp_http_client_perform() . I understand that If the user gives a big enough buffer when calling it , it should solve those issues, but  If the user doesn't give a buffer (which as I understand is the default behavior) then why those crashes happen? if the buffer was dynamically allocated to fit the response size that shouldn't happen no?
[image]
[image]

shubhamkulkarni97, 2020-04-27:
@agentzex Can you please share your test code in which you get `multi_heap_asssert`?

agentzex, 2020-04-27:
`void send_get(char* m_buffer){
    char r_url[512];
    sprintf(r_url, "%s", "https://www.howsmyssl.com");

    esp_http_client_config_t config = {
        .url = r_url,
    };

    esp_http_client_handle_t client = esp_http_client_init(&config);

    esp_err_t err = esp_http_client_perform(client);
    if (err == ESP_OK) {
        int length_read = esp_http_client_get_content_length(client);
        ESP_LOGI(TAG, "HTTP GET Status = %d, content_length = %d",esp_http_client_get_status_code(client), length_read);

        esp_http_client_read(client, m_buffer, MAX_RESPONSE_SIZE);
        ESP_LOGI(TAG, "HTTP returned data:\n%s", m_buffer);
    }
    else {
        ESP_LOGE(TAG, "HTTP GET request failed: %s", esp_err_to_name(err));
    }
    esp_http_client_cleanup(client);
}

static void http_task(void *pvParameters){
    char m_buffer[MAX_RESPONSE_SIZE + 1] = {0};

    send_get(m_buffer);
    vTaskDelete(NULL);
}

void app_main()
{
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
      ESP_ERROR_CHECK(nvs_flash_erase());
      ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    ESP_LOGI(TAG, "starting network scanner");
    wifi_init_sta(); //init for wifi
    
    

    xTaskCreate(&http_task, "http_task", 2000, NULL, 5, NULL);

    

}
`

mahavirj, 2020-04-30:
@agentzex 

>     xTaskCreate(&http_task, "http_task", 2000, NULL, 5, NULL);

This is very small stack size. Even for simple `printf` call in your task main routine, recommended stack size would be slightly more than 3KB. For typical TLS connection, requirement would be close to 7-8KB. You can fine tune by using `uxTaskGetStackHighWaterMark ` which helps to find peak utilization of stack memory.

- With such small stack, either there will be `stack overflow` hook getting triggered (if there is immediate context switch)
- Or it is also possible to have heap corruption issues, since stacks are also allocated from heap region itself

Can you please try to increase stack size and retry (recommended to use values fro existing IDF examples)?

agentzex, 2020-04-30:
I know that if I increase the stack size (in app_main) or the stack depth in a task it will solve this issue. The thing is that if I called a task and used most of it's memory let's say for other things, and then call  esp_http_client_perform() the outcome will be the same no? 
What I asked is why the SDK won't try to read only the content-length header before reading the whole response and so avoiding this errors. If the http response is too big, it doesn't matter what will be the stack size, it will still crash no?

shubhamkulkarni97, 2020-04-30:
@agentzex ,

> What I asked is why the SDK won't try to read only the content-length header before reading the whole response and so avoiding this errors. If the http response is too big, it doesn't matter what will be the stack size, it will still crash no?

Can you provide test code which can help reproduce reproduce your issue?

secureschwob, 2020-06-05:
Hi,
I am experiencing similar problems concerning `multi_heap_assert`, while trying to perform a POST request in a separate thread. Please find my minimal code, based on the original `esp_http_client` example, stripped of all methods except basic POST, below:

```C
#include <string.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "esp_system.h"
#include "nvs_flash.h"
#include "esp_event.h"
#include "esp_netif.h"
#include "protocol_examples_common.h"
#include "esp_tls.h"

#include "esp_http_client.h"

#define MAX_HTTP_RECV_BUFFER 512
#define MAX_HTTP_OUTPUT_BUFFER 2048
static const char *TAG = "HTTP_CLIENT";

// I haven't touched this handler function
esp_err_t _http_event_handler(esp_http_client_event_t *evt)
{
    static char *output_buffer;  // Buffer to store response of http request from event handler
    static int output_len;       // Stores number of bytes read
    switch(evt->event_id) {
        case HTTP_EVENT_ERROR:
            ESP_LOGD(TAG, "HTTP_EVENT_ERROR");
            break;
        case HTTP_EVENT_ON_CONNECTED:
            ESP_LOGD(TAG, "HTTP_EVENT_ON_CONNECTED");
            break;
        case HTTP_EVENT_HEADER_SENT:
            ESP_LOGD(TAG, "HTTP_EVENT_HEADER_SENT");
            break;
        case HTTP_EVENT_ON_HEADER:
            ESP_LOGD(TAG, "HTTP_EVENT_ON_HEADER, key=%s, value=%s", evt->header_key, evt->header_value);
            break;
        case HTTP_EVENT_ON_DATA:
            ESP_LOGD(TAG, "HTTP_EVENT_ON_DATA, len=%d", evt->data_len);
            /*
             *  Check for chunked encoding is added as the URL for chunked encoding used in this example returns binary data.
             *  However, event handler can also be used in case chunked encoding is used.
             */
            if (!esp_http_client_is_chunked_response(evt->client)) {
                // If user_data buffer is configured, copy the response into the buffer
                if (evt->user_data) {
                    memcpy(evt->user_data + output_len, evt->data, evt->data_len);
                } else {
                    if (output_buffer == NULL) {
                        output_buffer = (char *) malloc(esp_http_client_get_content_length(evt->client));
                        output_len = 0;
                        if (output_buffer == NULL) {
                            ESP_LOGE(TAG, "Failed to allocate memory for output buffer");
                            return ESP_FAIL;
                        }
                    }
                    memcpy(output_buffer + output_len, evt->data, evt->data_len);
                }
                output_len += evt->data_len;
            }

            break;
        case HTTP_EVENT_ON_FINISH:
            ESP_LOGD(TAG, "HTTP_EVENT_ON_FINISH");
            if (output_buffer != NULL) {
                // Response is accumulated in output_buffer. Uncomment the below line to print the accumulated response
                // ESP_LOG_BUFFER_HEX(TAG, output_buffer, output_len);
[... 133 lines cut ...]

0x4008741d: esp_system_abort at /opt/esp32_development/esp-idf/components/esp_system/system_api.c:68

0x40095b66: abort at /opt/esp32_development/esp-idf/components/newlib/abort.c:46

0x4014cbe3: __assert_func at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

0x400842cf: heap_caps_free at /opt/esp32_development/esp-idf/components/heap/heap_caps.c:271 (discriminator 1)

0x40095a61: free at /opt/esp32_development/esp-idf/components/newlib/heap.c:47

0x4014aca9: esp_transport_list_destroy at /opt/esp32_development/esp-idf/components/tcp_transport/transport.c:117

0x400f8f83: esp_http_client_cleanup at /opt/esp32_development/esp-idf/components/esp_http_client/esp_http_client.c:623

0x400d51b1: http_rest_with_url at /home/martin/1_Git/ba-sensor-privacy/00_code/00_sensor_node/esp32/code/workspace_experimental/esp_http_client_minimal/build/../main/esp_http_client_example.c:123

0x400d51bb: http_test_task at /home/martin/1_Git/ba-sensor-privacy/00_code/00_sensor_node/esp32/code/workspace_experimental/esp_http_client_minimal/build/../main/esp_http_client_example.c:133

0x40092941: vPortTaskWrapper at /opt/esp32_development/esp-idf/components/freertos/xtensa/port.c:143
```

Creating a large `output_buffer` at the stack and getting rid of malloc/free, it works fine. Could you please provide some advice on what am I doing wrong? Have I stripped some critical lines of code from the original example?

Thank you!

secureschwob, 2020-06-05:
Something similar is discussed in the ESP32 forums: https://esp32.com/viewtopic.php?t=1806. He also sees the `esp_wifi_internal_free_rx_buffer` in the monitor log and finds: "The problem only shows in a certain phase of the application, namely in the beginning when there is a period where data is consumed more quickly than can be delivered and the buffer runs empty."
While in my case it does not happen in the beginning, can it indeed be related to the RX buffer trying to be `free`d while it is not allocated?

shahpiyushv, 2020-06-05:
@secureschwob , irrespective of where the firmware crashes, this is not a desirable code. You are creating a new task every second and there is a possibility that some task takes more time to execute than the other, resulting into multiple simultaneous http_test_task running. Since they would be sharing the same static output_buffer and output_len, without any protection, there is a high chance of double free, which can cause a crash.

I hope I myself ain't missing anything in this code.

Alvin1Zhang, 2020-06-17:
@agentzex Thanks for reporting and sharing updates. Would you please help share if any updates/progress of the issue? Thanks.

agentzex, 2020-06-17:
Hi
Sorry I didn't have the time to look deeper at your latest comment about the behavior of the this lib. I think you can close this if you believe there wasn't a bug here. Thanks for the support :)

closed 2020-06-17 by agentzex

shubhamkulkarni97, 2020-06-17:
@secureschwob,

Response length was not cleared in event handler when response buffer is passed in `user_data`. We have added a fix which will be available on GitHub soon. Also, as @shahpiyushv mentioned, it is not desirable to create a task at each iteration.

For now, you can try the patch attached below.
[http_client_example_fix.patch.zip](https://github.com/espressif/esp-idf/files/4790772/http_client_example_fix.patch.zip)