espressif/esp-idf #5167: LEDC: Guru Meditation when calling `ledc_fade_func_install` after `esp_restart`

hb-kaertech, 2020-04-22:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version: v4
- Module or chip used: ESP32-WROOM-32D
- IDF version: v4.2-dev-1126-gd85d3d969f
- Build System: idf.py
- Compiler version: (crosstool-NG esp-2020r1) 8.2.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

I boot my DevKitC and do 'things" (ble, wifi, ...). Specifically, LEDC is set to blink a led on `LEDC_CHANNEL_0`.
At some point, I call `esp_restart`. When I call `ledc_fade_func_install(0)` after the restart, I get a guru meditation.

## Debug Logs

```
I (207) main: LEDC BUG DEMO (reset_reason:1)
I (207) main: LEDC BUG DEMO (reset_reason:3)
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x40084086  PS      : 0x00050034  A0      : 0x40082490  A1      : 0x3ffbe290  
0x40084086: ledc_fade_isr at .../esp-idf/components/driver/ledc.c:567 (discriminator 2)

0x40082490: _xt_lowint1 at .../esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105

A2      : 0x3ffc08b0  A3      : 0x00000000  A4      : 0x3ffc08b0  A5      : 0x400888cc  
0x400888cc: _frxt_int_enter at .../esp-idf/components/freertos/xtensa/portasm.S:119

A6      : 0x3ffc0200  A7      : 0x1300324c  A8      : 0x00000000  A9      : 0x00000020  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x0000cdcd  A15     : 0x00000804  SAR     : 0x0000000f  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
Core  0 was running in ISR context:
EPC1    : 0x400e6213  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40084086
0x400e6213: uart_hal_write_txfifo at .../esp-idf/components/soc/src/hal/uart_hal_iram.c:35

0x40084086: ledc_fade_isr at .../esp-idf/components/driver/ledc.c:567 (discriminator 2)

Backtrace:0x40084083:0x3ffbe290 0x4008248d:0x3ffbe2d0 0x4000bfed:0x3ffb2010 0x40088851:0x3ffb2020 0x400d577b:0x3ffb2040 0x400d5ca9:0x3ffb2070 0x400d2e7d:0x3ffb2090 0x400d17da:0x3ffb2100 0x400885f9:0x3ffb2130
0x40084083: ledc_fade_isr at .../esp-idf/components/driver/ledc.c:572

0x4008248d: _xt_lowint1 at .../esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105

0x40088851: vPortExitCritical at .../esp-idf/components/freertos/xtensa/port.c:419

0x400d577b: ledc_isr_register at .../esp-idf/components/driver/ledc.c:272 (discriminator 2)

0x400d5ca9: ledc_fade_func_install at .../esp-idf/components/driver/ledc.c:815

0x400d2e7d: app_main at .../ledc_bug/build/../src/main.c:43

0x400d17da: main_task at .../esp-idf/components/esp32/cpu_start.c:571

0x400885f9: vPortTaskWrapper at .../esp-idf/components/freertos/xtensa/port.c:143

Core  1 register dump:
PC      : 0x400e5f96  PS      : 0x00060634  A0      : 0x800d26fe  A1      : 0x3ffb2fd0  
0x400e5f96: esp_pm_impl_waiti at .../esp-idf/components/esp32/pm_esp32.c:486

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x80000001  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x800d24d2  A9      : 0x3ffb2fa0  
A10     : 0x00000000  A11     : 0x00060623  A12     : 0x00060620  A13     : 0x00060623  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x400e5f93:0x3ffb2fd0 0x400d26fb:0x3ffb2ff0 0x400899d6:0x3ffb3010 0x400885f9:0x3ffb3030
0x400e5f93: esp_pm_impl_waiti at .../esp-idf/components/esp32/pm_esp32.c:484

0x400d26fb: esp_vApplicationIdleHook at .../esp-idf/components/esp_common/src/freertos_hooks.c:63

0x400899d6: prvIdleTask at .../esp-idf/components/freertos/tasks.c:3385 (discriminator 1)

0x400885f9: vPortTaskWrapper at .../esp-idf/components/freertos/xtensa/port.c:143
```
I have added 'in' and 'out' logs in  `ledc_fade_isr`, and I can confirm that it gets called repetadly until the Guru Meditation. More specifically, `ledc_fade_isr` does nothing, as the `p_ledc_obj[speed_mode] == NULL)` condition is always true.
https://github.com/espressif/esp-idf/blob/d85d3d969ff4b42e2616fd40973d637ff337fae6/components/driver/ledc.c#L568

## Workaround

Calling `ledc_stop(LEDC_LOW_SPEED_MODE, LEDC_CHANNEL_0, 0);` before `esp_restart();` seems to fix the issue.

## Code
I made a small reproducible example here:
main.c: https://gist.github.com/hb-kaertech/08a3af0ab91b66c313c3b34723064175
sdkconfig: https://gist.github.com/hb-kaertech/69730a7fc1b5f9c52f05e5f4f599fac3

Alvin1Zhang, 2020-04-23:
@hb-kaertech Thanks for the detailed report, we will look into. Thanks.

hb-kaertech, 2020-05-06:
@Alvin1Zhang : did you have time to look into it?

allanchen24, 2020-05-06:
@hb-kaertech Sorry, I can't access your code link, Could you please attach your code here.

hb-kaertech, 2020-05-06:
@allanchen24 : I forgot to make them public. Can you access them now?

allanchen24, 2020-05-06:
@hb-kaertech Still can not access.

hb-kaertech, 2020-05-06:
@allanchen24 
main.c:
```
#include <driver/gpio.h>
#include <driver/ledc.h>
#include <driver/uart.h>
#include <stddef.h>

#define LOG_LOCAL_LEVEL ESP_LOG_VERBOSE
#include <esp_log.h>

static const char *TAG = "main";
#define GPIO_NUMBER 26

static xQueueHandle event_queue = NULL;

static void xgpio_task(void *pvParameters)
{
    (void) pvParameters;
    for(;;) {
        int dummy;
        xQueueReceive(event_queue, &dummy, portMAX_DELAY);
        esp_err_t eret = ledc_set_duty_and_update(LEDC_LOW_SPEED_MODE, LEDC_CHANNEL_0, 1U << LEDC_TIMER_13_BIT, 0);
        if (eret != ESP_OK) {
            ESP_LOGE(TAG, "ledc_set_duty_and_update failed (%d)", eret);
        }
    }
}

void app_main(void)
{
    uart_set_baudrate(0, 2000000);
    ESP_LOGI(TAG, "LEDC BUG DEMO (reset_reason:%d)", esp_reset_reason());

    esp_err_t eret;

    event_queue = xQueueCreate(10, sizeof(int));

    xTaskCreatePinnedToCore(xgpio_task, "xgpio_task", 2048, NULL, tskIDLE_PRIORITY+3, NULL, 0);

    eret = gpio_install_isr_service(0);
    if (eret != ESP_OK) {
        ESP_LOGE(TAG, "gpio_install_isr_service failed (%d)", eret);
    }

    eret = ledc_fade_func_install(0);
    if (eret != ESP_OK) {
        ESP_LOGE(TAG, "ledc_fade_func_install failed (%d)", eret);
    }

    ledc_timer_config_t ledc_timer = {
        .duty_resolution = LEDC_TIMER_13_BIT,
        .freq_hz = 1,
        .speed_mode = LEDC_LOW_SPEED_MODE,
        .timer_num = LEDC_TIMER_0,
        .clk_cfg = LEDC_AUTO_CLK,
    };

    eret = ledc_timer_config(&ledc_timer);
    if (eret != ESP_OK) {
        ESP_LOGE(TAG, "ledc_timer_config failed (%d)", eret);
    }

    ledc_channel_config_t provisioning_led_conf = {
        .channel    = LEDC_CHANNEL_0,
        .duty       = 1U << LEDC_TIMER_13_BIT,
        .gpio_num   = GPIO_NUMBER,
        .speed_mode = LEDC_LOW_SPEED_MODE,
        .hpoint     = 0,
        .timer_sel  = LEDC_TIMER_0
[... 1564 lines cut ...]
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# End of deprecated options
```

allanchen24, 2020-05-06:
@hb-kaertech  The reason why your code crashed is you always wait queue in ISR, it will cause watchdog timeout.  
> `xQueueReceive(event_queue, &dummy, portMAX_DELAY);`

hb-kaertech, 2020-05-06:
> @hb-kaertech The reason why your code crashed is you always wait queue in ISR, it will cause watchdog timeout.
> 
> > `xQueueReceive(event_queue, &dummy, portMAX_DELAY);`

What do you mean exactly? `xgpio_task` is a regular low-priority freertos task. How is this related to an ISR?

allanchen24, 2020-05-06:
Aha, I'm soooo sorry, I read it wrong.

hb-kaertech, 2020-05-11:
@allanchen24 : Makes senses.

However, I am not sure what the right fix is here. There is not such function as a generic 'LEDC initialize'.
Imagine this convoluted case:
1- ESP32 is running with fw v1.0. This firmware makes use of ledc channel 1.
2- OTA kicks in, and updates the device to fw v2.0
3- Device reboots
4- fw v2.0 switches to using ledc channel 2, thus only initializes ledc channel 2 before calling `ledc_fade_func_install`. Channel 1 is left in the previous case and sends spurious irqs.
5- => Same problem as my issue?

Does it means I should call `ledc_stop` on every channel on startup? Seems like to the the only way to make sure to get back in a 'known' configuration.

koobest, 2020-05-13:
I think you should call `ledc_fade_func_install(0);` after ` ledc_channel_config`

hb-kaertech, 2020-05-18:
I refactored my code to init and deinit all ledc_channels before calling `ledc_fade_func_install(0)`.

I think the ledc lib should do it on init, but I am fine with my workaround.

closed 2020-05-18 by hb-kaertech

hoeken, 2023-10-20:
@hb-kaertech i'm having the same issue here.  do you mind explaining how you did the deinit / init?  i've tried all sorts of combinations and am still having issues.

Its slightly better based on this thread.  I went from the code hanging on the first ledc_fade_func_install(0) call after a reboot during fading. Now it will reboot, then crash with a "Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)." on the first time I call a fade.  The 2nd reboot it will be fine.  Better, but I would love to figure this thing out 100%.

brightproject, 2025-07-21:
> i'm having the same issue here. do you mind explaining how you did the deinit / init? i've tried all sorts of combinations and am still having issues.

Hello @hoeken and @hb-kaertech 
Unfortunately, it is difficult to obtain official support🙄
My backlight function is:
```c++
void guition_esp32s3_backlight_init(void)
{ 
  ledc_timer_config_t ledc_timer = { 
  .speed_mode = LEDC_LOW_SPEED_MODE, 
  .duty_resolution = LEDC_TIMER_8_BIT, 
  .timer_num = GUITION_ESP32_BACKLIGHT_LEDC_TIMER, 
  .freq_hz = 1000, // Set output frequency at 1 kHz 
  // .clk_cfg = LEDC_USE_RC_FAST_CLK, 
  .clk_cfg = LEDC_AUTO_CLK, 
  }; 
  ESP_ERROR_CHECK(ledc_timer_config(&ledc_timer)); 
  
  // Prepare and then apply the LEDC PWM channel configuration 
  ledc_channel_config_t ledc_channel = { 
  .gpio_num = GUITION_ESP32_PIN_BCKL, 
  .speed_mode = LEDC_LOW_SPEED_MODE, 
  .channel = GUITION_ESP32_BACKLIGHT_LEDC_CHANNEL, 
  .intr_type = LEDC_INTR_DISABLE, 
  .timer_sel = GUITION_ESP32_BACKLIGHT_LEDC_TIMER, 
  .duty = 0, // Set duty to 0% 
  .hpoint = 0, 
  }; 
  ESP_ERROR_CHECK(ledc_channel_config(&ledc_channel)); 
  // ESP_ERROR_CHECK(ledc_set_duty(LEDC_LOW_SPEED_MODE, GUITION_ESP32_BACKLIGHT_LEDC_CHANNEL, 50));
  // ESP_ERROR_CHECK(ledc_update_duty(LEDC_LOW_SPEED_MODE, GUITION_ESP32_BACKLIGHT_LEDC_CHANNEL));
  
  // Install the fade-in service (must be done once)
  static bool fade_service_installed = false;
  if (!fade_service_installed) {
  ESP_ERROR_CHECK(ledc_fade_func_install(0)); // Install the fade-in function
  fade_service_installed = true; // Mark setup complete
  }
  
  // Start fade to 100% (255)
  ESP_ERROR_CHECK(ledc_set_duty(LEDC_LOW_SPEED_MODE, GUITION_ESP32_BACKLIGHT_LEDC_CHANNEL, 255));
  ESP_ERROR_CHECK(ledc_update_duty(LEDC_LOW_SPEED_MODE, GUITION_ESP32_BACKLIGHT_LEDC_CHANNEL));
  
  // Start fade
  ESP_ERROR_CHECK(ledc_fade_start(LEDC_LOW_SPEED_MODE, GUITION_ESP32_BACKLIGHT_LEDC_CHANNEL, LEDC_FADE_NO_WAIT));

}
``` 
`esp32s3` reboots with this error:

> E (501) ledc: ledc_fade_start(1471): LEDC fade service not installed
> ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x4203a7c4
> file: "src/RTOS_LVGL_9_simple.cpp" line 83
> func: void guition_esp32s3_backlight_init()
> expression: ledc_fade_start(LEDC_LOW_SPEED_MODE, GUITION_ESP32_BACKLIGHT_LEDC_CHANNEL, LEDC_FADE_NO_WAIT)
> 
> abort() was called at PC 0x4037d353 on core 1