espressif/esp-idf #5168: Updated from IDF 2019r1 to 2020r1 - Can't flash using esptool

akhilv1, 2020-04-22:
## Environment

- Development Kit:      Custom Board
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.2-dev-1126-gd85d3d969
- Build System:         [Make|CMake|idf.py]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System:     macOS
- Using an IDE?: No
- Power Supply:  USB
- Programmer: SparkFun FT231X Breakout

## Problem Description

Before I updated the IDF to the latest version, make flash monitor used to work as expected. Now, it compiles, but when it tries to flash, it can't find the serial port specified in menuconfig.

`Toolchain path: /Users/aveluru/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp-2020r1
Compiler version: 8.2.0
Python requirements from /Users/aveluru/esp/esp-idf/requirements.txt are satisfied.

App "mqtt_tcp" version: v4.2-dev-1126-gd85d3d969
Flashing binaries to serial port '/dev/cu.usbserial-DN01QCY8' (app at offset 0x10000)...
esptool.py v3.0-dev
Serial port "/dev/cu.usbserial-DN01QCY8"
Traceback (most recent call last):
  File "/Users/aveluru/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 3430, in <module>
    _main()
  File "/Users/aveluru/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 3423, in _main
    main()
  File "/Users/aveluru/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 3087, in main
    esp = chip_class(each_port, initial_baud, args.trace)
  File "/Users/aveluru/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 261, in __init__
    self._port = serial.serial_for_url(port)
  File "/Library/Python/2.7/site-packages/serial/__init__.py", line 88, in serial_for_url
    instance.open()
  File "/Library/Python/2.7/site-packages/serial/serialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port "/dev/cu.usbserial-DN01QCY8": [Errno 2] No such file or directory: '"/dev/cu.usbserial-DN01QCY8"'
make: *** [flash] Error 1`

make simple_monitor also can't find the port, so I assume this is an issue with esptool.
`Toolchain path: /Users/aveluru/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp-2020r1
Compiler version: 8.2.0
Python requirements from /Users/aveluru/esp/esp-idf/requirements.txt are satisfied.
--- forcing DTR inactive
--- forcing RTS inactive
could not open port '"/dev/cu.usbserial-DN01QCY8"': [Errno 2] could not open port "/dev/cu.usbserial-DN01QCY8": [Errno 2] No such file or directory: '"/dev/cu.usbserial-DN01QCY8"'
make: *** [simple_monitor] Error 1`

If I run make and flash using the supplied flash command, it works as expected.
`python /Users/aveluru/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port "/dev/cu.usbserial-DN01QCY8" --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0xd000 /Users/aveluru/esp/esp-idf/examples/protocols/mqtt/tcp/build/ota_data_initial.bin 0x1000 /Users/aveluru/esp/esp-idf/examples/protocols/mqtt/tcp/build/bootloader/bootloader.bin 0x10000 /Users/aveluru/esp/esp-idf/examples/protocols/mqtt/tcp/build/mqtt_tcp.bin 0x8000 /Users/aveluru/esp/esp-idf/examples/protocols/mqtt/tcp/build/partitions.bin`

The only way to see the serial output is through another program such as screen or pyserial

igrr, 2020-04-22:
Hi @akhilv1, this issue was reported as https://github.com/espressif/esp-idf/issues/5124. Please check that thread for the temporary workaround and click subscribe there to get notified when the fix is merged.

akhilv1, 2020-04-22:
Awesome, thanks for the quick reply, looks like specifying the port works well.

closed 2020-04-22 by akhilv1