espressif/esp-idf #5169: Issues with xtensa-esp32-elf-gcc

JoeyMulder, 2020-04-23:
Hi there,

While trying to make frozen bytecode for an ESP32 device, I ran into the following error:

xtensa-esp32-elf-gcc: error: unrecognized command line option '-mfix-esp32-psram-cache-issue'

When I run "xtensa-esp32-elf-gcc --version" I get:

xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-61-gab8375a) 5.2.0

Any way to avoid the command line option or to allow it to work?

Thanks.

Alvin1Zhang, 2020-04-23:
@JoeyMulder Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

JoeyMulder, 2020-04-23:
@Alvin1Zhang I'll try add what I can find. Note I'm following this split up tutorial: https://forum.pycom.io/topic/1919/how-i-built-a-custom-firmware-for-frozen-bytecode

This is all for pycom devices though the hardware isn't used. 

``git describe --tags`` returns v1.20.2.rc6-1-ge264b4c24 which is just the newest version of pycom firmware. 
This is on Linux Mint 19.2 Cinnamon using VirtualBox on windows 10. 

The actual command being run is very convoluted: 
`xtensa-esp32-elf-gcc -DESP_PLATFORM -Wpointer-arith -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wl,-EL -mlongcalls -Wall -ffunction-sections -fdata-sections -fno-common -Wno-sign-compare -Wno-old-style-declaration -fstrict-volatile-bitfields -Wno-error=deprecated-declarations -Wno-unused-parameter -DWITH_POSIX  -mfix-esp32-psram-cache-issue -Os -nostdlib -std=gnu99 -g3 -ggdb -fstrict-volatile-bitfields -Iboards/WIPY -I. -I.. -Ihal -Iutil -Imods -Itelnet -Iftp -Ilora -Ilte -Ican -Ibootloader -Ifatfs/src/drivers -Ilittlefs -Ibuild/WIPY/release -Ibuild/WIPY/release/genhdr -I/root/pycom-esp-idf/components/bootloader_support/include -I/root/pycom-esp-idf/components/bootloader_support/include_priv -I/root/pycom-esp-idf/components/bootloader_support/include_bootloader -I/root/pycom-esp-idf/components/mbedtls/mbedtls/include -I/root/pycom-esp-idf/components/mbedtls/port/include -I/root/pycom-esp-idf/components/driver/include -I/root/pycom-esp-idf/components/driver/include/driver -I/root/pycom-esp-idf/components/heap/include -I/root/pycom-esp-idf/components/esp32 -I/root/pycom-esp-idf/components/esp32/include -I/root/pycom-esp-idf/components/esp_ringbuf/include -I/root/pycom-esp-idf/components/esp_event/include -I/root/pycom-esp-idf/components/esp_adc_cal/include -I/root/pycom-esp-idf/components/soc/include -I/root/pycom-esp-idf/components/soc/esp32/include -I/root/pycom-esp-idf/components/expat/include -I/root/pycom-esp-idf/components/freertos/include -I/root/pycom-esp-idf/components/json/include -I/root/pycom-esp-idf/components/expat/include -I/root/pycom-esp-idf/components/lwip/include/lwip -I/root/pycom-esp-idf/components/lwip/include/lwip/port -I/root/pycom-esp-idf/components/lwip/include/apps -I/root/pycom-esp-idf/components/lwip/port/esp32/include -I/root/pycom-esp-idf/components/lwip/lwip/src/include -I/root/pycom-esp-idf/components/lwip/include/lwip/posix -I/root/pycom-esp-idf/components/newlib/include -I/root/pycom-esp-idf/components/newlib/platform_include -I/root/pycom-esp-idf/components/nvs_flash/include -I/root/pycom-esp-idf/components/spi_flash/include -I/root/pycom-esp-idf/components/tcpip_adapter/include -I/root/pycom-esp-idf/components/log/include -I/root/pycom-esp-idf/components/sdmmc/include -I/root/pycom-esp-idf/components/vfs/include -I/root/pycom-esp-idf/components/bt/include -I/root/pycom-esp-idf/components/bt/bluedroid/device/include -I/root/pycom-esp-idf/components/bt/bluedroid/device/include/device -I/root/pycom-esp-idf/components/bt/bluedroid/bta/dm -I/root/pycom-esp-idf/components/bt/bluedroid/bta/hh -I/root/pycom-esp-idf/components/bt/bluedroid/bta/include -I/root/pycom-esp-idf/components/bt/bluedroid/bta/sys/include -I/root/pycom-esp-idf/components/bt/bluedroid/common/include -I/root/pycom-esp-idf/components/bt/bluedroid/stack/include -I/root/pycom-esp-idf/components/bt/bluedroid/stack/gatt/include -I/root/pycom-esp-idf/components/bt/bluedroid/stack/gap/include -I/root/pycom-esp-idf/components/bt/bluedroid/stack/l2cap/include -I/root/pycom-esp-idf/components/bt/bluedroid/btcore/include -I/root/pycom-esp-idf/components/bt/bluedroid/osi/include -I/root/pycom-esp-idf/components/bt/bluedroid/hci/include -I/root/pycom-esp-idf/components/bt/bluedroid/gki/include -I/root/pycom-esp-idf/components/bt/bluedroid/api/include -I/root/pycom-esp-idf/components/bt/bluedroid/btc/include -I/root/pycom-esp-idf/components/coap/libcoap/include/coap -I/root/pycom-esp-idf/components/coap/libcoap/examples -I/root/pycom-esp-idf/components/coap/port/include -I/root/pycom-esp-idf/components/coap/port/include/coap -I/root/pycom-esp-idf/components/mdns/include -I../lib/mp-readline -I../lib/netutils -I../lib/oofatfs -I../lib -I../drivers/sx127x -I../ports/stm32 -I/root/pycom-esp-idf/components/openthread/src -DMICROPY_NLR_SETJMP=1 -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H -DESP_PLATFORM -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -DWITH_POSIX -DREGION_AS923 -DREGION_AU915 -DREGION_EU868 -DREGION_US915 -DREGION_CN470 -DREGION_IN865 -DBASE=0 -DPYBYTES=1  -DVARIANT=0 -DNDEBUG -D BOOTLOADER_BUILD=1 -c -MD -o build/WIPY/release/bootloader/bootloader.o bootloader/bootloader.c`

And removing the command line option that is giving me trouble allows it to run but I don't know if it can be done without issues later.

Alvin1Zhang, 2020-04-23:
@JoeyMulder Thanks for the details, we will look into. Thanks.

negativekelvin, 2020-04-24:
Toolchain is too old, guide is from 2017 but you are cloning latest version of their code.

projectgus, 2021-01-04:
Hi @JoeyMulder,

Sorry for the long gap between replies by @negativekelvin is right. The tutorial from 2017 is specifying an old version of the toolchain to download. However you have a newer pycom firmware that uses a newer pycom-esp-idf (pycom have their own "esp-idf" fork with their own modifications), so it requires a different toolchain.

There are probably some warnings in the build log talking about the toolchain version being wrong as well, if you look for them.

If you're still using the same Pycom version, it may be enough to download the toolchain linked by pycom here for the version of the firmware you have, and replace the older one with it:
https://github.com/pycom/pycom-micropython-sigfox/tree/v1.20.2.rc6#the-esp32-version

If this isn't enough, suggest replying to the forum post or opening an issue on the [pycom-esp-idf](https://github.com/pycom/pycom-esp-idf) repo for help.

closed 2021-01-04 by projectgus