espressif/esp-idf #5170: xTimerCreateStatic aborts (TLSF)

boarchuz, 2020-04-23:
**Using feature/freertos_upstream_plus_tlsf_allocator_preview** (v4.2-dev-1181-g1757af235)

## Problem Description

abort() is called on xTimerCreateStatic, via:

https://github.com/espressif/esp-idf/blob/1757af23500f67b28d6ff70446bdb72de5444950/components/freertos/timers.c#L326-L327

CONFIG_FREERTOS_SUPPORT_STATIC_ALLOCATION=y

It seems these timer structs have changed in FreeRTOS 10 yet this assertion hasn't been updated (upstream?).

## Debug Log

<details>
<summary>Click</summary>

> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0030,len:4
> load:0x3fff0034,len:7196
> load:0x40078000,len:13696
> load:0x40080400,len:4000
> 0x40080400: _init at ??:?
> 
> entry 0x40080688
> I (29) boot: ESP-IDF v4.2-dev-1181-g1757af235 2nd stage bootloader
> I (29) boot: compile time 11:52:27
> I (29) boot: chip revision: 1
> I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
> I (40) boot.esp32: SPI Speed      : 40MHz
> I (45) boot.esp32: SPI Mode       : DIO
> I (50) boot.esp32: SPI Flash Size : 4MB
> I (54) boot: Enabling RNG early entropy source...
> I (60) boot: Partition Table:
> I (63) boot: ## Label            Usage          Type ST Offset   Length
> I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
> I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
> I (85) boot:  2 factory          factory app      00 00 00010000 00100000
> I (93) boot: End of partition table
> I (97) boot_comm: chip revision: 1, min. application chip revision: 0
> I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x05758 ( 22360) map
> I (122) esp_image: segment 1: paddr=0x00015780 vaddr=0x3ffb0000 size=0x02310 (  8976) load
> I (126) esp_image: segment 2: paddr=0x00017a98 vaddr=0x40080000 size=0x00404 (  1028) load
> 0x40080000: _WindowOverflow4 at /home/boarchuz/Espressif/IDF/tlsf/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1815
> 
> I (131) esp_image: segment 3: paddr=0x00017ea4 vaddr=0x40080404 size=0x08174 ( 33140) load
> I (154) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x128cc ( 75980) map
> 0x400d0020: _stext at ??:?
> 
> I (184) esp_image: segment 5: paddr=0x000328f4 vaddr=0x40088578 size=0x02204 (  8708) load
> 0x40088578: xQueueGenericSendFromISR at /home/boarchuz/Espressif/IDF/tlsf/esp-idf/components/freertos/queue.c:1106
> 
> I (193) boot: Loaded app from partition at offset 0x10000
> I (194) boot: Disabling RNG early entropy source...
> I (195) cpu_start: Pro cpu up.
> I (199) cpu_start: Application information:
> I (204) cpu_start: Project name:     template-app
> I (209) cpu_start: App version:      1d699ff-dirty
> I (215) cpu_start: Compile time:     Apr 23 2020 12:21:47
> I (221) cpu_start: ELF file SHA256:  b3f98fcfecb20244...
> I (227) cpu_start: ESP-IDF:          v4.2-dev-1181-g1757af235
> I (233) cpu_start: Starting app cpu, entry point is 0x40081280
> 0x40081280: call_start_cpu1 at /home/boarchuz/Espressif/IDF/tlsf/esp-idf/components/esp32/cpu_start.c:277
> 
> I (225) cpu_start: App cpu up.
> I (244) heap_init: Initializing. RAM available for dynamic allocation:
> I (251) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
> I (257) heap_init: At 3FFB2C98 len 0002D368 (180 KiB): DRAM
> I (263) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
> I (269) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
> I (276) heap_init: At 4008A77C len 00015884 (86 KiB): IRAM
> I (282) cpu_start: Pro cpu start user code
> I (300) spi_flash: detected chip: generic
> I (301) spi_flash: flash io: dio
> I (301) cpu_start: Starting scheduler on PRO CPU.
> I (0) cpu_start: Starting scheduler on APP CPU.
> /home/boarchuz/Espressif/IDF/tlsf/esp-idf/components/freertos/timers.c:327 (xTimerCreateStatic)- assert failed!
> 
> abort() was called at PC 0x4008a303 on core 0
> 0x4008a303: xTimerCreateStatic at /home/boarchuz/Espressif/IDF/tlsf/esp-idf/components/freertos/timers.c:327 (discriminator 1)
> 
> 
> Backtrace:0x4008519b:0x3ffb6310 0x40084e6d:0x3ffb6330 0x400875da:0x3ffb6350 0x4008a303:0x3ffb63c0 0x400d2925:0x3ffb63f0 0x400d14b6:0x3ffb6410 0x400877a5:0x3ffb6440
> 0x4008519b: panic_abort at /home/boarchuz/Espressif/IDF/tlsf/esp-idf/components/esp_system/panic.c:330
> 
> 0x40084e6d: esp_system_abort at /home/boarchuz/Espressif/IDF/tlsf/esp-idf/components/esp_system/system_api.c:68
> 
> 0x400875da: abort at /home/boarchuz/Espressif/IDF/tlsf/esp-idf/components/newlib/abort.c:46
> 
> 0x4008a303: xTimerCreateStatic at /home/boarchuz/Espressif/IDF/tlsf/esp-idf/components/freertos/timers.c:327 (discriminator 1)
> 
> 0x400d2925: app_main at /home/boarchuz/Espressif/Projects/tlsf_test/build/../main/main.c:16
> 
> 0x400d14b6: main_task at /home/boarchuz/Espressif/IDF/tlsf/esp-idf/components/esp32/cpu_start.c:571
> 
> 0x400877a5: vPortTaskWrapper at /home/boarchuz/Espressif/IDF/tlsf/esp-idf/components/freertos/xtensa/port.c:142

</details>

## Code

<details>
<summary>Click</summary>

```
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/timers.h"
#include "esp_log.h"

static const char *TAG = "tlsftest";

static void timer_expiry_handler(TimerHandle_t timer_handle)
{
    ESP_LOGD(TAG, "expired");
}

void app_main(void)
{
    static StaticTimer_t timer_data;
    xTimerCreateStatic(TAG, (2000 / portTICK_PERIOD_MS), pdTRUE, NULL, timer_expiry_handler, &timer_data);
    vTaskDelay(portMAX_DELAY);
}
```
</details>

uLipe, 2020-04-23:
Thanks for reporting this issue @boarchuz.

We are working to provide a fix of this soon.

uLipe, 2020-04-24:
@boarchuz , the fix is now available.

Thanks again for your help.

boarchuz, 2020-04-25:
Compiling now, thanks @uLipe

closed 2020-04-25 by boarchuz