espressif/esp-idf #5173: RMT WS2812 Driver flickering and crash problem

mburger82, 2020-04-23:
## Environment

- Development Kit:      `none`
- Module or chip used:  `ESP32-WROOM-32D`
- IDF version:  `v4.2-dev`
- Build System:  `Make`
- Compiler:  `xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0`
- Operating System:     `Linux in virtualBox on Windows`
- Using an IDE?:  `No, VSCode as Editor`
- Power Supply:       `external 5V (12V -> 5V)`

## Problem Description

Im tinkering with WS2812/WS2815 since some time.
The RMT seams to be the perfect Hardware Module to drive those LED's with their weird timing needs. 
But, the RMT-WS2812-Implementations had always the Problem, that there was some occasional flickering involved.

In v4.2-dev there is a official example/driver for ws2812. I gave the driver a chance, but could reconstruct the flickering again.

### Expected Behavior
* No Flickering when driving up to 8 lanes of WS2812 LED-Strips.
** as long as there is not too much load on the rest of the processor at least...

### Actual Behavior
* If only the original example is running, the led-strip works fine. Even with 300 LED's
* with wifi active and connected, the flickering happens occasionally. The more LED's, the more likely. ** min 22 I guess, (3 Bytes * 22 = 66 -> more then one memory bank in RMT module)
* Driver works up to 6 stripes (RMT0 - RMT5)
** if RMT7 & RMT8 are activated, the system crashes.

### Steps to reproduce

[image]

### Code to reproduce this issue
I made an [example](https://github.com/mburger82/flickering_led_stripe_demo) with compiler switches to activate different features

## Debug Logs
* crashlog when 8 LED-Stripes are active (wifi ssid and password was not tpyed in, but does not change anything)

```
Hard resetting via RTS pin...
MONITOR
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7148
load:0x40078000,len:14848
load:0x40080400,len:4568
entry 0x400806f0
I (28) boot: ESP-IDF v4.2-dev-dirty 2nd stage bootloader
I (28) boot: compile time 14:20:02
I (28) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 16MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
[... 108 lines cut ...]
Core 1 register dump:
PC      : 0x40144f5e  PS      : 0x00060534  A0      : 0x800d21c1  A1      : 0x3ffbbf40  
0x40144f5e: esp_pm_impl_waiti at /home/mburger/esp/esp-idf/components/esp32/pm_esp32.c:486

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x80000001  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x800d2066  A9      : 0x3ffbbf10  
A10     : 0x00000000  A11     : 0x00060f23  A12     : 0x00060f20  A13     : 0x00000001  
A14     : 0x00060f20  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 32be4efeed26e8e4e0b82c14b0cabb7222d62de45651aab87bdfc91b46bf7fd3

Backtrace: 0x40144f5b:0x3ffbbf40 0x400d21be:0x3ffbbf60 0x400946cd:0x3ffbbf80 0x400935a9:0x3ffbbfa0
0x40144f5b: esp_pm_impl_waiti at /home/mburger/esp/esp-idf/components/esp32/pm_esp32.c:484

0x400d21be: esp_vApplicationIdleHook at /home/mburger/esp/esp-idf/components/esp_common/src/freertos_hooks.c:63

0x400946cd: prvIdleTask at /home/mburger/esp/esp-idf/components/freertos/tasks.c:3382 (discriminator 1)

0x400935a9: vPortTaskWrapper at /home/mburger/esp/esp-idf/components/freertos/port.c:143
```

## Other items if possible

I can upload sdkconfig/coredump and stuff if needed. But the Problem is not particularly difficult to reproduce...

suda-morris, 2020-04-24:
Suggest to put RMT init code and RMT task into another core (e.g. Core1), which can help to install RMT interrupt on the core different to WiFi.  Currently the example code will install the RMT interrupt on Core0 (because you do the initialization just in the app_main).

mburger82, 2020-04-24:
Very good hint. (It seams to work so far... the flickering. Not the 8 channel thing)
I remember trying around with that some time ago, but never got it to work good enough.
Im also putting in some more code in my flicker-demo-project to test further.
In the End it should be capable of receiving data for about 1600 LED's via UDP and/or MQTT and update it ideally about 30times/second (at least via UDP)

The 8 Channel Problem is still present. 
But in the End I could work around that.
I push to my flicker-demo-project as soon as i find something.

Thank you anyway for the fast response!

suda-morris, 2022-05-16:
A good news is, on esp32-s3, RMT has the DMA ability, which will reduce the CPU interrupt of RMT ping-pong transaction by a lot. We can allocate a big enough buffer to store the RGB data and the DMA will handle the transaction without CPU intervention.
See the [doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/rmt.html#_CPPv4N23rmt_tx_channel_config_t8with_dmaE) for more information.

Enable the DMA feature in the driver is quite simple, only need to set the `with_dma` flag of `rmt_tx_channel_config_t`.

[Test code](https://github.com/espressif/esp-idf/blob/master/components/driver/test_apps/rmt/main/test_rmt_tx.c#L158) that can be helpful.

closed 2022-05-16 by suda-morris