espressif/esp-idf #5175: OTA Download problem from aws s3

lakshmisivareddy, 2020-04-24:
Hi i am trying to download .bin file from aws s3

i am  downloading file by 512 bits chunks and writing to spi flash   
but after downlading 10K  i am getting esp_tls_conn_read error  
this is the error message :

ota_size downloaded  is 10240:
HTTP chunk encoding Status = 206, content_length = 512
esp_get_free_heap_size  left 77672
Range Downloading : bytes=10240-1075
E (24445) esp-tls: read error :-29184:
E (24445) TRANS_SSL: esp_tls_conn_read error, errno=Success
esp_get_free_heap_size  OTA  73208
Range Downloading : bytes=10240-1075
E (24462) esp-tls: read error :-29184:
E (24463) TRANS_SSL: esp_tls_conn_read error, errno=Success

i tried changing the mbedtls max message content length to 4096 
but ended up with same error 
any help would be appreciated

Reagrds
Siva

mahavirj, 2020-04-24:
@lakshmisivareddy You may check on value of `CONFIG_MBEDTLS_SSL_IN_CONTENT_LEN` https://github.com/espressif/esp-idf/blob/master/components/mbedtls/Kconfig#L77, it should match to default value, i.e. 16384. If you still face please provide all information per issue template (includes ESP-IDF version, application code, sdkconfig etc.).

lakshmisivareddy, 2020-04-28:
solved the issues thanks

closed 2020-04-28 by lakshmisivareddy