espressif/esp-idf #5176: ESP32 BLE Central using NimBLE getting crashed(reset) after few seconds of communication.

promptps, 2020-04-24:
## Environment

- Development Kit: ESP32 DOIT (ESP32 WROOM 4MB)
- Kit version: DevKit: v1
- Module or chip used: ESP32-WROOM-32
- IDF version: v4.2-dev-532-g6f7be93
- Build System: CMake | idf.py
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Linux
- Using an IDE?: Yes ( Eclipse IDE [Version: 2019-06 (4.12.0)] with ESP-IDF Eclipse plugin v1.0.1 )
- Power Supply: USB

## Problem Description
ESP32 Central BLE(NimBLE) module crashes after few seconds of communication with peripheral.

//Detailed problem description goes here.
ESP32 here is configured as BLE central device and it is using NimBLE stack for Bluetooth operation. 
Central BLE here scans for required BLE, connects to peripheral BLE and Discovers  all the required services for application to continue communication.
Central also initiates security here. (but issue also occurs without using security).
Once the encryption process is completed. The central BLE starts communication with Read and Write request to peripheral characteristics. It works fine for few seconds but then devices resets giving error.
Please check the debug logs for error description.
Please update me if there is any error in utilising or configuring the BLE stack as I am new to Bluetooth.

### Expected Behavior
After communication gets started, the device must not reset. It should not break connection until requested form either side. 

### Actual Behavior
The central BLE starts communication with Read and Write request to peripheral characteristics. It works fine for few seconds but then devices resets giving error.
This reset occurrence happens at random interval. maximum time for which the device works is approximately 5mins.

### Steps to reproduce
Here, unfortunately I don't have another BLE module available to configure it as peripheral device.
So, I am using nRF Connect app configured as BLE peripheral.
1. I have configured nRF Connect-->"advertiser" adding 128bit thingy configuration service uuid as advertising data and checked on Connectable.

2. In nRF Connect-->"Configure Gatt Server" option I have created my service list, which includes only "Thingy Configuration service" and I've added three characteristics namely "Thingy Air Quality", "Thingy Button" and "Thingy Light Intensity". Provided each characteristic a random text data.

3. Start the advertiser from nRF Connect and load the configured services. 

4. Then program the esp32 module with the attached application code. The device will reset after few seconds of communication.

Please check the debug logs for error desciption.
[app_ble_central.zip](https://github.com/espressif/esp-idf/files/4527194/app_ble_central.zip)

## Debug Logs
```

GATT procedure initiated: read; att_handle=46
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400fa1d6  PS      : 0x00060530  A0      : 0x800f81c2  A1      : 0x3ffc8770  
0x400fa1d6: ble_gattc_rx_read_rsp at /home/prompt/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:4612

A2      : 0x3ffc15fc  A3      : 0x00000000  A4      : 0x3ffc1460  A5      : 0x0000001f  
A6      : 0x00000001  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffc8740  
A10     : 0x3ffc15fc  A11     : 0x3f409628  A12     : 0x00000003  A13     : 0x3ffc8770  
A14     : 0x310b0004  A15     : 0x310b0004  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  

ELF file SHA256: 68504a50eda10054cb568d1ccdc02168b844242248da84daff09b8b01bd1fe86

Backtrace: 0x400fa1d3:0x3ffc8770 0x400f81bf:0x3ffc87a0 0x400f295d:0x3ffc87d0 0x400f3e92:0x3ffc8800 0x400f2489:0x3ffc8830 0x400f249b:0x3ffc8850 0x400fc4b2:0x3ffc8870 0x400dc1d6:0x3ffc8890 0x40092725:0x3ffc88b0
0x400fa1d3: ble_gattc_rx_read_rsp at /home/prompt/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:4611

0x400f81bf: ble_att_clt_rx_read at /home/prompt/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_clt.c:505

0x400f295d: ble_att_rx at /home/prompt/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att.c:503

0x400f3e92: ble_hs_hci_evt_acl_process at /home/prompt/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:946

0x400f2489: ble_hs_process_rx_data_queue at /home/prompt/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:238

0x400f249b: ble_hs_event_rx_data at /home/prompt/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:539

0x400fc4b2: ble_npl_event_run at /home/prompt/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:121
 (inlined by) nimble_port_run at /home/prompt/esp/esp-idf/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:81

0x400dc1d6: blecent_host_task at /home/prompt/esp/esp32_workspace/app_ble_central/main/ble_main.c:800

0x40092725: vPortTaskWrapper at /home/prompt/esp/esp-idf/components/freertos/xtensa/port.c:143
```
The file list in logs aren't visible in Eclipse terminal. So, to capture it, I have used "make monitor" command.

## Other items if possible

[blecent_elf.zip](https://github.com/espressif/esp-idf/files/4527211/blecent_elf.zip)

dhrishi, 2020-04-24:
@promptps I quickly browsed through your code and what I could see was that you are sending read and write requests back to back after every 1 sec (in the timer callback). This behavior is invalid as per Core 4.2, Vol 3, Part F, Section 3.3.2 which says:

> Once a client sends a request to a server, that client shall send no other request to the same server until a response PDU has been received.

One way for this is, your functions `read_ble` and `write_ble` can return only when a response to corresponding read and write request is received (in the callback) respectively

Please refer https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/blecent for more details wherein the next GATT request is sent only after the response of the previous one is received.

promptps, 2020-04-25:
hello @dhrishi,

Thanks for the quick response. 

As per your suggestion, I have made changes for waiting till the read / write event completes. 
And this solution works. 
Now Central BLE doesn't reset at random time.

closed 2020-04-25 by promptps