espressif/esp-idf #5177: esp-tls-mbedtls: mbedtls_x509_crt_parse returned -0x2180

umer-ilyas, 2020-04-24:
## Environment

- Development Kit:      custom
- Kit version  
- Module or chip used:  ESP32-WROOM-32D
- IDF version v4.1-beta1-112-g4fe18a925
- Build System:         idf.py
- Compiler version   xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Windows
- (Windows only) PowerShell
- Using an IDE?: 
- Power Supply:         Battery

## Problem Description

i am using simple_ota example to update firmware, my firmware file is hosted on microsoft blob storage, but during init this error is being returned `esp-tls-mbedtls: mbedtls_x509_crt_parse returned -0x2180 `   MBEDTLS_ERR_X509_INVALID_FORMAT

this function is returning parse error in file `x509.crt.c`
```
    if( ( ret = mbedtls_asn1_get_tag( &p, end, &len,
            MBEDTLS_ASN1_CONSTRUCTED | MBEDTLS_ASN1_SEQUENCE ) ) != 0 )
    {
        mbedtls_x509_crt_free( crt );
        return( MBEDTLS_ERR_X509_INVALID_FORMAT );
    }
```
### Expected Behavior
certificate should parse successfully.
 
### Actual Behavior
error in certificate parsing.
 
### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
                esp_http_client_config_t config = {
                    .cert_pem = (char *)server_cert_pem_start,
                    .transport_type = HTTP_TRANSPORT_OVER_SSL,
                    .event_handler = _http_event_handler,
                    .skip_cert_common_name_check = true,
                    .buffer_size_tx = 1 * 1024,
                    .buffer_size = 1 * 1024,
                    .url = fimeareurl_local,
                    .use_global_ca_store = false,
                };
```
server_cert_pem_start is null terminated and also attached below

```
-----BEGIN CERTIFICATE-----
MIIFtDCCBJygAwIBAgIQDywQyVsGwJN/uNRJ+D6FaTANBgkqhkiG9w0BAQsFADBa
MQswCQYDVQQGEwJJRTESMBAGA1UEChMJQmFsdGltb3JlMRMwEQYDVQQLEwpDeWJl
clRydXN0MSIwIAYDVQQDExlCYWx0aW1vcmUgQ3liZXJUcnVzdCBSb290MB4XDTE2
MDUyMDEyNTE1N1oXDTI0MDUyMDEyNTE1N1owgYsxCzAJBgNVBAYTAlVTMRMwEQYD
VQQIEwpXYXNoaW5ndG9uMRAwDgYDVQQHEwdSZWRtb25kMR4wHAYDVQQKExVNaWNy
b3NvZnQgQ29ycG9yYXRpb24xFTATBgNVBAsTDE1pY3Jvc29mdCBJVDEeMBwGA1UE
AxMVTWljcm9zb2Z0IElUIFRMUyBDQSAyMIICIjANBgkqhkiG9w0BAQEFAAOCAg8A
MIICCgKCAgEAnqoVwRuhY1/mURjFFrsR3AtNm5EKukBJK9zWBgvFd1ksNEJFC06o
yRbwKPMflpW/HtOfzIeBliGk57MwZq18bgASr70sPUWuoD917HUgBfxBYoF8zA7Z
Ie5zAHODFboJL7Fg/apgbQs/GiZZNCi0QkQUWzw0nTUmVSNQ0mz6pCu95Dv1WMsL
GyPGfdN9zD3Q/QEDyJ695QgjRIxYA1DUE+54ti2k6r0ycKFQYkyWwZ25HD1h2kYt
3ovW85vF6y7tjTqUEcLbgKUCB81/955hdLLsbFd6f9o2PkU8xuOc3U+bUedvv6Sb
tvGjBEZeFyH8/CaQhzlsKMH0+OPOFv/bMqcLarPw1V1sOV1bl4W9vi2278niblzI
bEHt7nN888p4KNIwqCcXaGhbtS4tjn3NKI6v1d2XRyxIvCJDjgoZ09zF39Pyoe92
sSRikZh7xns4tQEQ8BCs4o5NBSx8UxEsgyzNSskWGEWqsIjt+7+A1skDDZv6k2o8
VCHNbTLFKS7d72wMI4ErpzVsBIicxaG2ezuMBBuqThxIiJ+G9zfoP9lxim/9rvJA
xbh3nujA1VJfkOYTJIojEAYCxR3QjEoGdapJmBle97AfqEBnwoJsu2wav8h9v+po
DL4h6dRzRUxY1DHypcFlXGoHu/REQgFLq2IN30/AhQLN90Pj9TT2RQECAwEAAaOC
AUIwggE+MB0GA1UdDgQWBBSRnjtEbD1XnEJ3KjTXT9HMSpcs2jAfBgNVHSMEGDAW
gBTlnVkwgkdYzKz6CFQ2hns6tQRN8DASBgNVHRMBAf8ECDAGAQH/AgEAMA4GA1Ud
DwEB/wQEAwIBhjAnBgNVHSUEIDAeBggrBgEFBQcDAQYIKwYBBQUHAwIGCCsGAQUF
BwMJMDQGCCsGAQUFBwEBBCgwJjAkBggrBgEFBQcwAYYYaHR0cDovL29jc3AuZGln
aWNlcnQuY29tMDoGA1UdHwQzMDEwL6AtoCuGKWh0dHA6Ly9jcmwzLmRpZ2ljZXJ0
LmNvbS9PbW5pcm9vdDIwMjUuY3JsMD0GA1UdIAQ2MDQwMgYEVR0gADAqMCgGCCsG
AQUFBwIBFhxodHRwczovL3d3dy5kaWdpY2VydC5jb20vQ1BTMA0GCSqGSIb3DQEB
CwUAA4IBAQBsf+pqb89rW8E0rP/cDuB9ixMX4C9OWQ7EA7n0BSllR64ZmuhU9mTV
2L0G4HEiGXvOmt15i99wJ0ho2/dvMxm1ZeufkAfMuEc5fQ9RE5ENgNR2UCuFB2Bt
bVmaKUAWxscN4GpXS4AJv+/HS0VXs5Su19J0DA8Bg+lo8ekCl4dq2G1m1WsCvFBI
oLIjd4neCLlGoxT2jA43lj2JpQ/SMkLkLy9DXj/JHdsqJDR5ogcij4VIX8V+bVD0
NCw7kQa6Ulq9Zo0jDEq1at4zSeH4mV2PMM3LwIXBA2xo5sda1cnUWJo3Pq4uMgcL
e0t+fCut38NMkTl8F0arflspaqUVVUov
-----END CERTIFICATE-----
```

ESP-Marius, 2020-04-26:
The certificate you posted here seems to work fine when I tried it. Could you upload the .pem file you are using?

umer-ilyas, 2020-04-28:
[microsoft_it_tls.zip](https://github.com/espressif/esp-idf/files/4544325/microsoft_it_tls.zip)

ESP-Marius, 2020-04-28:
That one also seems to parse just fine when I try it using the simple_ota_example. Are you doing anything else in your code that could be modifying the certificate?

ESP-Marius, 2020-05-19:
@umer-ilyas Hi, did you manage to resolve this issue? If yes, the we'll close this issue.

closed 2020-05-19 by umer-ilyas

DazaiZero, 2021-07-27:
Hello Please try removing space in the .pem file. I was also facing the same issue. it was solved by removing space.

pedros89, 2021-08-03:
Hello. I had the same problem. I had a problem in the formatting of the SSL certificate.
Formatted as below it worked for me.

`#include "esp_http_client.h"   `
```
esp_http_client_config_t config;
  memset(&config, 0, sizeof(esp_http_client_config_t));
  config.url = URL_SPIFFS;
  config.cert_pem = UPGRADE_SERVER_CERT;
```
 

```
#define UPGRADE_SERVER_CERT "-----BEGIN CERTIFICATE-----\n\
MIIDSjCCAjKgAwIBAgIQRK+wgNajJ7qJMDmGLvhAazANBgkqhkiG9w0BAQUFADA/\n\
MSQwIgYDVQQKExtEaWdpdGFsIFNpZ25hdHVyZSBUcnVzdCBDby4xFzAVBgNVBAMT\n\
DkRTVCBSb290IENcIFgzMB4XDTAwMDkzMDIxMTIxOVoXDTIxMDkzMDE0MDExNVow\n\
PzEkMCIGA1UEChMbRGlnaXRhbCBTaWduYXR1cmUgVHJ1c3QgQ28uMRcwFQYDVQQD\n\
Ew5EU1QgUm9vdCBDQSBYMzCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoCggEB\n\
AN+v6ZdQCINXtMxPZfaQguzH0yxrMMpb7NnDfcdAwRgUi+DoM3ZJKuM/IUmTrE4O\n\
rz5Iy2Xu/NMhD2XSKtkyj4zl93ewEnu1lcCJo6m67XMuegwGMoOifooUMM0RoOEq\n\
OLl5CjH9UL2AZd+3UWODyOKIYepLYYHsUmu5ouJLGiifSKOeDNoJjj4XLh7dIN9b\n\
xiqKqy69cK3FCxolkHRyxXtqqzTWMIn/5WgTe1QLyNau7Fqckh49ZLOMxt+/yUFw\n\
7BZy1SbsOFU5Q9D8/RhcQPGX69Wam40dutolucbY38EVAjqr2m7xPi71XAicPNaD\n\
aeQQmxkqtilX4+U9m5/wAl0CAwEAAaNCMEAwDwYDVR0TAQH/BAUwAwEB/zAOBgNV\n\
HQ8BAf8EBAMCAQYwHQYDVR0OBBYEFMSnsaR7LHH62+FLkHX/xBVghYkQMA0GCSqG\n\
SIb3DQEBBQUAA4IBAQCjGiybFwBcqR7uKGY4Or+Dxz9LwwmglSBd49lZRNI+DT69\n\
ikugdB/OEIKcdBodfpga3csTS7MgROSR6cz8faXbauX+5v3gTt23ADq1cEmv8uXr\n\
AvHRAosZy5Q6XkjEGB5YGV8eAlrwDPGxrancWYaLbumR9YbK+rlmM6pZW87ipxZz\n\
R8srzJmwN0jP41ZL9c8PDHIyh8bwRLtTcm1D9SZImlJnt1ir/md2cXjbDaJWFBM5\n\
JDGFoqgCWjBH4d1QB7wCCZAA62RjYJsWvIjJEubSfZGL+T0yjWW06XyxV3bqxbYo\n\
Ob8VZRzI9neWagqNdwvYkQsEjgfbKbYK7p2CNTUQ\n\
-----END CERTIFICATE-----\n"
```

CraigeHales, 2021-08-12:
Also, MUST be nul terminated. If you embed the files, you MUST use EMBED_TXTFILES, not EMBED_FILES in CMakeLists.txt
`idf_component_register(...
                    INCLUDE_DIRS "."
                    EMBED_FILES
                        "upload_script.html"
                    EMBED_TXTFILES
                        "certs/cacert.pem"
                        "certs/prvtkey.pem"
                    )
...`
TXTFILES adds a trailing nul. Even though the interface uses a pointer/length to supply the data in config.cacert_pem/config.cacert_len, something parses off the end without the trailing nul. That's one source of the -0x2180 error.

I wound up using the EMBED_FILES by copy/paste examples together...I figured it out after printing the length from a working example and a failing example using the same certificates.

chill117, 2022-05-30:
I had the same problem as described by others in this issue, which I did finally manage to solve. I am using the ESP WebSocket Client but the same solution can be applied for anyone using the HTTP Client. From the [espressif docs](https://espressif.github.io/esp-protocols/esp_websocket_client/index.html#_CPPv429esp_websocket_client_config_t):
> `const char *cert_pem`
> 
> **Pointer to certificate data in PEM ... format** for server verify (with SSL), default is NULL, not required to verify the server. PEM-format must have a terminating NULL-character. DER-format requires the length to be passed in cert_len.

I emphasized the important part - that the PEM certificate must be a **pointer** to the certificate data (ie. string) which must remain un-garbage-collected until the WebSocket or HTTP client has managed to read from it. So in my application, I saved the PEM certificate data to a `std::string` variable which never gets garbage-collected.