espressif/esp-idf #5178: WPS saved connection

Wtpaul, 2020-04-24:
When using the WPS feature.. where is the wifi connection ssid and password saved to?

the wps feature connects to my wifi... but on restart of the device the connection is lost!

sagb2015, 2020-04-30:
Hi @Wtpaul - WiFi driver stored the credential of last AP to which the device connects in NVS. Just calling esp_wifi_connect should be sufficient for you. Can you post your example here and also mention the IDF version?

Wtpaul, 2020-04-30:
Hi Sagb2015
Just found the issue...
Was running the wifi provisioning manager for BLE.
When the WPS button was pressed, i called wifi_prov_mgr_stop_provisioning() and then started the WPS functionality.... but even tho WPS was successful.. it would not save the credentials over a reset/restart.
The issue was.. WIFI manager moves over to save credentials in RAM... and even on calling wifi_prov_mgr_stop_provisioning() this remained... so the WPS credentials were being saved only on RAM.

Possible change request.. call esp_wifi_set_storage(WIFI_STORAGE_FLASH); at the end of wifi_prov_mgr_stop_provisioning()

Hope this all makes sense..
And hopefully helps someone else along the way!

closed 2020-04-30 by Wtpaul

sagb2015, 2020-05-04:
@Wtpaul  Noted and forwarded the request to corresponding developers internally.