espressif/esp-idf #5180: Unable to write to BLE server(external device) from ESP32(BLE client) using example "gattc_gatts_coex"

ROYMURLI, 2020-04-24:
IDF version- v4.1-dev-1086-g93a8603c5
OS- windows

I am using gattc_gatts_coex example to communicate with a "blood pressure monitor device" which is a ble server. 
How does this BP monitor work - It works as both manually and by giving values over BLE. For example to start the BP monitor BLE client shou;d send byte array of {00,02,03,64,69} also in hex {0x00,0x02,0x03,0x40,0x45} and after receiving the value the device start NOTIFY the sensed value of Blood Pressure to the client. I tried with nrF BLE scanner app and it work perfectly fine. 

Now, I want to implement it wth ESP32 by using gattc_gatts_coex example, The connection is successful but when I tried to write the above mentioned byte array it doesn,t turned on to record the value. I tried to turn it on manually, which can be done by press button and  I am able to receive data in ESP_GATTC_NOTIFY_EVT. I tried different alteration to write to the BLE server (BP monitor device) through ESP32 but none of them are working.

I think there is some UUID issue because the write write_uuid is different from service_uuid or I am doing some mistake in writing the byte array. I am attaching my code, BLE scanner screenshot and debug logs here. Kindly review. 

[image]

[image]

[code.zip](https://github.com/espressif/esp-idf/files/4531239/code.zip)

Alvin1Zhang, 2020-04-26:
@ROYMURLI Thanks for your contribution. @GYC-Espressif

ROYMURLI, 2020-05-08:
Hi, Any progress in this issue.

ROYMURLI, 2020-06-03:
Hi, Any suggestions. @Alvin1Zhang  @GYC-Espressif

WCCWCC, 2020-06-18:
Hi, @ROYMURLI 

Use the esp_ble_gattc_write_char_descr API to set the value of Client Characteristic Configuration to notify. Use as follows:

```c
                    if (count > 0 && descr_elem_result[0].uuid.len == ESP_UUID_LEN_16 && descr_elem_result[0].uuid.uuid.uuid16 == ESP_GATT_UUID_CHAR_CLIENT_CONFIG) {
                        ret_status = esp_ble_gattc_write_char_descr( gattc_if,
                                                                     gattc_profile_tab[GATTC_PROFILE_C_APP_ID].conn_id,
                                                                     descr_elem_result[0].handle,
                                                                     sizeof(notify_en),
                                                                     (uint8_t *)&notify_en,
                                                                     ESP_GATT_WRITE_TYPE_RSP,
                                                                     ESP_GATT_AUTH_REQ_NONE);
                        
                    }
```

[image]

Alvin1Zhang, 2020-08-19:
@ROYMURLI Would you please help share if any updates for this issue? Thanks.

ROYMURLI, 2020-08-20:
I am not working on this issue. I will start again and will check if it is working or not. Till then I am closing this issue. Thanks for your help.

closed 2020-08-20 by ROYMURLI