espressif/esp-idf #5181: install.sh failing to install openocd

KaeLL, 2020-04-24:
## Environment

- IDF version: v4.1-beta1
- Operating System: Linux
- Using an IDE?: No

## Problem Description
Just like [here](https://github.com/espressif/esp-idf/issues/4095), I'm getting this when running `install.sh`
```
WARNING: tool openocd-esp32 version v0.10.0-esp32-20191114 is installed, but the tool failed to run
ERROR: tool openocd-esp32 has no installed versions. Please run 'install.sh' to install it.
```

### Expected Behavior
Successful install

### Steps to reproduce
Have a GoCD agent running on AlpineLinux execute a script that clones the said version of esp-idf
Run install.sh
Get error.

igrr, 2020-04-26:
Could you please check if libusb is installed?

KaeLL, 2020-04-30:
It isn't, and unfortunately I can't fix that right now, so I'm just ignoring the error, since the compiler still works fine. Does this serve as an argument for a selective install of the tools? We've talked about something like this, but in the scope of esp32 models, like the S2 and the regular. What about the suite of tools that comes within a package for each of them? I'd like to not need to install openOCD, for example.

igrr, 2020-04-30:
We don't have selective tool support in install.sh yet, unfortunately.

For now you may be able to work around the issue by calling the underlying `idf_tools.py` script to install specific tools, and then to export the paths, ignoring the error:

```
$IDF_PATH/tools/idf_tools.py install xtensa-esp32-elf
eval $(IDF_TOOLS_PATH=/tmp/tools tools/idf_tools.py export 2>/dev/null || true)
```

Another option which gives you full control of the installed tools is to download specific archives listed in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-tools.html#list-of-idf-tools, unpack them, and add to PATH. Then install Python packages listed in `requirements.txt`. This is the same process that was used in IDF 3.x and earlier, and it is still available if needed.

closed 2020-05-12 by KaeLL