espressif/esp-idf #5184: PPPoS sometimes looping when trying to connect network

Kunislaw, 2020-04-25:
Hello, I'm using the latest example from examples / pppos_client. The program can loop while getting the address, I checked older versions and it is the same. I use a roaming sim card

Modem: SIM800L

output in terminal: 
`E (32333) event_send: invalid ip event id 5
I (32333) pppos_example: Modem Connect to PPP Server
I (32343) pppos_example: ~~~~~~~~~~~~~~
I (32343) pppos_example: IP          : 10.235.33.226
I (32353) pppos_example: Netmask     : 255.255.255.255
I (32353) pppos_example: Gateway     : 192.168.254.254
I (32363) pppos_example: Name Server1: 10.2.9.7
I (32363) pppos_example: Name Server2: 10.2.9.8
I (32373) pppos_example: ~~~~~~~~~~~~~~
I (32373) pppos_example: GOT ip event!!!
E (32383) event_send: invalid ip event id 5
I (32383) pppos_example: Modem Connect to PPP Server
I (32393) pppos_example: ~~~~~~~~~~~~~~
I (32393) pppos_example: IP          : 10.235.33.226
I (32403) pppos_example: Netmask     : 255.255.255.255
I (32403) pppos_example: Gateway     : 192.168.254.254
I (32413) pppos_example: Name Server1: 10.2.9.7
I (32423) pppos_example: Name Server2: 10.2.9.8
I (32423) pppos_example: ~~~~~~~~~~~~~~
I (32433) pppos_example: GOT ip event!!!
E (32433) event_send: invalid ip event id 5
I (32433) pppos_example: Modem Connect to PPP Server
I (32443) pppos_example: ~~~~~~~~~~~~~~
I (32443) pppos_example: IP          : 10.235.33.226
I (32453) pppos_example: Netmask     : 255.255.255.255
I (32463) pppos_example: Gateway     : 192.168.254.254
I (32463) pppos_example: Name Server1: 10.2.9.7
I (32473) pppos_example: Name Server2: 10.2.9.8
I (32473) pppos_example: ~~~~~~~~~~~~~~
I (32483) pppos_example: GOT ip event!!!
E (32483) event_send: invalid ip event id 5
I (32493) pppos_example: Modem Connect to PPP Server
I (32493) pppos_example: ~~~~~~~~~~~~~~
I (32503) pppos_example: IP          : 10.235.33.226
I (32503) pppos_example: Netmask     : 255.255.255.255
I (32513) pppos_example: Gateway     : 192.168.254.254
I (32513) pppos_example: Name Server1: 10.2.9.7
I (32523) pppos_example: Name Server2: 10.2.9.8
I (32523) pppos_example: ~~~~~~~~~~~~~~
I (32533) pppos_example: GOT ip event!!!
E (32533) event_send: invalid ip event id 5
I (32543) pppos_example: Modem Connect to PPP Server
I (32543) pppos_example: ~~~~~~~~~~~~~~
I (32553) pppos_example: IP          : 10.235.33.226
I (32553) pppos_example: Netmask     : 255.255.255.255
I (32563) pppos_example: Gateway     : 192.168.254.254
I (32573) pppos_example: Name Server1: 10.2.9.7
I (32573) pppos_example: Name Server2: 10.2.9.8
I (32583) pppos_example: ~~~~~~~~~~~~~~
I (32583) pppos_example: GOT ip event!!!
E (32593) event_send: invalid ip event id 5
I (32593) pppos_example: Modem Connect to PPP Server
I (32603) pppos_example: ~~~~~~~~~~~~~~
I (32603) pppos_example: IP          : 10.235.33.226
I (32613) pppos_example: Netmask     : 255.255.255.255
I (32613) pppos_example: Gateway     : 192.168.254.254
I (32623) pppos_example: Name Server1: 10.2.9.7
I (32623) pppos_example: Name Server2: 10.2.9.8
I (32633) pppos_example: ~~~~~~~~~~~~~~
I (32633) pppos_example: GOT ip event!!!
E (32643) event_send: invalid ip event id 5
I (32643) pppos_example: Modem Connect to PPP Server
I (32653) pppos_example: ~~~~~~~~~~~~~~
I (32653) pppos_example: IP          : 10.235.33.226
I (32663) pppos_example: Netmask     : 255.255.255.255
I (32663) pppos_example: Gateway     : 192.168.254.254
I (32673) pppos_example: Name Server1: 10.2.9.7
I (32673) pppos_example: Name Server2: 10.2.9.8
I (32683) pppos_example: ~~~~~~~~~~~~~~
I (32683) pppos_example: GOT ip event!!!
E (32693) event_send: invalid ip event id 5
I (32693) pppos_example: Modem Connect to PPP Server
I (32703) pppos_example: ~~~~~~~~~~~~~~
I (32703) pppos_example: IP          : 10.235.33.226
I (32713) pppos_example: Netmask     : 255.255.255.255
I (32723) pppos_example: Gateway     : 192.168.254.254
I (32723) pppos_example: Name Server1: 10.2.9.7
I (32733) pppos_example: Name Server2: 10.2.9.8
I (32733) pppos_example: ~~~~~~~~~~~~~~
I (32743) pppos_example: GOT ip event!!!
E (32743) event_send: invalid ip event id 5
I (32753) pppos_example: Modem Connect to PPP Server
I (32753) pppos_example: ~~~~~~~~~~~~~~
I (32763) pppos_example: IP          : 10.235.33.226
I (32763) pppos_example: Netmask     : 255.255.255.255
I (32773) pppos_example: Gateway     : 192.168.254.254
I (32773) pppos_example: Name Server1: 10.2.9.7
I (32783) pppos_example: Name Server2: 10.2.9.8
I (32783) pppos_example: ~~~~~~~~~~~~~~
I (32793) pppos_example: GOT ip event!!!
E (32793) event_send: invalid ip event id 5
I (32803) pppos_example: Modem Connect to PPP Server
I (32803) pppos_example: ~~~~~~~~~~~~~~
I (32813) pppos_example: IP          : 10.235.33.226
I (32813) pppos_example: Netmask     : 255.255.255.255
I (32823) pppos_example: Gateway     : 192.168.254.254
I (32823) pppos_example: Name Server1: 10.2.9.7
I (32833) pppos_example: Name Server2: 10.2.9.8
I (32843) pppos_example: ~~~~~~~~~~~~~~
I (32843) pppos_example: GOT ip event!!!
E (32843) event_send: invalid ip event id 5
I (32853) pppos_example: Modem Connect to PPP Server
I (32853) pppos_example: ~~~~~~~~~~~~~~
I (32863) pppos_example: IP          : 10.235.33.226
I (32873) pppos_example: Netmask     : 255.255.255.255
I (32873) pppos_example: Gateway     : 192.168.254.254
I (32883) pppos_example: Name Server1: 10.2.9.7
I (32883) pppos_example: Name Server2: 10.2.9.8
I (32893) pppos_example: ~~~~~~~~~~~~~~
I (32893) pppos_example: GOT ip event!!!
E (32903) event_send: invalid ip event id 5
I (32903) pppos_example: Modem Connect to PPP Server
I (32913) pppos_example: ~~~~~~~~~~~~~~
I (32913) pppos_example: IP          : 10.235.33.226
I (32923) pppos_example: Netmask     : 255.255.255.255
I (32923) pppos_example: Gateway     : 192.168.254.254
I (32933) pppos_example: Name Server1: 10.2.9.7
I (32933) pppos_example: Name Server2: 10.2.9.8
I (32943) pppos_example: ~~~~~~~~~~~~~~
I (32943) pppos_example: GOT ip event!!!
E (32953) event_send: invalid ip event id 5
I (32953) pppos_example: Modem Connect to PPP Server
I (32963) pppos_example: ~~~~~~~~~~~~~~
I (32963) pppos_example: IP          : 10.235.33.226
I (32973) pppos_example: Netmask     : 255.255.255.255
I (32973) pppos_example: Gateway     : 192.168.254.254
I (32983) pppos_example: Name Server1: 10.2.9.7
I (32993) pppos_example: Name Server2: 10.2.9.8
I (32993) pppos_example: ~~~~~~~~~~~~~~
I (33003) pppos_example: GOT ip event!!!
E (33003) event_send: invalid ip event id 5
I (33003) pppos_example: Modem Connect to PPP Server
I (33013) pppos_example: ~~~~~~~~~~~~~~
I (33013) pppos_example: IP          : 10.235.33.226
I (33023) pppos_example: Netmask     : 255.255.255.255
I (33033) pppos_example: Gateway     : 192.168.254.254
I (33033) pppos_example: Name Server1: 10.2.9.7
I (33043) pppos_example: Name Server2: 10.2.9.8
I (33043) pppos_example: ~~~~~~~~~~~~~~
I (33053) pppos_example: GOT ip event!!!
E (33053) event_send: invalid ip event id 5
I (33063) pppos_example: Modem Connect to PPP Server`

Alvin1Zhang, 2020-04-26:
@Kunislaw Thanks for reporting, we will look into. @david-cermak

david-cermak, 2020-04-29:
Hi @Kunislaw 

Could you please post the log with the below options enabled in menuconfig?
* debug level in `CONFIG_LOG_DEFAULT_LEVEL_DEBUG`
* PPP debug options in `CONFIG_LWIP_PPP_DEBUG_ON`

Just to make sure, was this option off -- `CONFIG_LWIP_PPP_ENABLE_IPV6=n` ?

Kunislaw, 2020-04-29:
```

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:9248
ho 0 tail 12 room 4
load:0x40078000,len:14536
ho 0 tail 12 room 4
load:0x40080400,len:4388
entry 0x400806c4
I (32) boot: ESP-IDF v4.2-dev-1206-g741960d5c 2nd stage bootloader
I (32) boot: compile time 19:22:28
D (33) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (40) boot: chip revision: 1
I (44) boot_comm: chip revision: 1, min. bootloader chip revision: 0
D (51) boot.esp32: magic e9
D (54) boot.esp32: segments 04
D (57) boot.esp32: spi_mode 02
D (60) boot.esp32: spi_speed 00
D (63) boot.esp32: spi_size 02
I (66) boot.esp32: SPI Speed      : 40MHz
I (71) boot.esp32: SPI Mode       : DIO
I (76) boot.esp32: SPI Flash Size : 4MB
D (80) boot: Enabling RTCWDT(9000 ms)
I (84) boot: Enabling RNG early entropy source...
D (89) bootloader_flash: mmu set paddr=00000000 count=1 size=c00 src_addr=8000 src_addr_aligned=0
D (98) boot: mapped partition table 0x8000 at 0x3f408000
D (104) flash_parts: partition table verified, 4 entries
I (109) boot: Partition Table:
I (113) boot: ## Label            Usage          Type ST Offset   Length
D (120) boot: load partition table entry 0x3f408000
D (125) boot: type=1 subtype=2
I (128) boot:  0 nvs              WiFi data        01 02 00009000 00006000
D (136) boot: load partition table entry 0x3f408020
D (141) boot: type=1 subtype=1
I (144) boot:  1 phy_init         RF data          01 01 0000f000 00001000
D (151) boot: load partition table entry 0x3f408040
D (156) boot: type=0 subtype=0
I (159) boot:  2 factory          factory app      00 00 00010000 00100000
I (167) boot: End of partition table
D (171) boot: Trying partition index -1 offs 0x10000 size 0x100000
D (177) esp_image: reading image header @ 0x10000
D (182) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
D (189) esp_image: image header: 0xe9 0x06 0x02 0x02 40081588
I (195) boot_comm: chip revision: 1, min. application chip revision: 0
I (202) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1af68 (110440) map
D (211) esp_image: free data page_count 0x00000032
D (216) bootloader_flash: mmu set paddr=00010000 count=2 size=1af68 src_addr=10020 src_addr_aligned=10000
D (268) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)
I (268) esp_image: segment 1: paddr=0x0002af90 vaddr=0x3ffb0000 size=0x023d0 (  9168) load
D (272) esp_image: free data page_count 0x00000032
D (277) bootloader_flash: mmu set paddr=00020000 count=1 size=23d0 src_addr=2af90 src_addr_aligned=20000
D (290) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)
I (293) esp_image: segment 2: paddr=0x0002d368 vaddr=0x40080000 size=0x00404 (  1028) load
D (302) esp_image: free data page_count 0x00000032
D (307) bootloader_flash: mmu set paddr=00020000 count=1 size=404 src_addr=2d368 src_addr_aligned=20000
D (317) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)
I (324) esp_image: segment 3: paddr=0x0002d774 vaddr=0x40080404 size=0x028a4 ( 10404) load
D (332) esp_image: free data page_count 0x00000032
D (337) bootloader_flash: mmu set paddr=00020000 count=2 size=28a4 src_addr=2d774 src_addr_aligned=20000
D (351) bootloader_flash: mmu set block paddr=0x00030000 (was 0xffffffff)
I (354) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x68bb8 (428984) map
D (363) esp_image: free data page_count 0x00000032
D (367) bootloader_flash: mmu set paddr=00030000 count=7 size=68bb8 src_addr=30020 src_addr_aligned=30000
[... 292 lines cut ...]
I (19635) pppos_example: Name Server2: 10.2.9.8
I (19645) pppos_example: ~~~~~~~~~~~~~~
I (19645) pppos_example: GOT ip event!!!
E (19655) event_send: invalid ip event id 5
I (19655) pppos_example: Modem Connect to PPP Server
I (19665) pppos_example: ~~~~~~~~~~~~~~
I (19665) pppos_example: IP          : 10.224.60.38
I (19675) pppos_example: Netmask     : 255.255.255.255
I (19685) pppos_example: Gateway     : 192.168.254.254
I (19685) pppos_example: Name Server1: 10.2.9.7
I (19695) pppos_example: Name Server2: 10.2.9.8
I (19695) pppos_example: ~~~~~~~~~~~~~~
I (19705) pppos_example: GOT ip event!!!
E (19705) event_send: invalid ip event id 5
I (19715) pppos_example: Modem Connect to PPP Server
I (19715) pppos_example: ~~~~~~~~~~~~~~
I (19725) pppos_example: IP          : 10.224.60.38
I (19725) pppos_example: Netmask     : 255.255.255.255
I (19735) pppos_example: Gateway     : 192.168.254.254
I (19735) pppos_example: Name Server1: 10.2.9.7
I (19745) pppos_example: Name Server2: 10.2.9.8

```

[image]

david-cermak, 2020-04-30:
@Kunislaw Thanks for sharing the log and related configuration!
This looks like a bug in the default connection handler. Could you please try to apply this patch and see if that helps fixing the issue?
[0001-esp_netif-fix-default-connect-handler-to-check-dhcpc.patch.txt](https://github.com/espressif/esp-idf/files/4558193/0001-esp_netif-fix-default-connect-handler-to-check-dhcpc.patch.txt)

Thanks!

Kunislaw, 2020-05-01:
@david-cermak 
Now it looks like everything is working, all I get is a dhcpc error
```
I (23714) pppos_example: Modem Connect to PPP Server
local  IP address 10.198.18.159
I (23714) pppos_example: ~~~~~~~~~~~~~~
remote IP address 192.168.254.254
I (23724) pppos_example: IP          : 10.198.18.159
primary   DNS address 10.1.9.7
I (23734) pppos_example: Netmask     : 255.255.255.255
secondary DNS address 10.1.9.8
I (23744) pppos_example: Gateway     : 192.168.254.254
ppp phase changed[1]: phase=10
I (23754) pppos_example: Name Server1: 10.1.9.7
I (23754) pppos_example: Name Server2: 10.1.9.8
I (23764) pppos_example: ~~~~~~~~~~~~~~
I (23764) pppos_example: GOT ip event!!!
E (23774) esp_netif_handlers: esp_netif_action_connected 50 connected action dhcpc failed ret=0x102
I (23784) pppos_example: MQTT other event id: 7
pppos_netif_output[1]: proto=0x21, len = 61
pppos_netif_output[1]: proto=0x21, len = 61
pppos_input[1]: got 49 bytes
pppos_input[1]: got 337 bytes
ppp_input[1]: ip in pbuf len=380
pppos_netif_output[1]: proto=0x21, len = 44
pppos_input[1]: got 260 bytes
pppos_input[1]: got 127 bytes
ppp_input[1]: ip in pbuf len=380
pppos_netif_output[1]: proto=0x21, len = 56
pppos_input[1]: got 49 bytes
ppp_input[1]: ip in pbuf len=44
I (24444) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
pppos_netif_output[1]: proto=0x21, len = 40
pppos_netif_output[1]: proto=0x21, len = 61
pppos_input[1]: got 95 bytes
ppp_input[1]: ip in pbuf len=40
ppp_input[1]: ip in pbuf len=44
I (24844) pppos_example: MQTT_EVENT_CONNECTED
pppos_netif_output[1]: proto=0x21, len = 63
I (24844) pppos_example: sent subscribe successful, msg_id=35384
pppos_input[1]: got 51 bytes
ppp_input[1]: ip in pbuf len=45
I (25194) pppos_example: MQTT_EVENT_SUBSCRIBED, msg_id=35384
pppos_netif_output[1]: proto=0x21, len = 71
I (25194) pppos_example: sent publish successful, msg_id=0
pppos_input[1]: got 76 bytes
ppp_input[1]: ip in pbuf len=71
I (25494) pppos_example: MQTT_EVENT_DATA
TOPIC=/topic/esp-pppos
DATA=esp32-pppos
pppos_netif_output[1]: proto=0x21, len = 40

```

david-cermak, 2020-05-04:
@Kunislaw Thanks for testing the patch and sharing the result. This is a bug in IDF esp-netif component. Please keep the issue open, will be closed upon merging and publishing a fix.

PS: You may use this fix and ignore the error message before it gets fixed properly (the `dhcpc failed ret=0x102` error won't appear in the final log, just wanted to see what was happening at your end)

closed 2020-05-15 by espressif-bot