espressif/esp-idf #5185: Secure boot with flash encryption and OTA

Zsozsesz1997, 2020-04-25:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4] Sorry, I don't know.
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.2-dev-414-g132cc67
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux (Ubuntu)
- Using an IDE?: No
- Power Supply:         USB

## Problem Description
Hy! I want to make a "system", where secure boot is enabled with flash encryption and OTA updates. I've already done the secure boot part, and it worked perfectly (I made it with "hello world" example). After that I made the "simple_ota" example with secure boot enabled. I loaded the previous mentioned "hello-world" example's binary over-the-air, so it worked as well. After that I wanted to enable flash encryption. I generated a key with `espsecure.py generate_flash_encryption_key my_flash_encryption_key.bin` And I burned it to the device, I burned the flash_enc bit too. But this time I got "assertion" failed message. The output message what I got:
[image]

[image]

Now I will write down every detail and command what I ran. If you feel easier to just write down how to do the 3 feature at the same time than go ahead! :)

### Expected Behavior
The device boots the simple_ota_app (with flash encryption), it prints out that flash encryption is working, and the other things (see flash_encryption example program), after that is connects to my wifi, and starts the over-the-air update to "hello-world.bin".
### Actual Behavior
It says assertion "partition" failed, than abort() was called.
### Steps to reproduce

First things first:
I copied the main of the "flash encryption" example, and added to the "simple_ota" example's main. The reason of this was to "see" if flash encrytion is working or not. Down you will find my "simple_ota_example.c" file.

I want to "stick to" simple_ota_example, because without flash encryption it was working well, I think it's easier to add flash encryption to a working OTA, than reverse.

As I said above I already made the secure boot, with my own key, and I burned this key onto the device. I did this with flash encryption key too. After we cleared the basics, here are my commands in order:

1. `idf.py menuconfig`
Here are the most important things from menuconfig:
[image]
[image]
[image]
[image]
I use reflashable secure boot and flash encryption with development mode.

2. `make clean && make erase_flash && make -j8 && make bootloader`

3. `make partition_table`

Here is my partition table:
[image]
I've done a lot of searching here (I saw a lot of issues), and after all I found that I have to encrypt all of my binaries which I want to flash onto the device (bootloader.bin, partitions_two_ota.bin,  ota_data_initial.bin, simple_ota.bin, bootloader-reflash-digest.bin). I broadly understand why, but the docs said that the device can encrypt everything, so I don'k know why it isn't working with plaintext binaries. (if someone can make this clear to me, what can the device encrypt, what can't, and why I would be very pleased.) So as I said before, I encrypted the binaries one-by-one and flashed them one-by-one onto the device. 

My commands were:
4. Encrypt the bootloader.bin and flash onto the device:

`espsecure.py encrypt_flash_data --keyfile /home/zsolt/esp/esp-idf/examples/system/ota/simple_ota_example/flash_enc_key/my_flash_encryption_key.bin --address 0x1000 -o BinEncOut/bootloader_En.bin /home/zsolt/esp/esp-idf/examples/system/ota/simple_ota_example/build/bootloader/bootloader.bin`

`python /home/zsolt/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 /home/zsolt/esp/esp-idf/examples/system/ota/simple_ota_example/BinEncOut/bootloader_En.bin`

5. Encrypt the partitions_two_ota.bin and flash onto the device:

`espsecure.py encrypt_flash_data --keyfile /home/zsolt/esp/esp-idf/examples/system/ota/simple_ota_example/flash_enc_key/my_flash_encryption_key.bin --address 0x11000 -o BinEncOut/partitions_two_ota_enc.bin /home/zsolt/esp/esp-idf/examples/system/ota/simple_ota_example/build/partitions_two_ota.bin`

`python /home/zsolt/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x11000 /home/zsolt/esp/esp-idf/examples/system/ota/simple_ota_example/BinEncOut/partitions_two_ota_enc.bin`

6. Encrypt the ota_data_initial.bin and flash onto the device:

`espsecure.py encrypt_flash_data --keyfile /home/zsolt/esp/esp-idf/examples/system/ota/simple_ota_example/flash_enc_key/my_flash_encryption_key.bin --address 0x16000 -o BinEncOut/ota_data_initial_enc.bin /home/zsolt/esp/esp-idf/examples/system/ota/simple_ota_example/build/ota_data_initial.bin`

`python /home/zsolt/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x16000 /home/zsolt/esp/esp-idf/examples/system/ota/simple_ota_example/BinEncOut/ota_data_initial_enc.bin`

7. Encrypt the simple_ota.bin and flash onto the device:

[... 77 lines cut ...]
0x400d333f: __assert_func at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

0x400d2e51: example_read_write_flash at /home/zsolt/esp/esp-idf/examples/system/ota/simple_ota_example/main/simple_ota_example.c:146 (discriminator 1)

0x400d2f2f: app_main at /home/zsolt/esp/esp-idf/examples/system/ota/simple_ota_example/main/simple_ota_example.c:180

0x400d145e: main_task at /home/zsolt/esp/esp-idf/components/esp32/cpu_start.c:563

0x40093f41: vPortTaskWrapper at /home/zsolt/esp/esp-idf/components/freertos/xtensa/port.c:143

```

## Other items if possible

- [+] sdkconfig file (attach the sdkconfig file from your project folder)
https://gist.github.com/Zsozsesz1997/907e15ee9fe692003fff39ac893cf59f
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

My espefuse summary:
[image]

I really appreciate every help, because I'm hopeless :( 

ps.: If you have a working method for secure boot with flash encryption and OTA than please share it with me.

Alvin1Zhang, 2020-04-26:
@Zsozsesz1997 Thanks for the detailed report, we will look into.

igrr, 2020-04-26:
> assertion "partition" failed: file "/home/zsolt/esp/esp-idf/examples/system/ota/simple_ota_example/main/simple_ota_example.c", line 146, function: example_read_write_flash

(Looking at the source you have attached around the reported line number)

The failure is when the example tries to operate on `storage` partition which is included in the flash encryption example. But there is no storage partition in partitions_two_ota.bin. Since this part of the example is not needed for OTA, perhaps you don't need to call `example_read_write_flash`?

Alvin1Zhang, 2020-05-13:
@Zsozsesz1997 Thanks for reporting. Would you please help share if any updates or details for this issue? Thanks.

Zsozsesz1997, 2020-05-20:
> (Looking at the source you have attached around the reported line number)
>
> The failure is when the example tries to operate on `storage` partition which is included in the flash encryption example. [...]

That did the trick, thank you!

closed 2020-05-20 by Zsozsesz1997