espressif/esp-idf #5187: Documentation bug - rtc_gpio_set_direction_in_sleep()

stevewms, 2020-04-25:
The "read the docs" page at 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gpio.html
lists a function: 
esp_err_t rtc_gpio_set_direction_in_sleep(gpio_num_t gpio_num, rtc_gpio_mode_t mode)

This function does not seem to appear in any of the esp-idf component files, either in *.h or *.c.  At least, I can't find rtc_gpio_set_direction_in_sleep(), having "grep"ed for it, and having easily found rtc_gpio_set_direction().

This raises 2 questions:
1. Is rtc_gpio_set_direction_in_sleep() a missing function in esp-idf? or...
2. Does rtc_gpio_set_direction() work in deep sleep, making rtc_gpio_set_direction_in_sleep() redundant?

costaud, 2020-04-26:
https://github.com/espressif/esp-idf/blob/741960d5ce0394e8ecb99deef8a5ebf579cb45a2/components/driver/rtc_io.c#L109

stevewms, 2020-04-26:
My mistake.  I am using the most recent "stable" release, v4.0.  The API, 
rtc_gpio_set_direction_in_sleep() is neither documented nor instantiated in v4.0.  So when I found it in the docs after a google search for "esp-idf rtc gpio", and the write up for v4.1 came up, which described rtc_gpio_set_direction_in_sleep(),  yes, I was confused not to be able to find it. in either a .h or a .c file in v4.0 on my machine.  There was nothing obvious in the doc that came up that indicated that the doc applied to v4.1.
Perhaps the documentation bug is that in the various documents you make available, nothing appears to indicate which version of esp-idf they apply,  Some docs have this information.  For instance, "Getting Started" docs for v3.3 say there's a later version available, and that's very helpful.   Perhaps a date or a idf version number at the head of your many helpful docs would be nice to disambiguate these situations.

closed 2020-05-26 by costaud