espressif/esp-idf #5190: Error while building aws_iot (thing shadow) example in release/v4.2 SDK.

harshithakre38, 2020-04-26:
**Is your feature request related to a problem? Please describe.**

A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**

A clear and concise description of what you want to happen.

**Describe alternatives you've considered**

A clear and concise description of any alternative solutions or features you've considered.

Please give as many details as you can. Include suggestions for useful APIs or interfaces if relevant.

**Additional context**

Add any other context or screenshots about the feature request here.

mahavirj, 2020-04-27:
Please try examples from below repository pointer:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/libraries-and-frameworks/cloud-
frameworks.html

https://github.com/espressif/esp-aws-iot

mahavirj, 2020-04-29:
@harshithakre38 I will mark this as resolved, please feel to re-open if you still find any issues with examples in above repository pointer.

closed 2020-04-29 by mahavirj