espressif/esp-idf #5192: SPI Flash driver definitions not shared

alexanderturner, 2020-04-26:
Hey there - 

Doc's point to: spi_flash_host_drv.h
and refer to spi host driver files that dont appear to exist.

As an aside - does ESP have a list of support and tested flash chips that work with the ESP32?

```
Host driver

The host driver relies on an interface (spi_flash_host_driver_t) defined in the spi_flash_host_drv.h (in the soc/include/hal folder). This interface provides some common functions to communicate with the chip.

In other files of the SPI HAL, some of these functions are implemented with existing ESP32 memory-spi functionalities. However due to the speed limitations of ESP32, the HAL layer can’t provide high-speed implementations to some reading commands (So we didn’t do it at all). The files (memspi_host_driver.h and .c) implement the high-speed version of these commands with the common_command function provided in the HAL, and wrap these functions as spi_flash_host_driver_t for upper layer to use.

You can also implement your own host driver, even with the GPIO. As long as all the functions in the spi_flash_host_driver_t are implemented, the esp_flash API can access to the flash regardless of the low-level hardware.

Chip driver

The chip driver, defined in spi_flash_chip_driver.h, wraps basic functions provided by the host driver for the API layer to use.

Some operations need some commands to be sent first, or read some status after. Some chips need different command or value, or need special communication ways.

There is a type of chip called generic chip which stands for common chips. Other special chip drivers can be developed on the base of the generic chip.

```

mythbuster5, 2020-08-26:
Yeah... you are right. The document needs to be updated. We are doing that..

Now ,the interface spi_flash_host_driver is defined in `memspi_host_driver.h`.

The list of supported flash are updating in Doc. For now, we support most of 24-bit address flash.

Thanks a lot.

Alvin1Zhang, 2021-01-29:
Thanks for reporting and sorry for slow turnaround, the fix is available https://github.com/espressif/esp-idf/commit/82039cdbc15b0a5bc3279a17954171e3bc0b1c08, feel free to reopen, thanks.

closed 2021-01-29 by Alvin1Zhang