espressif/esp-idf #5193: idf.py double-loads "idf_ext.py" in project directory.

BrianPugh, 2020-04-26:
Starting [here](https://github.com/espressif/esp-idf/blob/master/tools/idf.py#L635), `idf.py` begins searching for and loading extensions. If the environment variable `IDF_EXTRA_ACTIONS_PATH` doesn't exist, it defaults to an empty string. When searching for python files ending in `'_ext'`, an empty string for a directory is interpretted as the current directory (which for many cases is the project directory). So, if the project has an `idf_ext.py`, it gets loaded here. A few lines down (under commend `# Load extensions from project dir`), `idf_ext.py` is searched for again and reloaded, duplicating the extension.

So there are a few solutions (I didn't open a PR because I'm not sure of external implications/backwards compatibility). I personally thing solution (1) is the cleanest and actually reduces the current amount of code:

1. Only append `extra_paths` if it's a non-empty. Not sure if other esp-idf elements relied on the current directory being added to the search path. I think this may have been the intended implementation, because it doesn't seem correct that we should only load the current directory if the environment variable `IDF_EXTRA_ACTIONS_PATH`. Additionally, by default, the current directory is already in the python path.

```
# Make a set to automatically handle duplicate paths
extension_dirs = set([os.path.abspath(os.path.normpath(idf_py_extensions_path))])
extra_paths = os.environ.get("IDF_EXTRA_ACTIONS_PATH")
if extra_paths is not None:
    extension_dirs.update([os.path.abspath(os.path.normpath(path)) for path in extra_paths.split(';')])
```

In this event, its simpler and cleaner to append the `project_dir` to `extension_dirs` and get rid of the code under   `# Load extensions from project dir`:

```
extension_dirs.append(project_dir)
```

And then finally, convert it back to a list:

```
extension_dirs = list(extension_dirs)
```

2. If we do intend importing the current directory (which may be the project directory, this solution is absolutely backwards compatible), make sure all paths are absolute and normalized (i.e. `os.path.abspath(os.path.normpath(some_extension_dir))`). Prior to loading an extension, check if the absolute-normalized-path to the file is in a newly-created list-of-loaded-paths. If the extension is already loaded, don't reload it.

BrianPugh, 2020-04-26:
totally unrelated: the `run` method of `Task` would be more appropriate as the magic `__call__` method

Alvin1Zhang, 2020-04-27:
@BrianPugh Thanks for reporting, we will look into.

BrianPugh, 2020-05-11:
any update? My included code is basically a complete solution

kumekay, 2020-05-12:
Hi @BrianPugh thank you for the issue once again.
I think your solution is ok, could you please open a PR?

The only minor thing:
instead of using `os.path.abspath(os.path.normpath(path))` there is a `realpath` function defined in `idf_py_actions.tools`. Could you please use it?

And one more tiny note: `os.path.abspath` already calls  `normalize`, so in all cases `abspath` is enough https://github.com/python/cpython/blob/3.8/Lib/posixpath.py#L381

kumekay, 2020-05-12:
> totally unrelated: the `run` method of `Task` would be more appropriate as the magic `__call__` method

This won't break anything and make it a bit cleaner, so feel free to make a PR, thank you for the note!

kumekay, 2020-05-19:
Hi @BrianPugh thank you for you PRs once again, both of them are merged now. However, 
it was necessary to rebase your commits over more recent master, so there are new commit ids and github cannot handle it automatically, 
if you are ok with the changes, could you please close your PRs and this issue?

BrianPugh, 2020-05-19:
closed in https://github.com/espressif/esp-idf/commit/3a6b8bb6e3116c8780b6a5c8638489451bbca51a

closed 2020-05-19 by BrianPugh