espressif/esp-idf #5194: adjtime implements olddelta incorrectly

tve, 2020-04-26:
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.0 through v4.2

## Problem Description

The POSIX and GNU libc definition of adjtime specifies:
> If the olddelta argument is not a null pointer, the adjtime function returns information about any previous time adjustment that has not yet completed.

Note the word "previous". The ESP-IDF implementation incorrectly returns
the current time adjustment that has not yet completed, which is rather useless.

### Fix

Swap the two outer `if` blocks in the adjtime function https://github.com/espressif/esp-idf/blob/master/components/newlib/time.c#L193.

Alvin1Zhang, 2020-04-27:
@tve Thanks for reporting, we will look into.

KonstantinKondrashov, 2020-06-08:
Hi @tve!
Thanks for your report. I agree, now we need to call it twice to get the same result. I will fix it and provide backports up to 4.0.

closed 2020-07-10 by espressif-bot