espressif/esp-idf #5195: I2S DMA ADC stops working after esp_bt_controller_enable is called.

admiralmaggie, 2020-04-26:
## Environment

- Development Kit:      ESP32-DevKitC
- Module or chip used:  ESP32-WROOM-32
- IDF version: Tried both v4.2-dev-1054-g7d0f1536e and v4.0
- Build System: CMake
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System:     Windows
- Using an IDE?:  Yes (Eclipse 2020-03)
- Power Supply:         USB

## Problem Description

I2S DMA ADC stops working when `esp_bt_controller_enable` is called. I have tried both v4.2-dev-1054-g7d0f1536e and v4.0. The same issue exist. The example used is A2DP_SOURCE with an added I2S sampling task. I have tried initializing I2S before, and after BT with no luck. Hope I'm not doing anything wrong!

### Expected Behavior

I'm expecting I2S DMA ADC to continue to function even after BT stack is initialized. 

### Actual Behavior

I2S produces a dozen samples and then after `esp_bt_controller_enable` is called it constantly returns zero values.

### Steps to reproduce

1. Clone ESP-IDF repo.
2. Copy A2DP_SOURCE example.
3. Copy and paste main.c routine from attached file, or gist. It is identical to A2DP_SOURCE code but I have added a BT name, a I2S sampling task, and a flag to enable/disable BT routines.
4. Open serial terminal.
5. When BT is active (#define BT_ACTIVE 1), I2S ADC samples become zero after BT controller is initialized.
6. When BT is disabled (#define BT_ACTIVE 0) I2S ADC continues to function without an issue.

### Code to reproduce this issue

https://gist.github.com/admiralmaggie/65f27cb5291db50ce227b44e61c5a8d3

## Debug Logs

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7008
ho 0 tail 12 room 4
load:0x40078000,len:13696
load:0x40080400,len:3984
entry 0x40080688
I (30) boot: ESP-IDF v4.2-dev-1054-g7d0f1536e-dirty 2nd stage bootloader
I (30) boot: compile time 19:32:29
I (32) boot: chip revision: 1
I (35) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (53) boot.esp32: SPI Speed      : 40MHz
I (53) boot.esp32: SPI Mode       : DIO
I (53) boot.esp32: SPI Flash Size : 4MB
I (56) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  2 factory          factory app      00 00 00010000 001f0000
I (95) boot: End of partition table
I (99) boot_comm: chip revision: 1, min. application chip revision: 0
I (106) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1bb78 (113528) map
I (158) esp_image: segment 1: paddr=0x0002bba0 vaddr=0x3ffbdb60 size=0x03538 ( 13624) load
I (164) esp_image: segment 2: paddr=0x0002f0e0 vaddr=0x40080000 size=0x00404 (  1028) load
I (166) esp_image: segment 3: paddr=0x0002f4ec vaddr=0x40080404 size=0x00b2c (  2860) load
I (175) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x74cc8 (478408) map
I (365) esp_image: segment 5: paddr=0x000a4cf0 vaddr=0x40080f30 size=0x129f4 ( 76276) load
I (409) boot: Loaded app from partition at offset 0x10000
I (409) boot: Disabling RNG early entropy source...
I (410) cpu_start: Pro cpu up.
I (413) cpu_start: Application information:
I (418) cpu_start: Project name:     a2dp_source
I (423) cpu_start: App version:      f054da7-dirty
I (429) cpu_start: Compile time:     Apr 26 2020 19:31:58
I (435) cpu_start: ELF file SHA256:  c11fc5e60a23b166...
I (441) cpu_start: ESP-IDF:          v4.2-dev-1054-g7d0f1536e-dirty
I (448) cpu_start: Starting app cpu, entry point is 0x400814fc
I (0) cpu_start: App cpu up.
I (458) heap_init: Initializing. RAM available for dynamic allocation:
I (465) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (471) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (477) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (483) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (489) heap_init: At 3FFCC5B8 len 00013A48 (78 KiB): DRAM
I (496) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (502) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (508) heap_init: At 40093924 len 0000C6DC (49 KiB): IRAM
I (515) cpu_start: Pro cpu start user code
I (533) spi_flash: detected chip: generic
I (533) spi_flash: flash io: dio
I (534) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (600) BTDM_INIT: BT controller compile version [d9cc6b1]
I (600) system_api: Base MAC address is not set
I (600) system_api: read default base MAC address from EFUSE
I (600) I2S: DMA Malloc info, datalen=blocksize=512, dma_buf_count=16
I (610) I2S: PLL_D2: Req RATE: 44100, real rate: 2777.000, BITS: 16, CLKM: 30, BCK: 60, MCLK: 30.234, SCLK: 88864.000000, diva: 64, divb: 14
I (710) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (1100) BT_AV: Starting device discovery...
W (1110) BT_BTC: btc_avrc_tg_set_rn_supported_evt failed: AVRC TG not yet initialized

I (1130) BT_AV: event: 10
I (1140) BT_AV: event: 10
I (1140) BT_AV: event: 10
I (1140) BT_AV: event: 10
I (1140) BT_AV: Discovery started.
I (1630) I2S: PLL_D2: Req RATE: 44100, real rate: 2777.000, BITS: 16, CLKM: 30, BCK: 60, MCLK: 30.234, SCLK: 88864.000000, diva: 64, divb: 14
I (1630) I2S: 355
I (1630) I2S: 355
I (1630) I2S: 355
I (1640) I2S: 355
I (1640) I2S: 355
I (1640) I2S: 355
I (1640) I2S: 355
I (1650) I2S: 355
I (1650) I2S: 355
I (1650) I2S: 355
I (1650) I2S: 27
I (1660) I2S: 94
I (1660) I2S: 144
I (1660) I2S: 183
I (1670) I2S: 227
I (1670) I2S: 261
I (1670) I2S: 305
I (1670) I2S: 338
I (1680) I2S: 355
I (1680) I2S: 355
I (1680) I2S: 355
I (1680) I2S: 355
I (4440) BT_AV: Scanned device: ac:3f:a4:55:50:ad
I (4440) BT_AV: --Class of Device: 0x60680
I (4440) BT_AV: --RSSI: -89
I (4880) BT_AV: Scanned device: ac:3f:a4:55:50:ad
I (4880) BT_AV: --Class of Device: 0x60680
I (4880) BT_AV: --RSSI: -73
...
```
I2S ADC stops working right after BT controller is enabled.

## Other items if possible
sdkconfig file attached.
elf file is attached.
main.c is attached.
[gt.zip](https://github.com/espressif/esp-idf/files/4536705/gt.zip)

Alvin1Zhang, 2020-04-27:
@admiralmaggie Thanks for the detailed report, we will look into. Thanks.

koobest, 2020-04-28:
Hi, @admiralmaggie 
It should be that the chip has entered the low-power mode and the power of the ADC has been turned off. Please call `esp_bt_sleep_disable` to disable the low-power mode.

thanks !!

admiralmaggie, 2020-04-28:
@koobest Thank you! That makes a lot of sense. I will try and get back to you.

admiralmaggie, 2020-04-30:
Calling the `esp_bt_sleep_disable` after `esp_bt_controller_enable` does fix the issue... Thanks! Closing the issue.

closed 2020-04-30 by admiralmaggie