espressif/esp-idf #5196: Hostname failed to resolve in DHCP with fixed address

thecashit, 2020-04-27:
How to i correct set to work hostname resolving in DHCP with fixed address?

We are use ESP32 with ESP-IDF v4.0.

In our firmware we need to get information from some APIs endpoints. Endpoints listed by hostname.

Previously, the hostname resolve occurred through special functions, but now the HTTPClient component make all necessary work.

We setup Google DNS in the start of firmware with ::dns_setserver() function.
2 servers are marked: 8.8.8.8 and 8.8.4.4.

Everything works well in STA mode except one thing: when firmware connects to DHCP with fixed IP (for example, by MAC binding) device can't resolve hostnames and most of the logic stops working in this case.
Moreover, this does not happen immediately, but a few hours after the connection
We tried on three access points and the same behavior is visible on all

What is the correct way to setup DNS servers with ESP-IDF to works in all cases?

What we already tried:
1) Don't set DNS servers (After a while, the host stops resolving either in standart DHCP and DHCP with fixed IP)
2) Try to set DNS servers after connecting (didn't work)
3) Try to set DNS servers to AP ip (seems like didn't work)

freakyxue, 2020-05-18:
hi @thecashit 
Which function do you use to parse DNS.
`getaddrinfo `or ' gethostbyname'

thecashit, 2020-05-19:
> hi @thecashit
> Which function do you use to parse DNS.
> `getaddrinfo `or ' gethostbyname'

This is a very good question.
It seems I have not described the problem well enough. 
DNS resolve failed when using standard components included in esp-idf: esp-http-client and esp-mqtt.

However, for some reason, I can’t immediately find what exactly is used in these components. There was probably some kind of refactoring in this component, since today i installed 4.1-beta and the SUPPOSEDLY problem is solved.

freakyxue, 2020-05-19:
hi @thecashit 
Please continue to test. If you have any questions, please continue to ask. Thank you

freakyxue, 2020-06-10:
HI @thecashit Is there any problem with this? If not, I think this problem can be closed

thecashit, 2020-06-10:
@freakyxue As i see problem was solved
Sorry for long answer - we will try to us different routers to get exactly information.

I will close this issue.

closed 2020-06-10 by thecashit