espressif/esp-idf #5197: Bluetooth and SoftAP at same time not working

MHAN0, 2020-04-27:
Hello,

with IDF Branch/Release 4.0  "Bluetooth" and "SoftAP"  are not working in coexistence.

I have written an Android App, which uses the both networks WiFi (Udp Client) and Bluetooth (BDR Spp Client). First, i have individually tested both network . The communication over UDP Client or BDR Client with the ESP32 working really good.

if i try to use both network at the same time, Softap is lost and Bluetooth works further.
Softap is during a Bluetooth communication completly disabled from ESP32. 

What I have observed is:

1) After Reset ESP32, the network Bluetooth is ready (shows ESP_SPP_START_EVT ) and SoftAP also
2) I have connected succesfully my tablet with the SSID from SoftAP and pair successfully the  ESP32 BT device.
3) ESP32 is now connected with both networks without any error
4) Now I start the app. This first opens a UDP socket and starts communication. Works.
5) After that, i open in my Android APP a BT socket and connect only. ESP32 reports 
  "ESP_SPP_SRV_OPEN_EVTP" and also for three times "UDP sock error during sending". During this 
   phase the udp communication hangs but after that it works again.
6) Now, i start the communication with the Bluetooth Client (tx/rx of data). In this phase EPS32 reports "UDP sock error during sending" more times and then the wifi is disconnected. After this report SoftAP is not available more. During this phase the Bluetooth communication (tx/rx) working fine. 
7) If i now stop the Bluetooth communication, SoftAP is reactivating but i can not connect with the SSID to the ESP32. 
8) If i now disconnect Bluetooth completly, than it is possible to connect to the EPS32.

It seems, Bluetooth communication stop the SoftAP completely.
What is the reason for that. Can you help me please. Thanks.

For a better understanding I copied the terminal, the installed software and my sdkconfig

############################ TERMINAL #############################
rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6696
load:0x40078000,len:12612
load:0x40080400,len:3788
entry 0x40080684
I (47) boot: Chip Revision: 1
I (48) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot: ESP-IDF -128-NOTFOUND 2nd stage bootloader
I (39) boot: compile time 07:59:42
I (48) boot: Enabling RNG early entropy source...
I (48) boot: SPI Speed      : 80MHz
I (48) boot: SPI Mode       : DIO
I (52) boot: SPI Flash Size : 4MB
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00300000
I (90) boot: End of partition table
I (94) boot_comm: chip revision: 1, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x24060 (147552) map
I (158) esp_image: segment 1: paddr=0x00034088 vaddr=0x3ffbdb60 size=0x040a8 ( 16552) load
I (164) esp_image: segment 2: paddr=0x00038138 vaddr=0x40080000 size=0x00400 (  1024) load
I (165) esp_image: segment 3: paddr=0x00038540 vaddr=0x40080400 size=0x07ad0 ( 31440) load
I (185) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x99834 (628788) map
I (389) esp_image: segment 5: paddr=0x000d9854 vaddr=0x40087ed0 size=0x15f5c ( 89948) load
I (441) boot: Loaded app from partition at offset 0x10000
I (441) boot: Disabling RNG early entropy source...
I (441) cpu_start: Pro cpu up.
I (445) cpu_start: Application information:
I (450) cpu_start: Project name:     ESP-APP
I (454) cpu_start: App version:      1
I (459) cpu_start: Compile time:     Apr 26 2020 08:15:39
I (465) cpu_start: ELF file SHA256:  c4367f267c70797d...
I (471) cpu_start: ESP-IDF:          -128-NOTFOUND
I (476) cpu_start: Starting app cpu, entry point is 0x40081648
I (0) cpu_start: App cpu up.
I (487) heap_init: Initializing. RAM available for dynamic allocation:
[... 1164 lines cut ...]
CONFIG_TCP_OVERSIZE_MSS=y
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=32
CONFIG_TCPIP_TASK_STACK_SIZE=3000
# CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY is not set
CONFIG_TCPIP_TASK_AFFINITY_CPU0=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x0
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=4096
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_IP_LOST_TIMER_INTERVAL=120
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
# End of deprecated options

Alvin1Zhang, 2020-04-27:
@MHAN0 Thanks for the detailed report, we will look into. Thanks.

YouDONG-ESP, 2020-10-26:
Hi @MHAN0, ESP32 will not fully support softAP + Bluetooth. Just support WiFi softAP provision + Bluetooth.

KaeLL, 2020-10-26:
@YouDONG-ESP care to elaborate further, with examples?

YouDONG-ESP, 2020-10-27:
Hi @KaeLL, you can find the example here https://github.com/espressif/esp-idf/tree/master/examples/provisioning

TimXia, 2020-12-15:
@MHAN0 ESP32 can not well support data transaction of Wi-Fi softAP and Bluetooth simutaneously for Wi-Fi and Bluetooth can not receive simutaneously. Please try Wi-Fi station + Bluetooth.

jack0c, 2023-04-13:
There is some optimizations on SoftAP + Bluetooth coexistence, will release it later.