espressif/esp-idf #5199: Spiffs partition failure

NJenny, 2020-04-27:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version (for DevKitC): [v1|v2|v3|v4]
- Module or chip used:  ESP32-WROOM-32U
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
 ESP-IDF v3.3.1
- Build System: Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System: Windows
- (Windows only) environment type: MSYS2 mingw32
- Using an IDE?: No
- Power Supply: external 5V.

## Problem Description
Hey, I have been working on spiffs on  ESP-IDF v3.3.1. The Spiffs is successfully mounted but there is a failure in creating SPIFFS partion. I am using the sample code of Spiffs given in the example folder. I actually wanted to merge simple OTA example with SPIFFS. So i tried running the spiffs example code in the Spiffs example folder itself.  When i run the code in the Spiffs folder it works fine but when i copy and paste the same code in some other file like the simple OTA folder , there is a failure of SPIFFS partition.

### Expected Behavior
Spiffs should be mounted and should be readily available for reading and writing of files.
### Actual Behavior
A failure in Spiffs partition. 
### Steps to reproduce
A screenshot of the serial montor is provided below .
[image]

### Code to reproduce this issue
This is the example code present in the storage/spiffs directory.
```code 
void app_main(void)
{
    ESP_LOGI(TAG, "Initializing SPIFFS");
    
    esp_vfs_spiffs_conf_t conf = {
      .base_path = "/spiffs",
      .partition_label = NULL,
      .max_files = 5,
      .format_if_mount_failed = true
    };
    
    // Use settings defined above to initialize and mount SPIFFS filesystem.
    // Note: esp_vfs_spiffs_register is an all-in-one convenience function.
    esp_err_t ret = esp_vfs_spiffs_register(&conf);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            ESP_LOGE(TAG, "Failed to mount or format filesystem");
        } else if (ret == ESP_ERR_NOT_FOUND) {
            ESP_LOGE(TAG, "Failed to find SPIFFS partition");
        } else {
            ESP_LOGE(TAG, "Failed to initialize SPIFFS (%s)", esp_err_to_name(ret));
        }
        return;
    }
    
    size_t total = 0, used = 0;
    ret = esp_spiffs_info(NULL, &total, &used);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "Failed to get SPIFFS partition information (%s)", esp_err_to_name(ret));
    } else {
        ESP_LOGI(TAG, "Partition size: total: %d, used: %d", total, used);
    }

    // Use POSIX and C standard library functions to work with files.
    // First create a file.
    ESP_LOGI(TAG, "Opening file");
    FILE* f = fopen("/spiffs/hello.txt", "w");
    if (f == NULL) {
        ESP_LOGE(TAG, "Failed to open file for writing");
        return;
    }
    fprintf(f, "Hello World!\n");
    fclose(f);
    ESP_LOGI(TAG, "File written");
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
I (0) cpu_start: App cpu up.
I (271) heap_init: Initializing. RAM available for dynamic allocation:
I (278) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (284) heap_init: At 3FFB2ED0 len 0002D130 (180 KiB): DRAM
I (290) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (296) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (303) heap_init: At 400895C4 len 00016A3C (90 KiB): IRAM
I (309) cpu_start: Pro cpu start user code
I (327) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (329) example: Initializing SPIFFS
E (329) SPIFFS: spiffs partition could not be found
E (329) example: Failed to find SPIFFS partition

```
[image]

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

igrr, 2020-04-27:
> When i run the code in the Spiffs folder it works fine but when i copy and paste the same code in some other file like the simple OTA folder , there is a failure of SPIFFS partition.

It fails because in the partition table used by the simple OTA example there is no SPIFFS partition, hence the message that the partition can not be found. You need to create a partition table which includes both OTA and SPIFFS partitions, for example by adding spiffs partition to the partition table of the OTA example.

Please check https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html for more details about partition tables.

NJenny, 2020-05-05:
thank you so much for your help @igrr .

closed 2020-05-05 by NJenny