espressif/esp-idf #5200: OTA_Esp8266 using http

sowmimalu, 2020-04-27:
Hi,
 i am using esp8266 development board...i am try to download a file from local server by using http..
In my application i need to update ota  in esp8266 by using http.... I got one example in arduino(webupdater)when i tried with this example webpage has opend i am able to choose a bin file which i want form that i can able to update the bin file by OTA. Now i need  to download a file in my source code by using http ...i tried basic example in ardunio(httpupdate) but its not work ...i got connection refused error ....when I tried to download a bin  file from local server.

chegewara, 2020-04-27:
Wrong address.

sowmimalu, 2020-04-28:
can you please suggest me.. to how to download a file  by http from my local server..

http://ip/new/hello_world.bin
 *ip  i have given my system ip as in above command
what is the correct way to download a file and then to update 
Is there any example??

sowmimalu, 2020-04-28:
sorry, posted in esp8266 github

Alvin1Zhang, 2020-04-28:
@sowmimalu Tracking in https://github.com/espressif/ESP8266_RTOS_SDK/issues/884, will close this one, thanks.

closed 2020-04-28 by Alvin1Zhang