espressif/esp-idf #5201: OTA Failure

lakshmisivareddy, 2020-04-28:
Hi i am trying to download ota from aws s3 
i am passing url information to device  from OTA topic which device is subscribed 
once device receives the url info i am creating a http client in a sup rate task and initiating downloading 

i am ended up with below error 

E (306647) esp-tls: mbedtls_ssl_setup returned -0x7f00
E (306651) esp-tls: Failed to open new connection
E (306651) TRANS_SSL: Failed to open a new connection
E (306654) HTTP_CLIENT: Connection failed, sock < 0

i tried enabling  MBEDTLS_ASYMMETRIC_CONTENT_LEN but ended up with below error 

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40183cab  PS      : 0x00060530  A0      : 0x8010976d  A1      : 0x3fff9db0
0x40183cab: esp_http_client_set_header at D:/WorkSpace/esp-32/newesp32/esp-at/esp-idf/components/esp_http_client/esp_http_client.c:1226

my code:

void myotaSubCallBackHandler(char *topicName, int payloadLen, char *payLoad) {

	//printf("OTA Data Recived %s \r\n", payLoad);

	cJSON *object = cJSON_Parse((char*) payLoad);

	cJSON *url = cJSON_GetObjectItem(object, "url");
	if (url != NULL)
		printf("data %s\r\n", url->valuestring);

	cJSON *version = cJSON_GetObjectItem(object, "version");
	if (version != NULL)
		printf("data %s\r\n", version->valuestring);

	cJSON *size = cJSON_GetObjectItem(object, "size");
	if (size != NULL)
		printf("data %d\r\n", size->valueint);

	if(ota_data != NULL)
		free(ota_data);

  ota_data =(struct OTA_info* ) calloc(1,sizeof(struct OTA_info));

  if(ota_data != NULL)
  {
	  ota_data->ota_uri = (char*)calloc(strlen(url->valuestring),sizeof(char));
	  memcpy(ota_data->ota_uri, url->valuestring, strlen(url->valuestring));
	  ota_data->ota_size = size->valueint;
  }
  received_ota = true;
    ota_init(ota_data);
	if(object != NULL)
			cJSON_Delete(object);

}

void ota_init(struct OTA_info *ota_data)
{
   if((status = spi_flash_erase_range(HOST_MCU_FLASH_START,SPI_FLASH_SEC_SIZE)) == ESP_OK)
 {
				  printf("4096 bytes of flash erased starting  %x\r\n",HOST_MCU_FLASH_START);
 }

 xTaskCreate(&ota_task,"ota_task",8*1024,(void*)ota_data,4,NULL);
}

void ota_task(void *param)
{
	
	 struct OTA_info *data = (struct OTA_info *) param; 
	 if(data != NULL)
	 {
	    esp_http_client_config_t config = {
			.url = data->url,
			//.cert_pem = (char *)server_cert_pem_start,
			.event_handler = http_event_handler,
		};
		
		 client = esp_http_client_init(&config);
		  esp_err_t err = esp_http_client_perform(client);

    if (err == ESP_OK) {
        printf("HTTP chunk encoding Status = %d, content_length = %d \r\n",
                esp_http_client_get_status_code(client),
                esp_http_client_get_content_length(client));
    } else {
        printf("Error perform http request %s \r\n", esp_err_to_name(err));
    }
	 vTaskDelete( NULL );
	
}

esp_err_t http_event_handler(esp_http_client_event_t *evt)
{
    switch(evt->event_id) {
        case HTTP_EVENT_ERROR:
            ESP_LOGI(OTA_TAG, "HTTP_EVENT_ERROR");
            break;
        case HTTP_EVENT_ON_CONNECTED:
            ESP_LOGI(OTA_TAG, "HTTP_EVENT_ON_CONNECTED");
            break;
        case HTTP_EVENT_HEADER_SENT:
            ESP_LOGI(OTA_TAG, "HTTP_EVENT_HEADER_SENT");
            break;
        case HTTP_EVENT_ON_HEADER:
            ESP_LOGI(OTA_TAG, "HTTP_EVENT_ON_HEADER, key=%s, value=%s", evt->header_key, evt->header_value);
            break;
        case HTTP_EVENT_ON_DATA:
            printf("HTTP_EVENT_ON_DATA, len=%d \r\n", evt->data_len);
					
            if (!esp_http_client_is_chunked_response(evt->client)) {
	
            }

            break;
        case HTTP_EVENT_ON_FINISH:
            ESP_LOGI(OTA_TAG, "HTTP_EVENT_ON_FINISH");
            break;
        case HTTP_EVENT_DISCONNECTED:
            ESP_LOGI(OTA_TAG, "HTTP_EVENT_DISCONNECTED");
            break;
    }
    return ESP_OK;
}

any help would be appreciated

Reagrds
Siva

chegewara, 2020-04-28:
https://github.com/espressif/esp-idf/issues/3869#issuecomment-517793077

mahavirj, 2020-04-29:
@lakshmisivareddy 

Seems like insufficient dynamic memory issue, you can track that using `esp_get_free_heap_size()`.

> xTaskCreate(&ota_task,"ota_task",81024,(void)ota_data,4,NULL);

Stack size here is 8K, right? (Seems like formatting issue otherwise please reduce it to 8K)

In addition you have not provided any information per issue template so not sure what IDF version you are using but following can help to get some additional memory provided you are using `master` branch:

- Create file `sdkconfig.defaults` with following configuration options in application directory

```
CONFIG_FREERTOS_UNICORE=y
CONFIG_ESP32_IRAM_AS_8BIT_ACCESSIBLE_MEMORY=y
CONFIG_MBEDTLS_IRAM_8BIT_MEM_ALLOC=y
```
- Regenerate `sdkconfig` using

```
rm sdkconfig
idf.py reconfigure
idf.py build
```

This will enable some config options to redirect TLS connection buffers (in/out) to IRAM but it is restricted to single core mode only. Hope this helps.

mahavirj, 2020-05-11:
@lakshmisivareddy Did above suggestion helped you? Can we close this issue?

lakshmisivareddy, 2020-05-11:
Thanks For the support

closed 2020-05-11 by lakshmisivareddy