espressif/esp-idf #5204: Error while running install.sh : unrecognized arguments: --no-site-packages

amrutnrp, 2020-04-28:
- IDF version ( v4.0
- Operating System:     Linux Mint

Hi, 
I am trying to install esp-idf in linux from terminal. I have already installed all required tools.
while running ./install.sh, i am getting this error. 

'virtualenv: error: unrecognized arguments: --no-site-packages'

Detailed logs below.

Please help.
Thanks

> HP-241-G1:~/esp/esp-idf$ ./install.sh
>
> Installing ESP-IDF tools
> Installing tools: xtensa-esp32-elf, esp32ulp-elf, openocd-esp32
> Skipping xtensa-esp32-elf@esp-2019r2-8.2.0 (already installed)
> Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
> Skipping openocd-esp32@v0.10.0-esp32-20190313 (already installed)
> Installing Python environment and packages
> Creating a new Python environment in /home/kanha/.espressif/python_env/idf4.0_py2.7_env
> usage: virtualenv [--version] [--with-traceback] [-v | -q] [--app-data APP_DATA] [--clear-app-data] [--discovery {builtin}] [-p py] [--creator {builtin,cpython2-posix}] [--seeder {app-data,pip}] [--no-seed] [--activators comma_sep_list]
>                   [--clear] [--system-site-packages] [--symlinks | --copies] [--download | --no-download] [--extra-search-dir d [d ...]] [--pip version] [--setuptools version] [--wheel version] [--no-pip] [--no-setuptools] [--no-wheel]
>                   [--symlink-app-data] [--prompt prompt] [-h]
>                   dest
> virtualenv: error: unrecognized arguments: --no-site-packages
> Traceback (most recent call last):
>   File "/home/kanha/esp/esp-idf/tools/idf_tools.py", line 1372, in <module>
>     main(sys.argv[1:])
>   File "/home/kanha/esp/esp-idf/tools/idf_tools.py", line 1368, in main
>     action_func(args)
>   File "/home/kanha/esp/esp-idf/tools/idf_tools.py", line 1184, in action_install_python_env
>     stdout=sys.stdout, stderr=sys.stderr)
>   File "/usr/lib/python2.7/subprocess.py", line 190, in check_call
>     raise CalledProcessError(retcode, cmd)
> subprocess.CalledProcessError: Command '['/usr/bin/python', '-m', 'virtualenv', '--no-site-packages', '/home/kanha/.espressif/python_env/idf4.0_py2.7_env']' returned non-zero exit status 2
>

igrr, 2020-04-28:
Hi @amrutnrp, this has been fixed in commit 7ccf6eb on release/v4.0 branch. The fix will be part of the next bugfix release (v4.0.1). At the moment you can either:

* update to the latest release/v4.0, or 
* cherry-pick the commit 7ccf6eb

amrutnrp, 2020-04-28:
Hi @igrr , 
After implementing this change, the error went away.
Note- after release of v4.0.1, the error is fixed.

Thanks

closed 2020-04-28 by amrutnrp