espressif/esp-idf #5205: Smart config works unreliably.

alfonsohera, 2020-04-28:
**IDF version: v4.1-dev-281-g96b96ae24**

Running the smartconfig example alone works reliably. However when adapting my system to add the smart config functionality, the SC task seems not to run. I observed the debug output of the uC and the messages that the event handler (left as in the example) should output when the SC events are recognized (Scan done, found channel, got SSID PWD) are never shown. During this time no matter how many attempts are done with the ESP Touch app, the uC will not get the WiFi credentials. 

A little bit of background, my system is running several tasks along the SC task and among them, there is a HW timer configured to trigger an ADC sampling task every 100 ms. If I don't start this HW timer, the SC task runs without problems (output messages are observed and the WiFi credentials can be set via the app). 
This timer has proven problematic in the past with my system (causing problems in the execution of other tasks or even not allowing them to run) and back then my "solution" was to increase the timer period (originally set to 300 us). 100 ms is a period I can still accept (although the original would be much better) but more than that would not be acceptable. 

A couple of questions:
1.- Is this SC problem related to the HW timer? How could it be solved? I tried pinning the WiFi driver's and SC's tasks to core 1 as well as increasing the task's priority with no avail.
2.- Is the behavior of the HW timer expected as is? I understand that in order to keep track of the time accurately, it has to have a very high priority, but interfering with other tasks seems to make the use of the HW timer not so attractive.
3.- If the HW timer causes the aforementioned issues and there is no way to circumvent this, what other options are available to trigger fast acquisition of ADC samples?

Thanks

Alfonso

Alvin1Zhang, 2020-04-28:
@alfonsohera Thanks for reporting, we will look into.

howroyd, 2020-05-05:
Similar issue, not running the ADC though.  Never seems to get the SSID and password.

With logging turned on. IDF v4.2 and latest ESPTouch from the Play store:
```
D (00:00:01.166) WIFI: WIFI_EVENT_STA_START
I (2511) smartconfig: SC version: V2.9.0
I (5651) wifi:ic_enable_sniffer
I (5651) smartconfig: Start to find channel...
I (23611) smartconfig: T|once 1 80
I (23611) smartconfig: save rssi:-83 ,bssid:1c:b7:2c:c1:0a:60
I (23611) smartconfig: iCh lock
I (27991) smartconfig: confirm rssi:-83, bssid:1c:b7:2c:c1:0a:60
I (27991) smartconfig: TYPE: ESPTOUCH
I (27991) smartconfig: T|sniffer on ch:2,width:0
I (27991) smartconfig: T|AP MAC: 1c:b7:2c:c1:0a:60
I (27991) smartconfig: T|Head Len:80
I (28001) smartconfig: Found channel on 2-0. Start to get ssid and password...
I (28001) smartconfig: T|SYNC STATUS
I (62421) smartconfig: T|e-12
I (63611) smartconfig: T|e-22
I (73991) wifi:ic_disable_sniffer
I (73991) smartconfig: smartconfig restart
I (74041) smartconfig: SC version: V2.9.0
I (77181) wifi:ic_enable_sniffer
I (77181) smartconfig: Start to find channel...
```

alfonsohera, 2020-05-05:
Your issue might be different @howroyd than the one I reported. In my case the log messages from the smartconfig task are never seen if the HW timer is running (meaning that the task is probably being perpetually blocked). In your case, the task _is_ running as seen in your debug output. Are you trying to share the credentials of a network configured to 5GHz? According what I have read, the sniffer can't get the credentials from such a network (only the 2.4 GHz band works).

howroyd, 2020-05-05:
Ah ok.
Nope, I am using 2.4G for the ESP.
I have an event handler for `SC_EVENT_GOT_SSID_PSWD` which seems to never get called:
```
esp_event_handler_register(SC_EVENT, SC_EVENT_GOT_SSID_PSWD, _sc_event_handler, NULL);
```
```
void Wifi::_sc_event_handler(void *arg, esp_event_base_t base, int32_t event_id, void *data)
{
    smartconfig_event_got_ssid_pswd_t *evt = (smartconfig_event_got_ssid_pswd_t *)data;
    uint8_t ssid[33] = { 0 };
    uint8_t password[65] = { 0 };
    uint8_t cellphone_ip[4];
    esp_err_t err = ESP_OK;

    memcpy(ssid, evt->ssid, sizeof(evt->ssid));
    memcpy(password, evt->password, sizeof(evt->password));
    memcpy(cellphone_ip, evt->cellphone_ip, sizeof(evt->cellphone_ip));

    ESP_LOGD(LOG_TAG, "SSID:%s", ssid);
    ESP_LOGD(LOG_TAG, "PASSWORD:%s", password);
    ESP_LOGD(LOG_TAG, "Phone ip: %d.%d.%d.%d\n", cellphone_ip[0], cellphone_ip[1], cellphone_ip[2], cellphone_ip[3]);

    err = sc_send_ack_start(evt->type, evt->token, evt->cellphone_ip);
    if (err != ESP_OK) {
        ESP_LOGE(LOG_TAG, "Send smartconfig ACK error: %d", err);
    }

    memcpy(wifi_config.sta.ssid, ssid, sizeof(ssid));
    memcpy(wifi_config.sta.password, password, sizeof(password));

    xEventGroupSetBits(event_group, WIFI_SC_DONE_BIT);
}
```

howroyd, 2020-05-05:
I've opened a separate issue for my case, since it seems different to yours.  Apologies for hijacking!
https://github.com/espressif/esp-idf/issues/5243

alfonsohera, 2020-05-05:
No worries, I think there are many SC-related issues that in the end should be investigated as a whole. Good luck with your problem!

liuzfesp, 2020-07-13:
HI @alfonsohera, please help to provide following info:

- Which ADC are you using? 1 or 2?
- What's the priority of ADC sampling task?
- Does the problem still exist if disable ADC sampling but still keep the hardware timer and sampling task (but NOT do real ADC sampling)?

alfonsohera, 2020-07-20:
> HI @alfonsohera, please help to provide following info:
> 
> * Which ADC are you using? 1 or 2?
> * What's the priority of ADC sampling task?
> * Does the problem still exist if disable ADC sampling but still keep the hardware timer and sampling task (but NOT do real ADC sampling)?

Hi @liuzfesp. 
The used ADC is 1, the priority of the sampling task (which just blocks until the ADC interruption is triggered and unblocks it) when I created this issue was 11. I don't remember disabling just the sampling task, but I imagine that could be problematic since the used queue had just space for one element, and the delay used when sending the queue was the maximum.

zhangyanjiaoesp, 2020-10-28:
@alfonsohera Can you please pin the HW timer task  and SC's task to different core, then have a test?

zhangyanjiaoesp, 2021-03-01:
@alfonsohera 

I have tested locally using ADC1 and smartconfig example, it works well.

The test code and logs in the attachment.  

[code and log.tar.gz](https://github.com/espressif/esp-idf/files/6060552/code.and.log.tar.gz)

You said you will trigger an ADC sampling task every 100 ms, how long the sampling will take every time？

Alvin1Zhang, 2021-03-08:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more update, thanks.

closed 2021-03-08 by Alvin1Zhang