espressif/esp-idf #5206: Multiple files missing in v4.0.

amrutnrp, 2020-04-28:
Hi,

I'm using linux mint and downloaded the zip file from here.
while doing 'idf.py menuconfig' it is showing 'file missing' errors.
I tried putting the respective missing files from older version, but some other error keeps coming up. (logs below)
Please fix the basic issues in latest release.

> 
> kanha@kanha-HP-241-G1:~/esp/1/hello_world$ idf.py menuconfig
> Executing action: menuconfig
> Running cmake in directory /home/kanha/esp/1/hello_world/build
> Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/kanha/esp/1/hello_world"...
> Warn about uninitialized values.
> -- Found Git: /usr/bin/git (found version "2.17.1") 
> fatal: not a git repository (or any of the parent directories): .git
> -- The C compiler identification is GNU 8.2.0
> -- The CXX compiler identification is GNU 8.2.0
> -- The ASM compiler identification is GNU
> -- Found assembler: /home/kanha/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
> -- Check for working C compiler: /home/kanha/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
> -- Check for working C compiler: /home/kanha/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Check for working CXX compiler: /home/kanha/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
> -- Check for working CXX compiler: /home/kanha/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
> -- Building ESP-IDF components for target esp32
> CMake Error at ../tools/cmake/component.cmake:302 (message):
>   Include directory '/home/kanha/esp/1/components/mbedtls/mbedtls/include' is
>   not a directory.
> Call Stack (most recent call first):
>   ../tools/cmake/component.cmake:470 (__component_add_include_dirs)
>   ../components/mbedtls/CMakeLists.txt:4 (idf_component_register)
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/kanha/esp/1/hello_world/build/CMakeFiles/CMakeOutput.log".
> cmake failed with exit code 1
> kanha@kanha-HP-241-G1:~/esp/1/hello_world$ 
> kanha@kanha-HP-241-G1:~/esp/1/hello_world$ idf.py menuconfig
> Executing action: menuconfig
> Running cmake in directory /home/kanha/esp/1/hello_world/build
> Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/kanha/esp/1/hello_world"...
> Warn about uninitialized values.
> -- Found Git: /usr/bin/git (found version "2.17.1") 
> fatal: not a git repository (or any of the parent directories): .git
> -- The C compiler identification is GNU 8.2.0
> -- The CXX compiler identification is GNU 8.2.0
> -- The ASM compiler identification is GNU
> -- Found assembler: /home/kanha/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
> -- Check for working C compiler: /home/kanha/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
> -- Check for working C compiler: /home/kanha/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Check for working CXX compiler: /home/kanha/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
> -- Check for working CXX compiler: /home/kanha/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
> -- Building ESP-IDF components for target esp32
[... 93 lines cut ...]
> -- Found PythonInterp: /home/kanha/.espressif/python_env/idf4.2_py2.7_env/bin/python (found version "2.7.17") 
> -- Found Perl: /usr/bin/perl (found version "5.26.1") 
> -- App "hello-world" version: 1
> -- Adding linker script /home/kanha/esp/1/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
> -- Adding linker script /home/kanha/esp/1/components/esp_rom/esp32/ld/esp32.rom.ld
> -- Adding linker script /home/kanha/esp/1/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
> -- Adding linker script /home/kanha/esp/1/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
> -- Adding linker script /home/kanha/esp/1/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
> -- Adding linker script /home/kanha/esp/1/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
> -- Adding linker script /home/kanha/esp/1/hello_world/build/esp-idf/esp32/esp32_out.ld
> -- Adding linker script /home/kanha/esp/1/components/esp32/ld/esp32.project.ld.in
> -- Adding linker script /home/kanha/esp/1/components/esp32/ld/esp32.peripherals.ld
> CMake Error at ../tools/cmake/component.cmake:302 (message):
>   Include directory '/home/kanha/esp/1/components/cbor/tinycbor/src' is not a
>   directory.
> Call Stack (most recent call first):
>   ../tools/cmake/component.cmake:471 (__component_add_include_dirs)
>   ../components/cbor/CMakeLists.txt:1 (idf_component_register)
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/kanha/esp/1/hello_world/build/CMakeFiles/CMakeOutput.log".
> cmake failed with exit code 1
> kanha@kanha-HP-241-G1:~/esp/1/hello_world$ 
>

igrr, 2020-04-28:
> downloaded the zip file from here.

That will not work, sorry, because the project uses git submodules, and GitHub doesn't include them when you download a zip file.

Please follow the instructions in the getting started guide to download ESP-IDF using Git: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html

amrutnrp, 2020-04-29:
I tried again by git cmd and it solved the issue. Thanks for clarification. Ivan.

Conclusion- Only doing git command in terminal will work properly.

closed 2020-04-29 by amrutnrp