espressif/esp-idf #5207: Wifi stop responding after 1 hour (average) since 3.3.2 in micropython release

petrkr, 2020-04-28:
Using ESP IDF tag 3.3.2 with https://github.com/micropython/micropython causes wifi will stop work approximately within one hour.

At 3.3.1 we had problem with memory (see issue https://github.com/micropython/micropython/issues/5554) Which we fixed by update to 3.3.2, but that introduced bug https://github.com/micropython/micropython/issues/5984).

liuzfesp, 2020-04-30:
Hi @petrkr , 

WiFi stop/start/init/deinit is a basic regression/stress test before IDF releasing, our stress test case test the stop/start for more than 5 days before v3.3.2 is releasing, so I guess this issue should happen in very special scenario.

Could you help to provide following info:

- The detailed steps to reproduce this issue
- About how long time this issue can be reproduced? What's the frequency?
- Is it possible to give a minimal IDF example code that can reproduce this issue?

petrkr, 2020-05-01:
Trying get something about that. But it seems it is only at some set-ups.It occurs at our lab (but I will be there by Tuesday).

TP-Link with DD-WRT it seems to work
Netis seems to disconnect randomly
UPC router/wifi do that every hour
Mikrotik since 2 hours not occur yet. (This set-up I have with DEBUG on)

https://parallelgarden.surikata.info/d/WfPV1n6Wk/octopuslab-influxdemo?orgId=1&refresh=30s&var-places=All&from=now-3h&to=now

petrkr, 2020-05-01:
Timer 30sec RUN                                                                                                                  
Keepalive: 183, Error count: 0, Number of wifi reconnect: 0                                                                      
Error write to InfluxDB [Errno 113] EHOSTUNREACH                                                                                 
Error while sending to InfluxDb                                                                                                  
I (5525102) wifi:state: run -> init (0)                                                                                          
I (5525102) wifi:pm stop, total sleep time: 5196440559 us / 5510653348 us                                                        

I (5525102) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1                                                           
I (5525112) wifi: STA_DISCONNECTED, reason:8
Timer 30sec RUN                                                                                                                  
Keepalive: 184, Error count: 1, Number of wifi reconnect: 1                                                                      
Error write to InfluxDB 118                                                                                                      
Error while sending to InfluxDb                                                                                                  
I (5538662) network: event 1
I (5538832) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1                                                           
I (5538832) wifi:state: init -> auth (b0)                                                                                        
I (5538832) wifi:state: auth -> assoc (0)                                                                                        
I (5538842) wifi:state: assoc -> run (10)                                                                                        
I (5538852) wifi:connected with SSID, aid = 8, channel 1, BW20, bssid = xx:xx:xx:xx:xx:xx                     
I (5538852) wifi:security type: 3, phy: bgn, rssi: -70                                                                           
I (5538862) wifi:pm start, type: 1                                                                                               

I (5538862) network: CONNECTED
I (5538902) wifi:AP's beacon interval = 102400 us, DTIM period = 3                                                               
I (5541062) event: sta ip: 192.168.x.x, mask: 255.255.255.0, gw: 192.168.x.x
I (5541062) network: GOT_IP
Timer 30sec RUN                                                                                                                  
Keepalive: 185, Error count: 2, Number of wifi reconnect: 2                                                                      
Timer 30sec RUN                                                                                                                  
Keepalive: 186, Error count: 0, Number of wifi reconnect: 2                                                                      
Timer 30sec RUN

petrkr, 2020-05-05:
Here I have log with debug-on.
But seems nothing more than when debug of. Today I am whole day near that Access Point which do that.. Can you somehow assist how I can do more logs? Because you are right it seems to be very specific scenario and it does not happen on my other AP which I've tested

Take a look to those graphs
https://parallelgarden.surikata.info/d/WfPV1n6Wk/octopuslab-influxdemo?orgId=1&refresh=1m&from=now-1h&to=now

that with reconnect 'stairs' is on that AP, today I turn on another two at same AP to confirm it's not ESP32's fault (as it could be some "broken" piece)

[esp32-wifiupc-recnnect.txt](https://github.com/espressif/esp-idf/files/4579750/esp32-wifiupc-recnnect.txt)

petrkr, 2020-05-18:
@liuzfesp can you please provide me which DEBUG flags I should turn-on in sdkconfig to get more usesful logs?

It is already known it's this environment only and I can not discover why. Other devices (like cellphone, laptop...) does not have that issue. But all ESP32 there yes... I can also try ESP8266 if they do reconnect as-well, but I think nope or at least do not remember.

liuzfesp, 2020-06-04:
Hi @petrkr, we can't reproduce the behavior you observed, could you help to capture the WiFi packets and then we can do more analysis?

petrkr, 2020-06-04:
> Hi @petrkr, we can't reproduce the behavior you observed, could you help to capture the WiFi packets and then we can do more analysis?

I can do wireshark in promisc mode from other PC, but there will be just WPA2 traffic probably. If I can do something from ESP32 directly ? From wifi router I can not do anything than change SSID and password.

Maybe good point to test some cases like change WPA2/WPA TKIP/AES... Open... 

Bad is it is only once per hour, good is it's every hour and not random...

Also I do not want play with router much to not "broke" such situation, so we can do some debug on it...

For now we using auto-reconnect in code when data are not available, we do reconnect

liuzfesp, 2020-06-04:
Hi @petrkr, if this issue happens strictly every one hour, I suspect it may be related to group rekey.   Could you help to do following tests:

1. Keep the router in the mode that problem happens every one hour and capture the WiFi packets.
2. Change the router to open mode and have a test, capture the WiFi packets.

petrkr, 2020-06-09:
First attempt with packets:

[esp32-upc-reconnect-error2.zip](https://github.com/espressif/esp-idf/files/4752672/esp32-upc-reconnect-error2.zip)

petrkr, 2020-06-09:
Maybe it's connected with DHCP Lease time? But strange it happen to all devices at same time regardless on when been connected.. But when I had 1 hour lease time, I can see reconnect was every 30 minutes, and when I had 2 hours lease time, reconnect was around 1 hour... When I use lease time 5 minutes, it did not do reconnect within 1.5hour...

I will do more tests...

I have here scenario:

UPC Router (DHCP Disabled) + WiFi + Default GW to internet
Mikrotik router (DHCP Server)

Wifi set up as: WPA-PSK/WPA2-PSK TKIP+AES

petrkr, 2020-06-09:
One more logs and packets. And did ping from laptop.

[esp32-upc-reconnect-error3.zip](https://github.com/espressif/esp-idf/files/4754076/esp32-upc-reconnect-error3.zip)

liuzfesp, 2020-07-13:
Hi @petrkr according to the debug log:

```
Keepalive: 189, Error count: 0, Number of wifi reconnect: 0
Error write to InfluxDB -202
Error while sending to InfluxDb
I (5695297) wifi:state: run -> init (0)
```

Seems the WiFi is disconnected by the application by calling `esp_wifi_disconnect()` or calling `esp_wifi_connect()`, could you take a look at the place that `esp_wifi_disconnect()`/`esp_wifi_connect()`?  In what conditions these APIs will be called.

petrkr, 2020-07-13:
@liuzfesp yes, because network stack stop communication (os error -202 which means could not resolve hostname) then i have workaround to do reconnect, see code

https://github.com/octopusengine/octopuslab/blob/35b49f2b7407af6b53f74e1b98c17a3dc3fb3fe2/esp32-micropython/examples/database/influxdb.py#L50

But that reconnect (disconmect/connect) is did already when communication stop already.

HarveyRong-Esp, 2020-10-23:
@petrkr, I checked the Log you provided. This seems to be an active disconnection caused by the application layer calling disconnect. No abnormality was found in the network before disconnection. Can you check if your application has abnormal operations? ?

petrkr, 2020-10-23:
> @petrkr, I checked the Log you provided. This seems to be an active disconnection caused by the application layer calling disconnect. No abnormality was found in the network before disconnection. Can you check if your application has abnormal operations? ?

Seems you did not read while issue and also did not check source code.

Yes, i have 'WORKAROUND'!!!! Solution for this problem by doing software reconnect.

So whenever i'll get wifi not responding (getting timeouts on requests) i will do disconnect and connect again

Also at that LOG's I have mark WHERE exactly it stop responding and that Disconnect is after that as workaround.

Again check source-code, it's also in that issue

HarveyRong-Esp, 2020-12-08:
Hi @petrkr, 
```
if influx.write(keepalive=keepalive, errorcount=errorcount, reconnectcount=reconnectcount):
        errorcount = 0
else:
        print("Error while sending to InfluxDb")
        errorcount+=1
        if errorcount > 4:
            print("Too much errors, reboot")
            reset()
        else:
            net.sta_if.disconnect()
            net.connect()
 reconnectcount+=1
```
Does a write error mean that the network must go wrong? Can the error code be judged to prove that this is a network problem?
If it is determined that it is a network problem, can you provide both wireless data packets from STA to AP and wired data packets from AP to database server?

petrkr, 2020-12-10:
> If it is determined that it is a network problem, can you provide both wireless data packets from STA to AP and wired data packets from AP to database server?

Hello,
I will try to get some, but from latest testing there whenever it start to say this problem, there are no packets at all, so wireshark would be empty capture.

But because this issue regards to SDK 3.3.2 I think it it point-less to debug it against this version, more likely I will try compile it against latest 3.x SDK (I think 3.3.4 in this time?).

Also I will try to do example in esp-idf to minimalize code.... What I need to test:

1) Connect to wifi
2) send in while(true) every 1 minute HTTP(S) POST request to InfluxDB

if you can point me to FreeRTOS, it could save lot of time for studying... Or I will try use some example..

What environment for debugging you recommend ? VS Code + OpenOCD? Some other? What make options you suggest for better debugging of this Wireless and IP stack?

Thx

HarveyRong-Esp, 2020-12-11:
@petrkr，  you can simply modify  [esp_http_client_example](https://github.com/espressif/esp-idf/blob/release/v3.3/examples/protocols/esp_http_client/main/esp_http_client_example.cl)   to try reproduce.

Need to modify the url to the url of InfluxDB.
```
esp_http_client_config_t config = {
        .url = "http://httpbin.org/get",
        .event_handler = _http_event_handler,
    };
```

HarveyRong-Esp, 2021-01-18:
Hi @petrkr, Are there any updates?

petrkr, 2021-01-18:
Hello, not yet.  I had no free time to check that on IDF. I will try remember for tomorrow, because I will go to place where that wifi problem occurs.

HarveyRong-Esp, 2021-03-16:
Hi @petrkr , 
Are there any updates? If not, I will close issue. If there is issue, please open it at any time

petrkr, 2021-03-16:
Still had no time as many project ahead. But maybe we can close this as Micropython switched to SDK 4.0 and 4.2, so I have to rewrite whole network layer as it also changed from SDK 3 to 4.

So now I more need issue #6564 to be done as now it is not possible to change clock modes from code.

Alvin1Zhang, 2021-08-25:
Thanks for reporting and sharing updates, feel free to reopen.

closed 2021-08-25 by Alvin1Zhang