espressif/esp-idf #5209: Garbage values even after setting 26MHz XTAL

amrutnrp, 2020-04-29:
Hi,
Im using the following setup
Linux mint 64 bit,  ESP_IDF v4.0 , NodeMCU v1.0 and python 2.7

I am using the hello_world project from examples dir, after flashing, it is showing garbage values in monitor. I have tried correcting the XTAL freq to 26Mhz, but it still shows the same values.

Logs below for reference. 

> root@kanha-HP-241-G1:/home/kanha/esp/esp-idf/hello_world# idf.py -p /dev/ttyUSB0 flash monitor
Checking Python dependencies...
Python requirements from /home/kanha/esp/esp-idf/requirements.txt are satisfied.
Adding flash's dependency "all" to list of actions
Executing action: all (aliases: build)
Running ninja in directory /home/kanha/esp/esp-idf/hello_world/build
Executing "ninja all"...
[1/4] cd /home/kanha/esp/esp-idf/hello_world/build/esp-idf/partition_table && /usr/...o "*******************************************************************************"

> Partition table binary generated. Contents:
> *******************************************************************************
> # Espressif ESP32 Partition Table
> # Name, Type, SubType, Offset, Size, Flags
> nvs,data,nvs,0x9000,24K,
> phy_init,data,phy,0xf000,4K,
> factory,app,factory,0x10000,1M,
> *******************************************************************************
> [2/4] Performing build step for 'bootloader'
> ninja: no work to do.
> Executing action: flash
> Running esptool.py in directory /home/kanha/esp/esp-idf/hello_world/build
> Executing "/usr/bin/python /home/kanha/esp/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset write_flash @flash_project_args"...
> esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_freq 26m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 hello-world.bin
> esptool.py v2.8
> Serial port /dev/ttyUSB0
> Connecting....
> Detecting chip type... ESP8266
> Chip is ESP8266EX
> Features: WiFi
> Crystal is 26MHz
> MAC: 80:7d:3a:3d:ac:d6
> Uploading stub...
> Running stub...
> Stub running...
> Changing baud rate to 460800
> Changed.
> Configuring flash size...
> Compressed 3072 bytes to 103...
> Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 4874.4 kbit/s)...
> Hash of data verified.
> Compressed 25344 bytes to 14958...
> Wrote 25344 bytes (14958 compressed) at 0x00001000 in 0.3 seconds (effective 594.6 kbit/s)...
> Hash of data verified.
> Compressed 147552 bytes to 76734...
> Wrote 147552 bytes (76734 compressed) at 0x00010000 in 1.8 seconds (effective 674.5 kbit/s)...
> Hash of data verified.
> 
> Leaving...
> Hard resetting via RTS pin...
> Executing action: monitor
> Running idf_monitor in directory /home/kanha/esp/esp-idf/hello_world
> Executing "/usr/bin/python /home/kanha/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 /home/kanha/esp/esp-idf/hello_world/build/hello-world.elf -m '/usr/bin/python' '/home/kanha/esp/esp-idf/tools/idf.py' '-p' '/dev/ttyUSB0'"...
> --- idf_monitor on /dev/ttyUSB0 115200 ---
> --- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
> r$l����$�|

> 
>             
>              �
>               $�b|����r�"�p
>                                ��on�loN���
>                                           "p��#$`r$p�N�
>                                                            �l
>                                                                 
>                                                                  
>                                                                   #
>                                                                    o��o��l�
> 
>                                                                             "��oN�$��l �no
>                                                                                               d`or����
> 
>                                                                                                          ��`p�N�
>                                                                                                                     ����l
>                                                                                                                               B
>                                                                                                                                N�|l#����������
>                                                                                                                                                  #�|�o�
>                                                                                                                                                         l rll����$�|
>                                                                                                                                                                         
>                                                                                                                                                                         ��l�
>     #|����{�#�
>                 B��oN�$oN���
>                             "p��#$`r$p�o�
> 
>                                              �l
>                                                  
>                                                   
>                                                    #
>                                                     o��o����
>                                                              b��Nn�l��$`�no
>                                                                                 $`or���o
>                                                                                           �� p�n�
>                                                                                                        ۄ��$co�|$c�����$
>                                                                                                                                ���
>                                                                                                                                   c�|~�o�
> 
>                                                                                                                                           d`
>

Alvin1Zhang, 2020-04-29:
@amrutnrp Thanks for reporting. It seems you are using ESP8266 hardware, so you would like to develop software based on ESP32 or you would like to develop software based on ESP8266? Thanks.

amrutnrp, 2020-04-29:
@Alvin1Zhang, Thanks for pointing out.  I was going to use the ESP8266_RTOS_SDK repository, but there it was mentioned that it'll migrate to ESP-IDF eventually. 
Any comments on which repo to use for NodeMCU?

dobairoland, 2020-04-29:
@amrutnrp ESP-IDF isn't supporting ESP8266 at this moment. You just flashed an ESP8266 with a binary build for ESP32. I'm surprised that this was even possible.

amrutnrp, 2020-04-29:
haha. got it.

Conclusion- this repository only applicable for ESP32 module based boards. It'll not work for NodeMCU as that's based on the ESP-12 module.

closed 2020-04-29 by amrutnrp