espressif/esp-idf #5210: mesh和蓝牙同时使用

fzboffice, 2020-04-29:
mesh是固定根节点的，在node节点上会周期出现

E (345625) wifi:Error! Should use default passive scan time parameter for WiFi scan when Bluetooth is enabled!!!!!!

E (345878) wifi:scan number is 0
I (345879) mesh: [SCAN][ch:8]AP:0, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:2,i:2][00:00:00:00:00:00][FIXED-ROOT]<weak>
I (345884) mesh: [FAIL][1]root:0, fail:1, normal:0[weak]

这样的信息
请问default passive scan time是多少

dongdong004, 2020-05-07:
- 同时开启 Wi-Fi 和 蓝牙的情况下，Wi-Fi 和蓝牙是不能在同一时刻收发包的。比如 Wi-Fi 扫描的时间过长会导致这段时间蓝牙丢包。在 mesh 中使用了 Wi-Fi 的扫描接口并自定义了扫描的时间所以会打印这行日志，目前这行日志不影响 Wi-Fi 正常工作。
- 另外从您的日志中看到了 `scan number is 0`
  - 如果您仅有 root 和这个 node 节点两个设备则这种情况是可能发生的，多次扫描结果中应该可以看到这个节点扫描到 root 节点。
  - 如果您多次扫描结果均为 0 ，可以尝试禁用蓝牙对比测试并提供相关信息。

fzboffice, 2020-05-07:
> * 同时开启 Wi-Fi 和 蓝牙的情况下，Wi-Fi 和蓝牙是不能在同一时刻收发包的。比如 Wi-Fi 扫描的时间过长会导致这段时间蓝牙丢包。在 mesh 中使用了 Wi-Fi 的扫描接口并自定义了扫描的时间所以会打印这行日志，目前这行日志不影响 Wi-Fi 正常工作。
> * 另外从您的日志中看到了 `scan number is 0`
>   
>   * 如果您仅有 root 和这个 node 节点两个设备则这种情况是可能发生的，多次扫描结果中应该可以看到这个节点扫描到 root 节点。
>   * 如果您多次扫描结果均为 0 ，可以尝试禁用蓝牙对比测试并提供相关信息。

请问有没有什么解决办法，或者设置，让这个错误不出现吗

dongdong004, 2020-05-08:
目前暂时没有办法单独使这行日志不出现。您如果使用`esp_log_level_set`可以控制各个模块的日志打印。
如果后续有关于这行日志的更新会在这里提到 。

fzboffice, 2020-05-08:
> 目前暂时没有办法单独使这行日志不出现。您如果使用`esp_log_level_set`可以控制各个模块的日志打印。
> 如果后续有关于这行日志的更新会在这里提到 。

感谢您的回复

Alvin1Zhang, 2020-07-01:
Thanks for reporting and late reply. We have a fix under internal reviewing, the issue will be updated and closed on the fix is available on GitHub.

closed 2020-07-10 by espressif-bot