espressif/esp-idf #5211: Can't add second characteristic with esp_ble_gatts_add_char

saloid, 2020-04-29:
I am trying to add second characteristic with `esp_ble_gatts_add_char` and this not works - I can see only first char. 
And yes, I've have seen table method for multiple charactere
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

saloid, 2020-04-29:
Looks like I should wait callback (ESP_GATTS_ADD_CHAR_EVT) after first char adding and then add second.

GitOfLance, 2020-05-28:
Hi, @saloid 
At the beginning, I failed to create two Characteristic in `ESP_GATTS_CREATE_EVT` in succession, but the reason is different from yours. When calling `esp_ble_gatts_create_service` to create a service, you need to set the number of request handles. This value is 4 in the demo, because the `service` needs one and the `characteristic` needs three. If you want to add second Characteristic, then the value of `num_handle` needs to be changed to **7**.
The ESP-IDF I am using is v4.0 stable version.You can try this method.

saloid, 2020-05-28:
Hi, @GitOfLance
Thank you for your reply. In my case it works as I described - just need to create second one in callback. But, in this case I am assuming that single `characteristic` consumes only 2 handles, not 3! In [this](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/ble/gatt_server/main/gatts_demo.c) example, 4 handlers used only because there also one handler for descriptor.
So single `service` definition looks like this: 1x`service` + 2x`characteristic` + 1x`descriptor`

GitOfLance, 2020-05-29:
Hi, @saloid
In the subsequent use, it was found that if two Characteristic were created at the same time, the communication would be disordered. For example, the `Listen for notifications` function of `Characteristic_A` was enabled, but the data was received on `Characteristic_B`. Therefore, the correct way to create multiple characteristics should be to create one by one.
I am using this [demo](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_server). In its [explanation document](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/ble/gatt_server/tutorial/Gatt_Server_Example_Walkthrough.md#creating-services), a complete characteristic needs three handles. Use this demo. Once the Characteristic uses less than three handles, an error will be reported. I think you should give up some features of Characteristic, so fewer handles can work.

closed 2020-09-09 by saloid