espressif/esp-idf #5212: wifi sta 模式下 UDP 收不到广播数据

MGDG, 2020-04-29:
## Environment
- IDF version: v3.3.1-265-g6c9c253-dirty
- Build System:Make
- Compiler version: 1.22.0-80-g6c4433a
- Operating System: Linux

## Problem Description

我创建了一个UDP任务用来做广播以及接收其他设备的广播，在以太网模式下，工作正常。
但是在wifi模式下，可以使用sendto发送数据，但是调用recvfrom却收不到其他设备的广播数据

### 以下是UDP任务的部分代码

```cpp
#include "fireLinkage.h"

#include <stdlib.h>
#include <stdint.h>
#include <string.h>

#include "lwip/sockets.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "mbedtls/aes.h"
#include "cJSON.h"

/*TODO:联动要支持分组，消息中增加分组字段，要可指定同样的组才联动*/
#define FIRELINKAGE_UDP_PORT                        (7001)
#define FIRELINKAGE_MSG_BROADCASE_INTERVAL          (1000)              /*消防联动消息广播间隔，ms*/
#define FIRELINKAGE_TRIGGER_KEEP_TIME               (10*1000)           /*消防联动触发后的保持时间，只针对UDP和MESH类型的消防广播*/
#define FIRELINKAGE_REPLAYATT_TIME_LIMIT            (30*1000)           /*重放攻击时间戳误差允许范围，ms*/

static const char *TAG = "fireLinkage";

/**
 * @brief Create a udp server sock object
 * 
 * @return int 
 */
static int create_udp_serv_sock(void) {
    int serv_sock = socket(AF_INET, SOCK_DGRAM, 0);
    if(serv_sock < 0) {
        debug_e("udp server socket create err");
        return -1;
    }
    
    do {
        struct sockaddr_in serv_adr = {
            .sin_family = AF_INET,
            .sin_addr.s_addr = htonl(INADDR_ANY),
            .sin_port = htons(FIRELINKAGE_UDP_PORT),
        };

        if (bind(serv_sock,(struct sockaddr*) &serv_adr,sizeof(serv_adr)) == -1) {
            debug_e("udp server bind err");
            break;
        }

        struct timeval socket_timeout = {0, 5 * 1000};
        if(setsockopt(serv_sock, SOL_SOCKET, SO_RCVTIMEO, &socket_timeout, sizeof(struct timeval)) < 0) {
            debug_e("udp server SO_RCVTIMEO err");
            break;
        }
        return serv_sock;
    }while(0);
    close(serv_sock);
    return -1;
}

/**
 * @brief Create a udp client sock object
 * 
 * @param clit_adr 
 * @return int 
 */
static int create_udp_clit_sock(void) {
    int clit_sock = socket(AF_INET, SOCK_DGRAM, 0);
    if(clit_sock < 0) {
        debug_e("udp client socket create err");
        return -1;
    }

    do {
        int result=-1;
        socklen_t optlen = sizeof(int);
        if(setsockopt(clit_sock, SOL_SOCKET, SO_BROADCAST, &result, optlen) < 0) {
            debug_e("udp client socket set broadcast err");
            break;
        }

        if(setsockopt(clit_sock, SOL_SOCKET, SO_REUSEADDR, &result, optlen) < 0) {
            debug_e("udp client socket set reuseaddr err");
            break;
        }
        return clit_sock;
    }while(0);

    close(clit_sock);
    return -1;
}

static void fire_linkage_task(void *p) {

    socklen_t optlen = sizeof(struct sockaddr_in);
    struct sockaddr_in client = {0};
    struct sockaddr_in udp_broadcast_addr = {
        .sin_family      = AF_INET,
        .sin_port        = htons(FIRELINKAGE_UDP_PORT),
        .sin_addr.s_addr = htonl(INADDR_BROADCAST),
    };

    int clit_sock = create_udp_clit_sock();
    if(clit_sock < 0) {
        debug_e("udp client socket err");
        assert(NULL);
    }

    int serv_sock = create_udp_serv_sock();
    if(serv_sock < 0) {
        debug_e("udp server socket err");
        assert(NULL);
    }
    char buf[200];

    while(1) {

        int s=recvfrom(serv_sock,buf,sizeof(buf),0,(struct sockaddr*)&client,&optlen);
        if(s > 0) {
            udp_msg_proc(buf,s);
        }

        if(/**/) {
            int msgLen = create_udp_msg(true,buf,sizeof(buf));
            if(msgLen > 0) {
                sendto(clit_sock, buf,msgLen, 0, (struct sockaddr *)&udp_broadcast_addr, sizeof(udp_broadcast_addr));
            }
        }

        vTaskDelay(5 / portTICK_RATE_MS);
    }
}

void fire_linkage_init(void) {
    if(pdPASS != xTaskCreatePinnedToCore(fire_linkage_task, TAG,1024*5, NULL, 15, NULL, 1)) {
        debug_e("fire linkage task create err");
        assert(NULL);
    }
}

```

Alvin1Zhang, 2020-05-15:
@MGDG Thanks for reporting, and sorry for the late reply. We will look into.

freakyxue, 2020-06-11:
hi @MGDG 
When you set the broadcast property, the parameter should be set to 1 instead of - 1。
 int result=-1;
        socklen_t optlen = sizeof(int);
        if(setsockopt(clit_sock, SOL_SOCKET, SO_BROADCAST, &result, optlen) < 0)

closed 2020-11-02 by Alvin1Zhang