espressif/esp-idf #5213: UART Select() Dec(13) changed to Dec(10)

hassance3, 2020-04-29:
Use UART Select() example:
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/uart_select

Add print hex value in line 70:

```
67: if (FD_ISSET(fd, &rfds)) {
68:    char buf;
69:    if (read(fd, &buf, 1) > 0) {
70:       ESP_LOGI(TAG, "Received: %c - Hex value is: (%02X)", buf, buf);
```

**Setup:** [STM32] <--- UART ---> [ESP32]

When [STM32] send data using Serial UART, ESP32 show the correct Hex value of every byte received "_buf_", except when [STM32] send '\r' (Dec: 13, Hex 0D), [ESP32] always show '\n' (Dec: 10, Hex: 0A).

**Note:** I'm 100% sure that [STM32] send correct data, I use Teraterm, it's show [0D].
**Note:** Data array in [STM32] is `unsigned char MyData[2] = { 10, 13 };`

**Resume:** [STM32] send (Dec: 13, Hex 0D), [ESP32] receive (Dec: 10, Hex: 0A).

Thank you,

dobairoland, 2020-04-29:
Hi @hassance3,

The default option of [NEWLIB_STDIN_LINE_ENDING](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-newlib-stdin-line-ending) is NEWLIB_STDIN_LINE_ENDING_CR which causes the described transformation.

Setting NEWLIB_STDIN_LINE_ENDING_CRLF should solve your issue.

negativekelvin, 2020-04-29:
I think you mean to suggest NEWLIB_STDIN_LINE_ENDING_LF  LF: no modification is applied, input is sent to stdin as is

hassance3, 2020-04-29:
I confirm, when I save the data to a buffer, then debugging, I found that the buffer hold the correct value (Dec: 13, Hex 0D). That mean `ESP_LOGI()` do default STDIN LINE ENDING conversion.

So, this it's not an issue since the conversion didn't change the original data.

Thank you guys.

closed 2020-04-29 by hassance3