espressif/esp-idf #5214: Copying WiFi credential in flash from RAM not working

vtunr, 2020-04-29:
## Environment

- Development Kit:      [Custom PCB]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
     `v4.0-2-g943a9b780 (Slightly modified to work with our code`
- Build System:         [CMake & idf.py]
- Compiler version:
    `xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0`
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt & regular CMD].
- Using an IDE?: [No]
- Power Supply:         [external 5V]

## Problem Description

You set WiFi credential in RAM, you validate them by waiting to have an IP, then you switch to FLASH and copy the credential. If you reboot, nothing is in FLASH.
My use case is to receive credential from BLE, test them, and only write them in flash when validated.

## To reproduce

I managed to reproduce by modifying a bit wifi station example, available on [github](https://github.com/airboxlab/esp_wifi_credential_bug): 
* Erase all, flash 
```
python %IDF_PATH%/tools/idf.py all erase_flash flash monitor
```
* Wait for this debug 
```
I (3546) wifi station: Setting successful config to FLASH (ssid, passphrase)
I (3556) wifi station: connected to ap SSID:ssidpassword:passphrase
```
* Reboot
* You should see this if the bug is reproduced : 
[image]

I also dumped the NVS partition and can't find anything about the credentials.

Maybe I misuse something, but i'd like to understand what is the problem.

vtunr, 2020-05-07:
Anybody can help me ?

Alvin1Zhang, 2020-05-08:
@vtunr Thanks for reporting and sorry for slow turnaround, we will look into and provide updates.

esp-HarveyRong, 2020-05-11:
Hi, @vtunr, we have confirmed that this is a bug and we will fix it in early June, thanks for your support. The way to provide a workaround is as follows, Modify line 63 of your code to:
`wifi_config_t conf;`

vtunr, 2020-05-11:
Hi @esp-HarveyRong,

Thanks for the feedback. I used your workaround in the example code I gave you, and it's working.
Though, in my real code, it's not working. Code is similar though : 
```
if (esp_wifi_set_storage(WIFI_STORAGE_RAM) != ESP_OK){
	ESP_LOGE(TAG, "Failed to set storage Wi-Fi");
	return ESP_FAIL;
}
wifi_config_t wifi_cfg;
if(esp_wifi_get_config(ESP_IF_WIFI_STA, &wifi_cfg) != ESP_OK) {
	ESP_LOGE(TAG, "Failed to get Wi-Fi configuration in RAM");
	return ESP_FAIL;
}
if (esp_wifi_set_storage(WIFI_STORAGE_FLASH) != ESP_OK) {
	ESP_LOGE(TAG, "Failed to set storage Wi-Fi");
	return ESP_FAIL;
}
/* Configure Wi-Fi station with host credentials
	* provided during provisioning */
if (esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_cfg) != ESP_OK) {
	ESP_LOGE(TAG, "Failed to set Wi-Fi configuration");
	return ESP_FAIL;
}
```

What could be wrong ? Different compilation option ?

esp-HarveyRong, 2020-05-12:
Hi @vtunr, 
The difference may be caused by the compiler, suggest modify the code as follows:

```
wifi_config_t wifi_cfg;
if(esp_wifi_get_config(ESP_IF_WIFI_STA, &wifi_cfg) != ESP_OK) {
	ESP_LOGE(TAG, "Failed to get Wi-Fi configuration in RAM");
	return ESP_FAIL;
}

wifi_cfg.sta.listen_interval = 2;

if (esp_wifi_set_storage(WIFI_STORAGE_FLASH) != ESP_OK) {
	ESP_LOGE(TAG, "Failed to set storage Wi-Fi");
	return ESP_FAIL;
}
/* Configure Wi-Fi station with host credentials
	* provided during provisioning */
if (esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_cfg) != ESP_OK) {
	ESP_LOGE(TAG, "Failed to set Wi-Fi configuration");
	return ESP_FAIL;
}
```

BTW, `esp_wifi_set_storage()` decides that `set_config` is stored in RAM or FLASH, and cannot determine where `get_config` is obtained from, so calling `esp_wifi_set_storage()` before `get_config` is meaningless.

vtunr, 2020-05-12:
Thanks @esp-HarveyRong , the workaround is working :)

So if I understand correctly, the `esp_wifi_get_config` will pick it where it was last time set ? 

Should I leave this issue open until the fix is available in this repository ?

esp-HarveyRong, 2020-05-13:
> So if I understand correctly, the `esp_wifi_get_config` will pick it where `esp_wifi_get_config` was lastly used?

Right！It is worth noting that when the system is powered on, it will be obtained from FLASH.

> Should I leave this issue open until the fix is available in this repository ?

Of course, wait for the issue to be resolved before closing.

liuzfesp, 2020-07-10:
Similar issue: https://github.com/espressif/esp-idf/issues/5129

Alvin1Zhang, 2020-12-09:
@vtunr Thanks for the report and sorry for the slow turnaround, we have a fix under internal reviewing, the issue will close automatically once the fix is available on GitHub.

closed 2021-03-12 by espressif-bot