espressif/esp-idf #5215: Can we have every non-mandatory component optional?

PeterBorisenko, 2020-04-29:
What we have now:
```#error "None of Modbus communication mode is enabled. Please enable one of ASCII or RTU mode in Kconfig."```

What if I don't want to build modbus or, say, pthreads? It would be great if I could switch it off in Kconfig at all.

igrr, 2020-04-29:
Hi @PeterBorisenko, which build system are you using (CMake or Make)?

PeterBorisenko, 2020-04-30:
Hi @igrr. I use CMake.

igrr, 2020-04-30:
You can specify a short list of components in the project CMakeLists.txt file:

```cmake
cmake_minimum_required(VERSION 3.5)

set(COMPONENTS "esptool_py esp32")
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(myapp)
```

This list will be expanded by the build system to include all required components (which is still a long list at the moment).

If you then run `idf.py reconfigure`, you should see the list of components printed:
```
app_trace app_update bootloader bootloader_support cxx driver efuse esp32 esp_common esp_eth esp_event esp_netif esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump esptool_py freertos heap log lwip mbedtls newlib nvs_flash partition_table perfmon pthread soc spi_flash tcpip_adapter vfs wpa_supplicant xtensa
```

Note that even if the the component is built, its code will not be included into the final application, unless it is referenced somewhere. You can check the per-component contributions to the application size using `idf.py size-components` command.

PeterBorisenko, 2020-05-01:
> Note that even if the the component is built, its code will not be included into the final application, unless it is referenced somewhere.
Sure it is. But I afraid of having static declarations in some components. But `idf.py size-components` is a good tool I think. I would use it (instead of staring into map files).

igrr, 2020-05-01:
Static data should also not be included if it is not referenced by any of the included functions. In some cases, there is a dependency from the startup code to the initialization code of a component (like in case of pthread), so some code and data will be pulled in even if you don't call functions from that component.
You can also use `idf.py size-files` to get a breakdown at object file level.

By the way, of your question is resolved, please close the issue.

PeterBorisenko, 2020-05-01:
Ok, I will check this.
Thanks for advices!

closed 2020-05-01 by PeterBorisenko