espressif/esp-idf #5217: esp restart issue

lakshmisivareddy, 2020-04-30:
Hi i am using esp_idf 3.3
i am restarting esp32 but i am seeing below error

I (33489) wifi: new:<8,0>, old:<8,0>, ap:<255,255>, sta:<8,0>, prof:1
E (33493) event: system_event_sta_disconnected_handle_default 294 esp_wifi_internal_reg_rxcb ret=0x3014
E (33501) wifi: esp_wifi_connect 1134 wifi not start
I (33507) wifi: flush txq
I (33508) wifi: stop sw txq
I (33513) wifi: lmac stop hw txq

it's not a show stopper as such but i want to understand how can i fix this

Regards
Siva

liuzfesp, 2020-05-08:
Hi @lakshmisivareddy, could you provide following info:

- IDF commit ID that generates this log
- How to reproduce this issue

I think the log should have been removed in latest IDF release v3.3, namely tag v3.3.2.

liuzfesp, 2020-06-04:
HI @lakshmisivareddy,  the following error info is not true bug, we already remove the log in v3.3.2, please check:

E (33493) event: system_event_sta_disconnected_handle_default 294 esp_wifi_internal_reg_rxcb ret=0x3014
E (33501) wifi: esp_wifi_connect 1134 wifi not start

Alvin1Zhang, 2020-06-04:
@lakshmisivareddy Thanks for reporting, feel free to reopen if any updates or file a new ticket for new issue. Thanks.

closed 2020-06-04 by Alvin1Zhang