espressif/esp-idf #5218: HTTPS OTA

varunsavita, 2020-04-30:
(https://docs.espressif.com/projects/esp-idf/en/latest
IDF branch master

Development Kit: ESP32-DevKitC
Module or chip used: ESP32-WROOM-32
Build System: Make
Compiler version (run xtensa-esp32-elf-gcc --version to find it):
Operating System: Windows
(Windows only) environment type: MSYS2 mingw32
Power Supply: USB

[image]

[image]
[image]

[image]

Error
 No rule to make target '/home/varun/esp32/esp-idf/examples/Varun/https_request/server_certs/certs.pem', needed by 'embed_txt/certs.pem'.  Stop.

I am able to read soon file but I want to download .bin file.

mahavirj, 2020-05-06:
@varunsavita You are specifying file name in `CMakeLists.txt` as `ca_cert.pem` but actual file name seems to be `certs.pem`?

mahavirj, 2020-05-13:
@varunsavita Did you happen to solve this? Can we close the issue?

closed 2020-05-13 by varunsavita