espressif/esp-idf #5219: Build failing for example project in latest esp-idf version.

NewStackLearner, 2020-04-30:
## Environmen

- Development Kit:      [none]
- Kit version (DevKitC): [v2]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version : // v3.2-dev-1148-g96cd3b75c
- Build System:         [idf.py]
- Compiler version:
    // xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [external 3.3V]

## Problem Description
Build fail of restful server example in the latest esp-idf version.

//Detailed problem description goes here.
after installing the latest toolchain from github, i am unable to build any examples. All build output shows errors in stdatomic.h when building example project. I specifically want to use restful server example. This example has been recently added.

### Expected Behavior
Default example build should pass
### Actual Behavior
Build failing due to errors directed to stdatomic.h 
### Steps to reproduce

1. step1: install the toolchain with latest esp-idf from git repository
2. step2: build example of restful server

### Build log

/esp/esp-idf/components/newlib/include/stdatomic.h: In function 'atomic_flag_test_and_set_explicit':
/esp/esp-idf/components/newlib/include/stdatomic.h:271:31: error: request for member '__val' in something not a structure or union
  __atomic_exchange_n(&(object)->__val, desired, order)
                               ^
/esp/esp-idf/components/newlib/include/stdatomic.h:387:10: note: in expansion of macro 'atomic_exchange_explicit'
  return (atomic_exchange_explicit(&__object->__flag, 1, __order));
          ^
/esp/esp-idf/components/newlib/include/stdatomic.h:387:2: error: incompatible type for argument 1 of '__atomic_exchange_n'
  return (atomic_exchange_explicit(&__object->__flag, 1, __order));
  ^
/esp/esp-idf/components/newlib/include/stdatomic.h: In function 'atomic_flag_clear_explicit':
/esp/esp-idf/components/newlib/include/stdatomic.h:285:28: error: request for member '__val' in something not a structure or union
  __atomic_store_n(&(object)->__val, desired, order)
                            ^
/esp/esp-idf/components/newlib/include/stdatomic.h:394:2: note: in expansion of macro 'atomic_store_explicit'
  atomic_store_explicit(&__object->__flag, 0, __order);
  ^
/esp/esp-idf/components/newlib/include/stdatomic.h:394:2: error: incompatible type for argument 1 of '__atomic_store_n'
/esp/esp-idf/components/esp_eth/src/esp_eth.c: In function 'esp_eth_driver_install':
/esp/esp-idf/components/newlib/include/stdatomic.h:91:47: error: request for member '__val' in something not a structure or union
 #define atomic_init(obj, value)  ((void)((obj)->__val = (value)))
                                               ^
/esp/esp-idf/components/esp_eth/src/esp_eth.c:174:5: note: in expansion of macro 'atomic_init'
     atomic_init(&eth_driver->ref_count, 1);
     ^
/esp/esp-idf/components/esp_eth/src/esp_eth.c: In function 'esp_eth_driver_uninstall':
/esp/esp-idf/components/newlib/include/stdatomic.h:283:27: error: request for member '__val' in something not a structure or union
  __atomic_load_n(&(object)->__val, order)
                           ^
/esp/esp-idf/components/esp_eth/src/esp_eth.c:27:15: note: in definition of macro 'ETH_CHECK'
         if (!(a))                                                              
               ^
/esp/esp-idf/components/newlib/include/stdatomic.h:365:2: note: in expansion of macro 'atomic_load_explicit'
  atomic_load_explicit(object, memory_order_seq_cst)
  ^
/esp/esp-idf/components/esp_eth/src/esp_eth.c:215:15: note: in expansion of macro 'atomic_load'
     ETH_CHECK(atomic_load(&eth_driver->ref_count) == 1,
               ^
/esp/esp-idf/components/esp_eth/src/esp_eth.c:215:5: error: incompatible type for argument 1 of '__atomic_load_n'
     ETH_CHECK(atomic_load(&eth_driver->ref_count) == 1,
     ^
In file included from /esp/esp-idf/components/esp_eth/src/esp_eth.c:15:0:
/esp/esp-idf/components/esp_eth/src/esp_eth.c: In function 'esp_eth_increase_reference':
/esp/esp-idf/components/newlib/include/stdatomic.h:273:30: error: request for member '__val' in something not a structure or union
  __atomic_fetch_add(&(object)->__val, operand, order)
                              ^
/esp/esp-idf/components/newlib/include/stdatomic.h:355:2: note: in expansion of macro 'atomic_fetch_add_explicit'
  atomic_fetch_add_explicit(object, operand, memory_order_seq_cst)
  ^
/esp/esp-idf/components/esp_eth/src/esp_eth.c:341:5: note: in expansion of macro 'atomic_fetch_add'
     atomic_fetch_add(&eth_driver->ref_count, 1);
     ^
/esp/esp-idf/components/esp_eth/src/esp_eth.c:341:5: error: incompatible type for argument 1 of '__atomic_fetch_add'
In file included from /esp/esp-idf/components/esp_eth/src/esp_eth.c:15:0:
/esp/esp-idf/components/esp_eth/src/esp_eth.c: In function 'esp_eth_decrease_reference':
/esp/esp-idf/components/newlib/include/stdatomic.h:279:30: error: request for member '__val' in something not a structure or union
  __atomic_fetch_sub(&(object)->__val, operand, order)
                              ^
/esp/esp-idf/components/newlib/include/stdatomic.h:361:2: note: in expansion of macro 'atomic_fetch_sub_explicit'
  atomic_fetch_sub_explicit(object, operand, memory_order_seq_cst)
  ^
/esp/esp-idf/components/esp_eth/src/esp_eth.c:352:5: note: in expansion of macro 'atomic_fetch_sub'
     atomic_fetch_sub(&eth_driver->ref_count, 1);
     ^
/esp/esp-idf/components/esp_eth/src/esp_eth.c:352:5: error: incompatible type for argument 1 of '__atomic_fetch_sub'
/esp/esp-idf/components/esp_eth/src/esp_eth.c: At top level:
cc1: warning: unrecognized command line option '-Wno-frame-address'
esp-idf/esp_eth/CMakeFiles/__idf_esp_eth.dir/build.make:62: recipe for target 'esp-idf/esp_eth/CMakeFiles/__idf_esp_eth.dir/src/esp_eth.c.obj' failed
make[2]: *** [esp-idf/esp_eth/CMakeFiles/__idf_esp_eth.dir/src/esp_eth.c.obj] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:2517: recipe for target 'esp-idf/esp_eth/CMakeFiles/__idf_esp_eth.dir/all' failed
make[1]: *** [esp-idf/esp_eth/CMakeFiles/__idf_esp_eth.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

igrr, 2020-04-30:
@NewStackLearner sorry, the description is a bit confusing: you mention using the latest toolchain, but crosstool-ng-1.22.0-80-g6c4433a 5.2.0 is actually not the latest one.

And IDF v3.2-dev-1148-g96cd3b75c is not latest version either.

Do these version numbers you have provided really match your environment? From the log it seems that you are using a GCC 5.2 based toolchain with ESP-IDF 4.x.

To get the correct version of the tools installed, please follow the getting started guides for your IDF version. This is the link for the latest ESP-IDF version: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/, and you can switch to a different version using a menu in the bottom left corner.

NewStackLearner, 2020-05-01:
@igrr :+1:  I messed up all the installations trying to manage 3.3.2 and 4.0 versions simultaneously.
Instead of figuring out what is going wrong and have decided to abandon this approach.

Read somewhere on the forum that docker seems to be a better option to try new releases. Could not find enough info. So for benefit of folks, below steps I have taken, to successfully build esp-idf examples with docker image:

1. Downloaded and installed docker as per the instruction here on [IDF docker Image handling](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-docker-image.html). 
2. I also downloaded the [esp-idf 4.0.zip from github](https://github.com/espressif/esp-idf/tree/v4.0) 
3. I pulled docker image [release-v4.0](https://hub.docker.com/layers/espressif/idf/release-v4.0/images/sha256-ea5034e29d988b0274bd020df1cba3326cae0f48b28970f54445cd09fa41e154?context=explore) using below command `docker pull espressif/idf:release-v4.0` 
4. In the example makefile directory i ran below command `docker run --rm -v $PWD:/project -w /project espressif/idf:release-v4.0 idf.py build`

- [x] this command format works for all the esp-idf docker images. 
- [x] this solves my problem on building restful server example code
- [x] I am also able to flash using esptool from command line 

However, i am still not able to figure out how to get through with 

`idf.py menuconfig`

Any suggestions?

NewStackLearner, 2020-05-01:
As am not to sure of menuconfig availability via docker. I ended up fixing the esp-idf 4.0 build problem with fresh git clone of esp-idf 4.0 and then merge with commit 8cb3cc5. So now both build and menuconfig are available via direct command line.

closed 2020-05-01 by NewStackLearner