espressif/esp-idf #5220: Blank "Bluetooth" option in menuconfig in IDF v4.1-dev-1086-g93a8603c5

ROYMURLI, 2020-04-30:
Hi,
IDF v4.1-dev-1086-g93a8603c5
OS- Windows (msys32)

I am getting black page to configure bluetooth option after "make menuconfig". Screenshots are attached below.

[image]

[image]

dobairoland, 2020-05-04:
Hi @ROYMURLI. Yes, you are right that this is broken in the mentioned commit. I was able to reproduce. Can you please update to a newer version of v4.1? There were a couple of related fixes in v4.1 and I have checked that the latest version of v4.1 doesn't have this issue.

ROYMURLI, 2020-05-04:
Ok, Thanks for the update. I will update and will check from my side.

closed 2020-05-05 by ROYMURLI