espressif/esp-idf #5221: Infinite loop on esp_tls_conn_new_sync

vtunr, 2020-04-30:
## Environment

- Development Kit:      [Custom PCB]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
     `v4.0-2-g943a9b780 (Slightly modified of v4.0 to work with our code`
- Build System:         [CMake & idf.py]
- Compiler version:
    `xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0`
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt & regular CMD].
- Using an IDE?: [No]
- Power Supply:         [external 5V]

## Problem Description
So, I managed to get http not respond. I can't reproduce easily but i managed to get this stack : 
```
#0  0x4009ad99 in xQueueGenericReceive (xQueue=0x3ffd2d20, pvBuffer=0x3ffede60, xTicksToWait=<optimized out>,
    xJustPeeking=0) at ../../SDK/esp-idf/components/freertos/include/freertos/portable.h:214
#1  0x40151050 in sys_arch_mbox_fetch (mbox=<optimized out>, msg=0x3ffede60, timeout=60000)
    at ../../SDK/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c:297
#2  0x40151d6b in netconn_recv_data (conn=0x3ffd2bf4, new_buf=0x3ffedf00, apiflags=<optimized out>)
    at ../../SDK/esp-idf/components/lwip/lwip/src/api/api_lib.c:628
#3  0x40151e5f in netconn_recv_data_tcp (conn=0x3ffd2bf4, new_buf=0x3ffedf00, apiflags=8 '\b')
    at ../../SDK/esp-idf/components/lwip/lwip/src/api/api_lib.c:740
#4  0x40151ee3 in netconn_recv_tcp_pbuf_flags (conn=<optimized out>, new_buf=0x3ffedf00, apiflags=8 '\b')
    at ../../SDK/esp-idf/components/lwip/lwip/src/api/api_lib.c:821
#5  0x40141328 in lwip_recv_tcp (sock=0x3ffcb5d0 <sockets>, mem=0x3ffef2de, len=<optimized out>, flags=0)
    at ../../SDK/esp-idf/components/lwip/lwip/src/api/sockets.c:1001
#6  0x401425bb in lwip_recvfrom (s=<optimized out>, mem=0x3ffef2de, len=1626, flags=0, from=0x0,
    fromlen=<optimized out>) at ../../SDK/esp-idf/components/lwip/lwip/src/api/sockets.c:1268
#7  0x4014265d in lwip_read (s=54, mem=0x3ffef2de, len=1626)
    at ../../SDK/esp-idf/components/lwip/lwip/src/api/sockets.c:1310
#8  0x400d5e4c in esp_vfs_read (r=<optimized out>, fd=54, dst=0x3ffef2de, size=1626)
    at ../../SDK/esp-idf/components/vfs/vfs.c:446
#9  0x4000bdbe in ?? ()
#10 0x400017ed in ?? ()
#11 0x401575e8 in mbedtls_net_recv (ctx=<optimized out>, buf=0x3ffef2de "", len=1626)
    at ../../SDK/esp-idf/components/mbedtls/port/net_sockets.c:334
#12 0x40155820 in mbedtls_ssl_fetch_input (ssl=0x3ffd2460, nb_want=2968)
    at ../../SDK/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:2715
#13 0x40155ec3 in ssl_get_next_record (ssl=0x3ffd2460)
    at ../../SDK/esp-idf/components/mbedtls/mbedtls/include/mbedtls/ssl_internal.h:714
#14 mbedtls_ssl_read_record (ssl=0x3ffd2460, update_hs_digest=1)
    at ../../SDK/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:4338
#15 0x401571cf in mbedtls_ssl_parse_certificate (ssl=0x3ffd2460)
---Type <return> to continue, or q <return> to quit---
    at ../../SDK/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:5695
#16 0x401583b2 in mbedtls_ssl_handshake_client_step (ssl=0x3ffd2460)
    at ../../SDK/esp-idf/components/mbedtls/mbedtls/library/ssl_cli.c:3559
#17 0x40156927 in mbedtls_ssl_handshake_step (ssl=0x3ffd2460)
    at ../../SDK/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:8064
#18 0x4015694a in mbedtls_ssl_handshake (ssl=0x3ffd2460)
    at ../../SDK/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:8088
#19 0x401322a7 in esp_tls_low_level_conn (hostname=0x3ffd23cc "api.foobot.io", hostlen=13, port=<optimized out>,
    cfg=0x3ffd1f60, tls=0x3ffd2460) at ../../SDK/esp-idf/components/esp-tls/esp_tls.c:626
#20 0x40132432 in esp_tls_conn_new_sync (hostname=0x3ffd23cc "api.foobot.io", hostlen=13, port=443, cfg=0x3ffd1f60,
    tls=0x3ffd2460) at ../../SDK/esp-idf/components/esp-tls/esp_tls.c:686
#21 0x40130e4b in ssl_connect (t=0x3ffd1f18, host=0x3ffd23cc "api.foobot.io", port=443, timeout_ms=60000)
    at ../../SDK/esp-idf/components/tcp_transport/transport_ssl.c:74
#22 0x401b68b2 in esp_transport_connect (t=<optimized out>, host=0x3ffd23cc "api.foobot.io", port=443,
    timeout_ms=60000) at ../../SDK/esp-idf/components/tcp_transport/transport.c:165
#23 0x400e40de in esp_http_client_connect (client=0x3ffbcb5c)
    at ../../SDK/esp-idf/components/esp_http_client/esp_http_client.c:987
#24 0x400e4497 in esp_http_client_perform (client=0x3ffbcb5c)
    at ../../SDK/esp-idf/components/esp_http_client/esp_http_client.c:861
```

I had 60000ms timeout, which is probably too long as it is each lwip/tls call that this timeout, not the full http request as I initially thought. Though, after more than an hour, the function still didn't return.

I checked a bit where this could happens, and it seems that `esp_tls_conn_new_sync` could be the culprit, as you could end up in an infinite loop.

### Steps to reproduce

1. Make `mbedtls_ssl_read_record` in `mbedtls_ssl_parse_certificate` always return an error (I saw some -26880)
2. You will never get out of this while loop.

On my side, I think I just have a bad connection, but it's hard to reproduce naturally.

Did I missed something ? 
If I didn't, I think we should have a way to get out of this loop after some retry/time so we're not stuck there indefinitely.

david-cermak, 2020-05-05:
Hi @vtunr 

This bug should have been fixed in https://github.com/espressif/esp-idf/commit/08219f3c , not sure if that's included in your version (it is not in `v4.0-beta2` nor in `v4.0-rc`, didn't find the `943a9b780`)

vtunr, 2020-05-05:
Hi @david-cermak,
I'm running tag `v4.0` so apparently not included :/ 
Good to know it's fixed in future release.  
Any idea when v4.1 will be released ?

closed 2020-05-05 by vtunr

david-cermak, 2020-05-05:
This fix is certainly available on the `release/v4.0` branch and will be included in the upcoming `v4.0.1` release.

As for the 4.1, approximately at the end of May, the  `v4.1-RC` should be available