espressif/esp-idf #5224: Build failure in esp-idf with custom library

ryan4volts, 2020-05-01:
After doing a fresh windows install, I felt it was time to try and move to the latest IDF toolchain with CMake. I made the modificiations to my custom library to bring it up to v4, the files compile fine, but it fails at the linker stage. This occurs "sometimes". I tried for serveral hours to slowly cut code down and grow it back up to get the below error, though I have been unable to successfully get it to occur. I only know that it happens 100% of the time when I use all of my code, 50% of the time when I use half of my code and none of the time if I stick to the basic libraries. For each test I would do an idf.py fullclean followed by a idf.py build.

Installed:
Git-2.26.0-64-bit.exe
python-3.8.2.exe
esp-idf-tools-setup-2.3.exe

When installing esp-idf I used all the suggested applications above (git and python that I had already installed) and used all the suggested paths.
c/user/ryan/.espressif/...
c/user/ryan/Desktop/esp-idf/...

open cmd.exe
cd C:\Users\Ryan\Desktop\esp-idf\examples\get-started\hello_world
idf.py build
Everything works well

Now move everything to C++
Rename hello_world_main.c => hello_world_main.cpp
change "hello_world_main.c" to "hello_world_main.cpp" in CMakeList.txt
use a simple class from my library initiated by doing:
```

cExampleClass * tst = NULL;
extern "C" void main(void) {
    tst = new cExampleClass("string 1", "string 2");
}
```
I saw a previous issue #4490 that identified a similar thing, though I am already running the latest stable release.

The compile/linker error is as follows:
```
C:\Users\Ryan\Desktop\esp-idf\examples\get-started\hello_world>idf.py build
Checking Python dependencies...
Python requirements from C:\Users\Ryan\Desktop\esp-idf\requirements.txt are satisfied.
Executing action: all (aliases: build)
Running cmake in directory c:\users\ryan\desktop\esp-idf\examples\get-started\hello_world\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 c:\users\ryan\desktop\esp-idf\examples\get-started\hello_world"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.26.0.windows.1")
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/Ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/Ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/Ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/Ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/Ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: v4.0-dirty
-- Building ESP-IDF components for target esp32
-- Found PythonInterp: C:/Program Files (x86)/Python38-32/python.exe (found version "3.8.2")
-- Found Perl: C:/Program Files/Git/usr/bin/perl.exe (found version "5.26.2")
-- Adding linker script C:/Users/Ryan/Desktop/esp-idf/examples/get-started/hello_world/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Users/Ryan/Desktop/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/Users/Ryan/Desktop/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script C:/Users/Ryan/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/Ryan/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/Ryan/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Users/Ryan/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Users/Ryan/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Components: 4volts_toolchain app_trace app_update asio bootloader bootloader_support bt coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_ringbuf esp_rom esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter test_lib ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
[... 138 lines cut ...]
c:/users/ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0x14): undefined reference to `_Unwind_SetIP'
c:/users/ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o): in function `base_of_encoded_value(unsigned char, _Unwind_Context*)':
/builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/../libgcc/unwind-pe.h:114: undefined reference to `_Unwind_GetTextRelBase'
c:/users/ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/../libgcc/unwind-pe.h:116: undefined reference to `_Unwind_GetDataRelBase'
c:/users/ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/../libgcc/unwind-pe.h:118: undefined reference to `_Unwind_GetRegionStart'
c:/users/ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o): in function `parse_lsda_header(_Unwind_Context*, unsigned char const*, lsda_header_info*)':
/builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc:55: undefined reference to `_Unwind_GetRegionStart'
c:/users/ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o): in function `__gxx_personality_v0':
/builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc:438: undefined reference to `_Unwind_GetLanguageSpecificData'
c:/users/ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc:448: undefined reference to `_Unwind_GetIPInfo'
[... cut ...]
c:/users/ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_globals.o): in function `eh_globals_dtor(void*)':
/builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_globals.cc:83: undefined reference to `_Unwind_DeleteException'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

projectgus, 2020-05-20:
Hi @ryan4volts ,

Thanks for being patient while someone got back to you.

Just to be certain, can you please confirm the exact ESP-IDF version you're using is v4.0 (meaning: if you go to the IDF directory and run `git describe --tags` then the output is `v4.0` and not `v4.0-something`)?

Would it be possible for you to please share a .map file from one of the times that the project built correctly? This will give some clues about what the linking sequence is when the project builds correctly, and referencing this against the error posted above would be very useful.

The .map file doesn't contain any of your project's logic or internal code, but it does contain symbol addresses and names for all the symbols (ie functions, variables) linked into the project. If anything in there is sensitive and you'd rather not post it on GitHub then I can give you the information to send it to me privately, just let me know.

ryan4volts, 2020-05-23:
Hi @projectgus,

I am happy to send my full source privately.

Nonetheless, I can confirm it is simply "v4.0".
```
C:\Users\Ryan\Desktop\esp-idf>git describe --tags
v4.0
```
[did not work.map.txt](https://github.com/espressif/esp-idf/files/4671021/did.not.work.map.txt)
[Working.map.txt](https://github.com/espressif/esp-idf/files/4671025/Working.map.txt)

projectgus, 2020-05-25:
Hi @ryan4volts ,

Thanks for sending those map files. To check I understand what I'm seeing, I have two questions:

* Are these map files from the simplified "hello world" described in the first post?
* Is the "did not work.map" from a build which gave the exact error messages shown above? 

(Asking as the map file doesn't seem to mention any of the symbols shown in the posted error, usually for this kind of error I'd expect to see the symbols listed in the error also appear in the "Cross Reference Table" at the end of the .map file but I don't see them there.)

The error seems to have something to do with the linker failing to find some libstdc++ symbols used by C++'s exception handler code. Calling `new <...>` in your "Working" build causes the exception handler code to be linked at that point (ie all the missing symbols are found in the Working.map file). Not calling `new` in `app.cpp` means that seemingly nothing in the linked object files references exception handler code, but somehow the linker produces this error anyhow. It's unclear exactly why this happens.

Do your "library" components have anything noteworthy in them - for example do they include precompiled static libraries, or include CMake commands to set global compile or link flags?

How are you adding the additional libraries to the project?

ryan4volts, 2020-05-29:
Hi gus,

I spent a bit of time playing with the files to try and get them working in that situation. It may be why the previously attached files are different. It appears I have finally managed to repeat it with minimal code.

It appears to occur when 2 things are being used at the same time. When I use the new keyword with a custom component folder named one of the following:
(These BREAK the linker):
"C:\Users\Ryan\Desktop\esp-idf\components\4volts_toolchain"      //test leading numbers
"C:\Users\Ryan\Desktop\esp-idf\components\_volts_toolchain"      //test other non-alphanum char
"C:\Users\Ryan\Desktop\esp-idf\components\aaaaaaaa_aaaaaaaaaa" //to test the length
"C:\Users\Ryan\Desktop\esp-idf\components\a_a_a_a_a_a"                 //to test multiple underscores

(These WORK):
"C:\Users\Ryan\Desktop\esp-idf\components\avolts_toolchain"   //to match similar named folders

In terms of the cpp file.

This WORKS when compiling regardless of folder naming:
`#include "storage_fatfs.h"
cExample *example = NULL;

extern "C" void app_main() {
    static cExample tmpCLass("test1", "test2");
	example = &tmpCLass;
}`

and this FAILS when compiling with one of the linker breaking file names:
`#include "example.h"
cExample *example = NULL;

extern "C" void app_main() {
    example = new cExample("test1", "test2");
}`

Attached below is the toolchain in one zip file. This is placed in  "C:/Users/Ryan/Desktop/esp-idf/components"
[4volts_toolchain.zip](https://github.com/espressif/esp-idf/files/4699156/4volts_toolchain.zip)

The second is the application with full compiled outputs. This one successfully compiled using "avolts_toolchain" as the folder name. This is currently being compiled into "C:/Users/Ryan/Desktop/esp-idf/examples/get-started/".
[hello_world-successful-compile.zip](https://github.com/espressif/esp-idf/files/4699167/hello_world-successful-compile.zip)

The third is the application with full compiled outputs. This one failed with directory name "4volts_toolchain". This is currently being compiled into "C:/Users/Ryan/Desktop/esp-idf/examples/get-started/".
[hello_world-fails.zip](https://github.com/espressif/esp-idf/files/4699183/hello_world-fails.zip)

RENAME ALL FILE EXTENSIONS FROM ".zip" TO ".7z". zip compressed to ~30mb. 7z compressed to 8mb. GitHub only allows .zip file uploads. 7zip will open the file regardless of extension anyway.

FINALLY, this is the build output when I compile the application on a fail. I hope this is everything you need to reproduce this issue. I will obviously change my toolchain name in the mean time to continue with development.

`C:\Users\Ryan\Desktop\esp-idf\examples\get-started\hello_world>idf.py fullclean
Checking Python dependencies...
Python requirements from C:\Users\Ryan\Desktop\esp-idf\requirements.txt are satisfied.
Executing action: fullclean

C:\Users\Ryan\Desktop\esp-idf\examples\get-started\hello_world>idf.py build
Checking Python dependencies...
Python requirements from C:\Users\Ryan\Desktop\esp-idf\requirements.txt are satisfied.
Executing action: all (aliases: build)
Running cmake in directory c:\users\ryan\desktop\esp-idf\examples\get-started\hello_world\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 c:\users\ryan\desktop\esp-idf\examples\get-started\hello_world"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.26.0.windows.1")
-- Unexpected file in components directory: C:/Users/Ryan/Desktop/esp-idf/components/4volts_toolchain.zip
-- Component directory C:/Users/Ryan/Desktop/esp-idf/components/4volts_toolchain.zip does not contain a CMakeLists.txt file. No component will be added
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/Ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/Ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/Ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
[... 132 lines cut ...]
c:/users/ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0x10): undefined reference to `_Unwind_SetGR'
c:/users/ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0x14): undefined reference to `_Unwind_SetIP'
[... cut ...]
c:/users/ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/ryan/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_globals.o): in function `eh_globals_dtor(void*)':
/builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_globals.cc:83: undefined reference to `_Unwind_DeleteException'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1`

projectgus, 2020-06-25:
Hi @ryan4volts,

Thanks for providing all this comprehensive information and being patient while I got back to you. I'm also glad to hear that you've found a workaround.

I think that the name of the component is not triggering the bug directly, but it's causing CMake to link the libraries in different orders and as a result for some reason this causes the bug to trigger or not trigger. Even after some long studies of the map files and linker command lines I can't explain why one order triggers the failure and the other doesn't, though.

However there is a deeper issue that may explain the whole thing, which is that when a project (such as yours) has exceptions disabled there is no need to link these unwind-related functions at all. This is a bug in the cxx component of ESP-IDF and a fix has been submitted as #5380.

I've massaged the patch from that PR to apply to ESP-IDF `v4.0.1` tag:  

[0001-cxx-properly-disable-C-exceptions-mechanism.patch.txt](https://github.com/espressif/esp-idf/files/4828870/0001-cxx-properly-disable-C-exceptions-mechanism.patch.txt)

I wasn't able to reproduce the failure to link (probably because my paths are a little different still), but if you have an opportunity could you please try downloading this patch and applying it to ESP-IDF with `git apply patchfile.patch` and then see if the issue is still reproducable on your system?

Thanks.

ryan4volts, 2020-06-26:
Thanks @projectgus.

I renamed the txt file to patchfile.patch and ran the command. Did a "fullclean" and a build. It looks like it compiled at this point!

Good stuff!

closed 2021-04-27 by espressif-bot