espressif/esp-idf #5227: panic handler crash

szmodz, 2020-05-01:
In case of stack pointer corruption, the panic handler will enter infinite recursion, which eventually results in an interrupt WDT timeout, without providing any useful information about the crash.

In my opinion, the panic handler should switch stacks and print at least basic information before touching the original stack pointer, which may be invalid.

A crash in the panic handler will result in a DoubleException, 
https://github.com/espressif/esp-idf/blob/ad60568151ae3226375ddb05f33c067213048404/components/freertos/xtensa/xtensa_vectors.S#L440

> Double exceptions are not a normal occurrence. They indicate a bug of some kind.

test case:
https://github.com/szmodz/esp-idf/commits/panic_crash

example output:
```
Hello world!
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:5132
load:0x40078000,len:14484
load:0x40080400,len:4068
entry 0x40080624
W (205) boot.esp32: PRO CPU has been reset by WDT.
W (205) boot.esp32: WDT reset info: PRO CPU PC=0x4008e4b2
0x4008e4b2: _xt_context_save at /home/szmodz/invendyne/esp-idf/components/freertos/xtensa/xtens
a_context.S:149                                                                                      
                                                                                                    
E (205) boot.esp32: WDT reset info: &s CPU not support!
```

Alvin1Zhang, 2020-05-12:
@szmodz Thanks for reporting, we will look into later. Thanks.

user897943, 2021-03-18:
> @szmodz Thanks for reporting, we will look into later. Thanks.

Later when, 2032?