espressif/esp-idf #5229: setjump / longjump (Lua) crash

szmodz, 2020-05-01:
There's something wrong with setjump / longjump. It seems that if a context switch occurs during a longjump, the context will get corrupted sometimes, resulting in stack pointer corruption and an awesome crash.

The problem doesn't seem to occur if I disable interrupts (or task switching) around the setjump / longjump calls, but that's not a proper solution.

Lua relies on setjump / longjump, so these (standard C) functions are required. I've also tried building Lua as C++ and using exceptions instead, but it doesn't work at all (immediate crash).

Test case:
https://github.com/szmodz/esp-idf/commits/sjlj_crash

Also related:
https://github.com/espressif/esp-idf/issues/5227

Example output:
```
Hello world!
This is esp32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB external flash
Free heap: 4492760
[1000] sjlj tick 0
[1001] sjlj tick 1
[2000] sjlj tick 0
[2001] sjlj tick 1
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:5132
load:0x40078000,len:14484
load:0x40080400,len:4068
entry 0x40080624
W (172) boot.esp32: PRO CPU has been reset by WDT.
W (173) boot.esp32: WDT reset info: PRO CPU PC=0x4008b3f6
0x4008b3f6: _xt_context_save at /home/szmodz/invendyne/esp-idf/components/freertos/xtensa/xtensa_context.S:142                                                                                          

E (173) boot.esp32: WDT reset info: &s CPU not support!

Hello world!

```

szmodz, 2020-05-04:
The problem exists in the stable SDKs as well (though the symptoms might look a little different).

Do you know what is the exact cause? Any ETA?

surfskidude, 2020-06-01:
I have the exact same problem. Longjmp does not appear to restore the stack correctly.

surfskidude, 2020-06-21:
Just tried the latest commit 625bd5e (full checkout including compiler) and my Lua VM now seems to work just fine.

0xjakob, 2021-04-09:
Hi @szmodz @surfskidude, We've probably found what the issue is and have been working on this already. It seems to be some corruption if a context switch occurs during the longjmp.

igrr, 2021-05-06:
The issue has been fixed in https://github.com/espressif/esp-idf/commit/fc22e3c645bbdd5568d52eff77f69431776fd212.

closed 2021-05-06 by igrr