espressif/esp-idf #5230: Coap Server memory leak

DKrepsky, 2020-05-02:
## Environment

- Development Kit: none
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.2-dev-1318-gad6056815 (also affects release/v4.2 branch)
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System:  Linux
- Using an IDE?: No
- Power Supply: external 5V

## Problem Description
There is a memory leak of a few bytes (~252 bytes) after every request in the coap server, which is causing system failure on the long run.

In order to verify that it was not on our application logic, I tested with the sample code provided by esp-idf, which confirmed the memory leak.

### Expected Behavior
All allocated resources are freed after usage.

### Actual Behavior
Memory is not being deallocated after a response is sent.

### Steps to reproduce

Below is the code used to test the leakage, which is the example code with a ESP_LOGI to report heap usage after every request.

Run the attached sample and make GET requests on the UDP interface to see the memory leaking.

I'm using coap-client CLI application to test.

```
    coap-client -m get coap://192.168.0.106:5683/Espressif
```

### Code to reproduce this issue
[coap_server_leak.zip](https://github.com/espressif/esp-idf/files/4566913/coap_server_leak.zip)

## Debug Logs
https://gist.github.com/DKrepsky/28a2911eeff538f5803c09585d2dc3e6

## Other items if possible

- [x ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

mrdeep1, 2020-05-13:
@DKrepsky Thanks for raising this issue and doing some troubleshooting.

It appears that your issue is that the libcoap logic is creating a new session (which is 5-tuple based) for every new instance of a coap-client doing a request and not closing the session down at the end of the request.

Server based Sessions are maintained on a per client connection basis so that they can subsequently be re-used (especially if they are (D)TLS based and they contain the (D)TLS keys etc.) for when the client sends in the next packet.

There is an undocumented way of limiting the sessions that are 'idle', and this can be done using the patch below.
````
diff --git a/examples/protocols/coap_server/main/coap_server_example_main.c b/examples/protocols/coap_server/main/coap_se
index 0da1642..3156ba7 100644
--- a/examples/protocols/coap_server/main/coap_server_example_main.c
+++ b/examples/protocols/coap_server/main/coap_server_example_main.c
@@ -183,6 +183,7 @@ static void coap_example_server(void *p)
             ESP_LOGE(TAG, "coap_new_context() failed");
             continue;
         }
+        ctx->max_idle_sessions = 10;
 #ifdef CONFIG_COAP_MBEDTLS_PSK
         /* Need PSK setup before we set up endpoints */
         coap_dtls_spsk_t dtls_psk;
````
With this change in place, I would expect to see the first 10 requests that come in cause a heap usage, but not after that.  The value of 10 used above is fine if you are expecting only 1 or 2 clients to be making requests at the same time - scale up if there a large number of clients that are setting up new 'connections' (i.e using different source ports) at the same time.

Please confirm that this solves your memory leak issue and I will look at getting a better fix into libcoap.

DKrepsky, 2020-05-16:
Hello @mrdeep1, thanks for the response.

Indeed, setting max_idle_sessions solves the issue.

Maybe this should be added to coap_server example with a note or added to libcoap documentation.

Anyway, thanks again for the response. I'll close this issue.

closed 2020-05-16 by DKrepsky