espressif/esp-idf #5231: Problems compiling with Xtensa GCC 8.2.0 (crosstool-NG esp-2019r2)

slaff, 2020-05-02:
I am trying to port a project to work on the ESP32 microcontroller using the stable esp-idf v4.0 release. While doing so I was faced with a strange problem: the provided compiler or the newlib version prevents a simple C++ code from compiling. Here is a minimal C++ code: `dodo.cpp` that triggers the issue:
```
#include <functional>
```

And a minimal compilation command:
```
xtensa-esp32-elf-g++ \
-I${IDF_PATH}/components/newlib/platform_include \
-I${IDF_PATH}/components/newlib/include \
-c dodo.cpp -o dodo.o
```

The compilation ends in flames with the following error reported:
```
/home/slavey/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:155:44: error: expected initializer before '__result_use_check'
 void *reallocarray(void *, size_t, size_t) __result_use_check __alloc_size(2)
                                            ^~~~~~~~~~~~~~~~~~
/home/slavey/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:340:52: error: expected initializer before '__alloc_align'
 void * aligned_alloc(size_t, size_t) __malloc_like __alloc_align(1)
```

I have tried instructing the compiler to use C++11 (-std=c++11), C+17, GNU+11 and GNU++ 17 but the error did not go away.

What am I doing wrong?

Here is the information from idf_tools.py check:
```
Checking for installed tools...
Checking tool xtensa-esp32-elf
    version found in PATH: esp-2019r2-8.2.0
    version installed in tools directory: 1.22.0-80-g6c4433a-5.2.0
Checking tool esp32ulp-elf
    version found in PATH: 2.28.51.20170517
    version installed in tools directory: 2.28.51.20170517
Checking tool cmake
    version found in PATH: 3.14.4
Checking tool openocd-esp32
    version found in PATH: v0.10.0-esp32-20190313
    version installed in tools directory: v0.10.0-esp32-20190313
Checking tool ninja
    version found in PATH: 1.7.1
```

And the GCC version
```
Using built-in specs.
COLLECT_GCC=xtensa-esp32-elf-g++
COLLECT_LTO_WRAPPER=/home/slavey/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/8.2.0/lto-wrapper
Target: xtensa-esp32-elf
Configured with: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf --prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf --with-headers=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/include --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG esp-2019r2' --disable-__cxa_atexit --enable-cxx-flags=-ffunction-sections --disable-libgomp --disable-libmudflap --disable-libmpx --disable-libssp --disable-libquadmath --disable-libquadmath-support --with-gmp=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpfr=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpc=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-isl=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-nls --enable-multiarch --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custom-rtio --enable-libstdcxx-time=yes
Thread model: posix
gcc version 8.2.0 (crosstool-NG esp-2019r2) 
```

igrr, 2020-05-02:
Hi @slaff, `$IDF_PATH/components/newlib/include` should not be among the include paths, only `components/newlib/platform_include`.

The reason why adding it causes a failure is that `components/newlib/include` directory contains a copy of header from an earlier version of newlib. This directory is not added to the list of include paths by the IDF build system (Make or CMake), which GCC 8.2 compiler is used. It was removed entirely later, some time in between releases v4.0 and v4.1.

slaff, 2020-05-02:
Thanks a lot @igrr! That worked like a treat. I guess the following issues are also related to this: https://github.com/espressif/esp-at/issues/215, https://esp32.com/viewtopic.php?f=13&t=11366&p=46390#p46387, https://github.com/micropython/micropython/issues/4850.

igrr, 2020-05-02:
That's right, these look similar. Usually this issue happens when trying to use GCC 8.2 + newlib 3.0 based toolchain with IDF 3.x.

If your question is resolved, please close the issue. Thanks!

closed 2020-05-02 by slaff

geza-pycom, 2020-08-11:
@igrr : Sorry for posting on this already closed thread, however the following caught my attention: "_$IDF_PATH/components/newlib/include should not be among the include paths, only components/newlib/platform_include._"

Does this mean that regardless of anything (e.g. CPP sources) when compiling the Firmware using esp-idf 4 with latest Toolchain (at this moment it is 2020r2) the "_components/newlib/include_" must not be on the include path only "_newlib/platform_include_" ?

Thanks !