espressif/esp-idf #5232: Add support of enum/enum class for NVS C++ API

huming2207, 2020-05-02:
**Is your feature request related to a problem? Please describe.**

So far it seems that enum or enum class does not support NVS's C++ API. For example, this code shown below won't work:

```cpp
enum class foo: int {
    FOO,
    BAR
};

... 
auto val = foo::BAR;
nvs_handle->set_item("setting_foo",  val);
```

In my opinion, enum is very useful for state management. So I believe NVS should support enum.

**Describe the solution you'd like**

I think it's possible to use `std::is_enum` and `std::is_same` with `std::underlying_type` to detect the enum/enum class's matched integer type. Then cast it to integer.

Here's a quick & dirty example after 3 minutes of google search:

```cpp
#include <cstdio>
#include <type_traits>

enum class foo: int {
    FOO,
    BAR
};

int main() 
{
    printf("is enum: %s, underlying type is int: %s", 
        std::is_enum<foo>::value ? "true" : "false",
        std::is_same<int, typename std::underlying_type<foo>::type>::value ? "true" : "false");

    return 0;
}
```
It will output: 

```
is enum: true, underlying type is int: true
```

**Describe alternatives you've considered**

N/A

**Additional context**

N/A

huming2207, 2020-05-02:
To be more specific: 

```cpp
#include <cstdio>
#include <type_traits>

enum class foo: int {
    FOO,
    BAR
};

int main() 
{
    printf("is enum: %s, underlying type is int: %s, size: %zu", 
        std::is_enum<foo>::value ? "true" : "false",
        std::is_same<int, typename std::underlying_type<foo>::type>::value ? "true" : "false",
        sizeof(typename std::underlying_type<foo>::type));

    return 0;
}
```

The code will output: 

```
is enum: true, underlying type is int: true, size: 4
```

So we probably can do something like:

```cpp
template<typename T, typename std::enable_if<std::is_enum<T>::value, void*>::type = nullptr>
constexpr ItemType itemTypeOf()
{
    return static_cast<ItemType>(((std::is_signed<T>::value)?0x10:0x00) | sizeof(typename std::underlying_type<T>::type));
}

```

0xjakob, 2020-05-06:
Hi @huming2207, thanks for the suggestion. We'll have a look during the next few days.

huming2207, 2020-05-06:
@0xjakob Thanks, looking forward to seeing it

0xjakob, 2020-05-13:
@huming2207 We have probably found a solution, which we will add to the repository during the next weeks. The only downside with saving enums as integral data types is that the enum type is that the type information is lost, so when retrieving the enum back again, you need to be careful.

huming2207, 2020-05-13:
@0xjakob  Thanks! I'm looking forward to seeing it coming out later.

By the way, I think as long as the size is correct, then it should be fine. For example if I have two enum class that is `enum class foo: uint8_t` and `enum class bar: int`, then the size stored into NVS is 1 byte and 4 bytes instead of two 4 bytes.

0xjakob, 2020-05-14:
@huming2207 All data of integral data types is stored in the same structure in NVS. The type is distinguished by a special field in that structure. In other words, all those items occupy the same space, namely 32 bytes; no matter whether a uint8_t or an int64_t is stored.
You can find the Item class here: https://github.com/espressif/esp-idf/blob/master/components/nvs_flash/src/nvs_types.hpp#L50, the field describing the type is `ItemType datatype`

huming2207, 2020-05-18:
@0xjakob Thanks for pointing that out. Now I understood.

closed 2020-06-12 by igrr