espressif/esp-idf #5233: ISO-layout exit-shortcut for miniterm

bubmann, 2020-05-02:
It's problematic for people with ISO-layout keyboards to exit the terminal, as the right bracket is accessed with alt-9. I modified idf.py and added ctrl-e and use that to exit. any thoughts on making the software a tiny bit more accessible for europe?

igrr, 2020-05-04:
Hi @bubmann, there are `Ctrl-T Ctrl-X` and `Ctrl-T X` shortcuts which exit from IDF monitor, would these work for you?

bubmann, 2020-05-04:
of course ... I must have overread something in the documentary. I should have searched for exit in the code, or maybe just consult the help-menu. I'm sorry :) I've been working with your device for two weeks now after essentialy not coding for at least five years, it was quite a lot to take in. I could make it log bt/ble-data and communicate that over http, even ota-update via a http-PUT-request, but that I overlooked. still keeping my E-key though, at least until the next update

closed 2020-05-04 by bubmann

igrr, 2020-05-04:
Yeah, this shortcut is listed as the last item in the table of shortcuts: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-monitor.html#keyboard-shortcuts, we should move it directly next to `Ctrl+]` to make it easier to find. Thanks for pointing this out!