espressif/esp-idf #5235: WiFi/BT PM crashes

fpgamaster, 2020-05-03:
- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

## Environment

- Module or chip used:  ESP32-WROOM-32D
- IDF version: v4.2-dev-1320-g1aebfdf6a
- Build System: Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: external 3.3V
## Problem Description
WiFi and BT enabled. These crashes happen on heavy WiFi traffic during OTA update over HTTPS.
The same configuration used to work with the idf master branch few weeks ago ...
I believe they are related to BT and WiFi coexistence and power management.
Some insights will be appreciated.

// Crashes

 I (24411) ota: OTA write 1024
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400f0baa  PS      : 0x00060630  A0      : 0x800f5230  A1      : 0x3ffdecd0
0x400f0baa: pm_coex_slice_timeout_process at ??:?

A2      : 0x3ffcb790  A3      : 0x3ffbdef0  A4      : 0x00000000  A5      : 0x00060623
A6      : 0x00000001  A7      : 0x3ffca034  A8      : 0x00000000  A9      : 0x60033cb8
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00060a20  A13     : 0x3ffded38
A14     : 0x0000000b  A15     : 0x3ffbb508  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000001f  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace:0x400f0ba7:0x3ffdecd0 0x400f522d:0x3ffdecf0 0x400f52d1:0x3ffded10 0x400f4811:0x3ffded30 0x4009cb11:0x3ffded60
0x400f0ba7: pm_coex_slice_timeout_process at ??:?

0x400f522d: dbg_lmac_init at ??:?

0x400f52d1: pp_timer_do_process at ??:?

0x400f4811: ppTask at ??:?

0x4009cb11: vPortTaskWrapper at /home/xxxx/esp/esp-idf/components/freertos/xtensa/port.c:143

ELF file SHA256: cb20d5041b05a485

Rebooting...

===============================

I (85993) ota: OTA write 1024
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013f588: f144f3df 916274ce 8d2af30e
0x4013f588: btdm_sleep_enter_phase2_wrapper at /home/xxxx/esp/esp-idf/components/bt/controller/bt.c:860

Core  0 register dump:
PC      : 0x4013f58c  PS      : 0x00060830  A0      : 0x8018194c  A1      : 0x3ffe5bb0
0x4013f58c: queue_delete_wrapper at /home/xxxx/esp/esp-idf/components/bt/controller/bt.c:689

A2      : 0x00000038  A3      : 0x000000ff  A4      : 0x3ffb8d40  A5      : 0x3ffb8bcc
A6      : 0x00000d09  A7      : 0x3ffc86c0  A8      : 0x8013f574  A9      : 0x3ffe5b90
A10     : 0x00000001  A11     : 0x3ffb2950  A12     : 0x07ffffff  A13     : 0x00000000
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001e  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace:0x4013f589:0x3ffe5bb0 0x40181949:0x3ffe5bd0 0x40181ed5:0x3ffe5bf0 0x4009cb11:0x3ffe5c20
0x4013f589: btdm_sleep_enter_phase2_wrapper at /home/xxxx/esp/esp-idf/components/bt/controller/bt.c:862

0x40181949: r_rw_schedule at ??:?

0x40181ed5: btdm_controller_task at ??:?

0x4009cb11: vPortTaskWrapper at /home/xxxx/esp/esp-idf/components/freertos/xtensa/port.c:143

ELF file SHA256: e8fa436513c68d6f

Rebooting...

================================

I (24196) ota: OTA write 1024
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4019cc60: 243e50ed 7d248fbc e11b589f
0x4019cc60: periph_ll_get_clk_en_reg at /home/xxxx/esp/esp-idf/components/soc/src/esp32/include/hal/clk_gate_ll.h:205
 (inlined by) periph_ll_enable_clk_clear_rst at /home/xxxx/esp/esp-idf/components/soc/src/esp32/include/hal/clk_gate_ll.h:235
 (inlined by) periph_module_enable at /home/xxxx/esp/esp-idf/components/driver/periph_ctrl.c:27

Core  0 register dump:
PC      : 0x4019cc66  PS      : 0x00060f33  A0      : 0x80082ee0  A1      : 0x3ffdddb0
0x4019cc66: periph_ll_enable_clk_clear_rst at /home/xxxx/esp/esp-idf/components/soc/src/esp32/include/hal/clk_gate_ll.h:235
 (inlined by) periph_module_enable at /home/xxxx/esp/esp-idf/components/driver/periph_ctrl.c:27

A2      : 0x0000001d  A3      : 0x00000000  A4      : 0x681de144  A5      : 0x00000000
A6      : 0x3ffdcf88  A7      : 0x5130f02b  A8      : 0xe0e31281  A9      : 0x3f4309ac
A10     : 0xffff8800  A11     : 0x000003c9  A12     : 0x00060f20  A13     : 0x00060f23
A14     : 0x00000001  A15     : 0x3ffca0fc  SAR     : 0x0000001e  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace:0x4019cc63:0x3ffdddb0 0x40082edd:0x3ffdddd0 0x400f4d53:0x3ffdddf0 0x4009cba1:0x3ffdde20
0x4019cc63: periph_ll_enable_clk_clear_rst at /home/xxxx/esp/esp-idf/components/soc/src/esp32/include/hal/clk_gate_ll.h:235
 (inlined by) periph_module_enable at /home/xxxx/esp/esp-idf/components/driver/periph_ctrl.c:27

0x40082edd: esp_phy_common_clock_enable at /home/xxxx/esp/esp-idf/components/esp_wifi/src/phy_init.c:225

0x400f4d53: ppTask at ??:?

0x4009cba1: vPortTaskWrapper at /home/xxxx/esp/esp-idf/components/freertos/xtensa/port.c:143

ELF file SHA256: 44a324ed4f2f706d

Rebooting...

============================
I (24682) ota: OTA write 1024
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400d3980: 5050df00 87dd35a1 451b1601
0x400d3980: esp_phy_rf_deinit at /home/xxxx/esp/esp-idf/components/esp_wifi/src/phy_init.c:358

Core  0 register dump:
PC      : 0x400d3986  PS      : 0x00060330  A0      : 0x40000001  A1      : 0x3ffddce0
0x400d3986: esp_phy_rf_deinit at /home/xxxx/esp/esp-idf/components/esp_wifi/src/phy_init.c:358

A2      : 0x00000002  A3      : 0x00000001  A4      : 0x00000003  A5      : 0x3ffdc9a4
A6      : 0x40082ca4  A7      : 0x00000000  A8      : 0x00000007  A9      : 0x00000001
0x40082ca4: task_ms_to_tick_wrapper at /home/xxxx/esp/esp-idf/components/esp_wifi/esp32/esp_adapter.c:371

A10     : 0x0000000e  A11     : 0x3ffc9208  A12     : 0xb8329e75  A13     : 0x00000000
A14     : 0x0000001a  A15     : 0x40082eb4  SAR     : 0x0000001e  EXCCAUSE: 0x00000000
0x40082eb4: esp_phy_common_clock_enable at /home/xxxx/esp/esp-idf/components/esp_wifi/src/phy_init.c:219

EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace:0x400d3983:0x3ffddce0 0x3ffffffe:0x3ffddd10 |<-CORRUPTED
0x400d3983: esp_phy_rf_deinit at /home/xxxx/esp/esp-idf/components/esp_wifi/src/phy_init.c:358

ELF file SHA256: 44a324ed4f2f706d

Rebooting...

fpgamaster, 2020-05-03:
Same with 4.2 branch:

I (23102) ota: OTA write 1024
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400fe190: 3d03d719 d4147efa a77696ff
0x400fe190: noise_check_loop at /home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7.c:2367

Core  0 register dump:
PC      : 0x400fe196  PS      : 0x00060733  A0      : 0x800ff08e  A1      : 0x3ffddc10
0x400fe196: noise_check_loop at /home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7.c:2375

A2      : 0x6001d044  A3      : 0x00000001  A4      : 0x00000014  A5      : 0x000000a5
A6      : 0x000000a5  A7      : 0x00060023  A8      : 0xfffffea0  A9      : 0x0000004d
A10     : 0x0000003e  A11     : 0x3ffccc20  A12     : 0x0000fea0  A13     : 0xfffffeba
A14     : 0x0291ecea  A15     : 0x00000000  SAR     : 0x0000001e  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
xxxx00fe193:0x3ffddc10 0x400ff08b:0x3ffddc30 0x400d3851:0x3ffddc50 0x400d392a:0x3ffddc80 0x400de5bc:0x3ffddcb0 0x400f0838:0x3ffddcd0 0x400f0e23:0x3ffddd00 0x400f53a9:0x3ffddd20 0x400f545d:0x3ffddd40 0x400f499d:0x3ffddd60 0x4009cba1:0x3ffddd90
0x400fe193: noise_check_loop at /home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7.c:2367

0x400ff08b: phy_close_rf at /home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7.c:3642

0x400d3851: esp_phy_rf_deinit at /home/xxxx/esp/esp-idf-4.2/components/esp_wifi/src/phy_init.c:394

0x400d392a: esp_modem_sleep_enter at /home/xxxx/esp/esp-idf-4.2/components/esp_wifi/src/phy_init.c:437

0x400de5bc: wifi_rf_phy_disable at ??:?

0x400f0838: is_lmac_idle at ??:?

0x400f0e23: pm_sleep_delay_timeout_process at ??:?

0x400f53a9: dbg_lmac_init at ??:?

0x400f545d: pp_timer_do_process at ??:?

0x400f499d: ppTask at ??:?

0x4009cba1: vPortTaskWrapper at /home/xxxx/esp/esp-idf-4.2/components/freertos/xtensa/port.c:143

ELF file SHA256: 96fe78dbcf0aaa15

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:10984
load:0x40078000,len:22600
load:0x40080400,len:3980
entry 0x40080680

fpgamaster, 2020-05-03:
With v4.1-dev-1935-g647cb628a all is fine!

TimXia, 2020-12-15:
@fpgamaster It seems that there is some problems with Flash driver. The crashes randomly happened at the code which is in Flash. There may be some regression about IDF code. Please use the latest IDFv4.2 to test.

Alvin1Zhang, 2021-01-13:
@fpgamaster Thanks for reporting, would you please help share if any updates for this issue? Thanks.

fpgamaster, 2021-01-13:
Hi @Alvin1Zhang,

I have switched to v4.2 stable branch a time ago and have not seen this panic recently.

Regards

Alvin1Zhang, 2021-01-13:
@fpgamaster Thanks for sharing the updates, feel free to reopen if it happens again.

closed 2021-01-13 by Alvin1Zhang