espressif/esp-idf #5236: Const correctness of the Bluedroid API

XaviDCR92, 2020-05-04:
**Is your feature request related to a problem? Please describe.**

Using branch release/v4.1.

From the ble_spp_server_demo.c example: 

```{c}
static const uint16_t primary_service_uuid = ESP_GATT_UUID_PRI_SERVICE;
/// SPP Service
static const uint16_t spp_service_uuid = 0xABF0;
/* Other variables. */
///Full HRS Database Description - Used to add attributes into the database
static const esp_gatts_attr_db_t spp_gatt_db[SPP_IDX_NB] =
{
    //SPP -  Service Declaration
    [SPP_IDX_SVC]                      	=
    {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&primary_service_uuid, ESP_GATT_PERM_READ,
    sizeof(spp_service_uuid), sizeof(spp_service_uuid), (uint8_t *)&spp_service_uuid}},
/* Other members. */
}
```

This example stores UUIDs and data into non-volatile memory via the `const` qualifier. However, [esp_attr_desc_t](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_gatt_defs.html#_CPPv415esp_attr_desc_t) takes a read/write pointer to both the UUID and characteristic data, which means the example has to cast the `const` away from all references to avoid compiler warnings. I don't see where this data is actually being modified and, if it really is, the example would be calling **undefined behaviour according to the standard**.

I've seen other functions on the BLE API that do not seem to be const-correct and should be also fixed:

**`esp_ble_gap_start_advertising()`:** according to the implementation, `adv_params` is not being modified anywhere, but is rather copied into another instance, so there isn't any reason for `adv_params` not to be `const esp_ble_adv_params_t *`.

```{c}
esp_err_t esp_ble_gap_start_advertising(esp_ble_adv_params_t *adv_params)
{
    btc_msg_t msg;
    btc_ble_gap_args_t arg;

    ESP_BLUEDROID_STATUS_CHECK(ESP_BLUEDROID_STATUS_ENABLED);

    msg.sig = BTC_SIG_API_CALL;
    msg.pid = BTC_PID_GAP_BLE;
    msg.act = BTC_GAP_BLE_ACT_START_ADV;
    memcpy(&arg.start_adv.adv_params, adv_params, sizeof(esp_ble_adv_params_t));

    return (btc_transfer_context(&msg, &arg, sizeof(btc_ble_gap_args_t), NULL) == BT_STATUS_SUCCESS ? ESP_OK : ESP_FAIL);
}
```
Also, on this particular implementation, I'd also suggest replacing the call to `mempcy()` by an assignment for increased type safety as shown below: 
```{c}
arg.start_adv.adv_params = *adv_params;
```
**`esp_ble_gap_update_conn_params()`, `esp_ble_gap_config_adv_data()`, `esp_ble_gap_set_scan_params()`:** exactly the same issues as in the function above.

**`esp_ble_gap_config_adv_data_raw()`:** this function only takes a pointer to given raw data instead of copying it, but since `arg.cfg_adv_data_raw.raw_adv` is a read/write pointer, `raw_data` must also be a read/write pointer. However, if `arg.cfg_adv_data_raw.raw_adv` is never modified by the API, all these pointers should be `const`-qualified.
_Hint: `btc_transfer_context()` does not modify `arg`, so it seems possible to add the `const` qualifier._

I didn't read more of the API, but I am pretty sure there are other places where const correctness should be reviewed. I know status quo works, but placing those `esp_gatts_attr_db_t` buffers into `.data` instead of `.rodata` (instead of casting the `const` qualifier away, which isn't recommended either) is less efficient memory-wise.

Could these suggestions be please reviewed by the esp-idf team and merged when possible?

**Describe the solution you'd like**

Please review the BT API and add the `const`qualifier where required. I guess I should explain the benefits of const-correctness as they are already well-known to both C and C++ developers.

**Describe alternatives you've considered**

There is no (or should not be) alternative to const correctness. The API should be carefully reviewed and affected symbols should be properly `const` qualified.

**Additional context**

[Const-correctness according to Wikipedia](https://en.wikipedia.org/wiki/Const_(computer_programming))
[Question about const-correctness on Stack Overflow](https://stackoverflow.com/questions/136880/sell-me-on-const-correctness)

XaviDCR92, 2020-05-09:
Could you please confirm this issue? Thank you in advance.

XaviDCR92, 2020-06-01:
Hi, could you please confirm this issue? Could it be worked out for future version? Thank you.

zhp0406, 2023-11-16:
This modification affects many areas, requiring changes from the application layer to the stack. Making changes to only a part of it may trigger constant warnings. We will evaluate accordingly.

Rodrigodd, 2024-09-20:
This issue also affects the function `esp_ble_gatts_send_indicate`. I am trying to write the following code:

```c
uint16_t length = 0;
const uint8_t *value;
esp_ble_gatts_get_attr_value(my_attr_handle, &length, &value);
esp_ble_gatts_send_indicate(gatts_if, conn_id, my_attr_handle, length, value, false);
```

But `esp_ble_gatts_get_attr_value` returns a `const uint8_t*`, while `esp_ble_gatts_send_indicate` consumes a non-const `uint8_t*`. The documentation does not suggest that `esp_ble_gatts_send_indicate` mutates the value passed, so I will probably just cast the const away. However, it would be much preferable if the documentation or the function signature clarified whether this is safe to do or not.